espressif/esp-idf #11842: nimble examples don't work with nrf-connect

zzzh, 2023-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

core-esp32-c3

### Power Supply used.

USB

### What is the expected behavior?

ble acts normal

### What is the actual behavior?

I was trying spp_server, bleprph and blehr with nRF Connect on my Adroid phone
They can't connect with default settings( which have  Security manager legacy pairing and Security manager secure connections (4.2) checked  in nimble options), disconnect reason 531.
After uncheck these two options, it can connect, but don't respond to read, write, enable notify operations etc, after a while, connection lost, disconnect reason 520 for spp_server and bleprph, 531 for blehr.

### Steps to reproduce.

Try it with nRF Connect, you'll see.

### Debug Logs.

```plain
I (371) NimBLE_BLE_PRPH: BLE Host Task Started
I (371) NimBLE: GAP procedure initiated: stop advertising.

I (381) NimBLE: Device Address: 
I (381) NimBLE: 60:55:f9:74:71:be
I (381) NimBLE: 

I (391) NimBLE: GAP procedure initiated: advertise; 
I (391) NimBLE: disc_mode=2
I (391) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (401) NimBLE: 

I (411) uart: queue free spaces: 8
I (33141) NimBLE: connection established; status=0 
I (33141) NimBLE: handle=1 our_ota_addr_type=0 our_ota_addr=
I (33141) NimBLE: 60:55:f9:74:71:be
I (33151) NimBLE:  our_id_addr_type=0 our_id_addr=
I (33151) NimBLE: 60:55:f9:74:71:be
I (33161) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (33161) NimBLE: 68:e6:84:6f:e3:42
I (33171) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (33171) NimBLE: 68:e6:84:6f:e3:42
I (33181) NimBLE:  conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

I (33191) NimBLE: 

I (78301) NimBLE: disconnect; reason=520 
I (78301) NimBLE: handle=1 our_ota_addr_type=0 our_ota_addr=
I (78301) NimBLE: 60:55:f9:74:71:be
I (78311) NimBLE:  our_id_addr_type=0 our_id_addr=
I (78311) NimBLE: 60:55:f9:74:71:be
I (78321) NimBLE:  peer_ota_addr_type=1 peer_ota_addr=
I (78321) NimBLE: 68:e6:84:6f:e3:42
I (78331) NimBLE:  peer_id_addr_type=1 peer_id_addr=
I (78331) NimBLE: 68:e6:84:6f:e3:42
I (78341) NimBLE:  conn_itvl=36 conn_latency=0 supervision_timeout=500 encrypted=0 authenticated=0 bonded=0

I (78351) NimBLE: 

I (78351) NimBLE: GAP procedure initiated: advertise; 
I (78361) NimBLE: disc_mode=2
I (78361) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (78371) NimBLE:
```

ESPAbhinav, 2023-07-11:
Hi , we tried to reproduce the issue for mentioned examples at our end, but it is working fine.  Below is one screenshot and log file for the success case run. 

[image]

[Success logs.txt](https://github.com/espressif/esp-idf/files/12014558/Log.2023-07-11.17_02_01.txt)

So, i think, this might be some other issue. For debugging this, can you please provide. 
1. Your sdkconfig
2. Any changes done, if any
3. Debug enabled console logs.

Following is the procedure for getting debug enable logs:

                            Component Config  --> Log Output --> Default Log Verbosity --> Debug

                            Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs 

If possible, can you please run bleprph and blecent and perform the connection operation to confirm more ?  If issue is seen even in this setup at your end, then debug logs would be helpful to narrow down the failure reason.

zzzh, 2023-07-12:
Hi，ESPAbhinav,

Thanks for the respond.

I realized that the failure may because of some ble bond info saved in nvs before, so I erased the whole chip and tried again with nothing changed, it works!
Though, there's still something wrong, the first time read on characteristic which's uuid end with f6, it would have GATT AUTH FAIL，but only after this step, all functions work well.
Without this step, write to the characteristic which's uuid end with f7 would fail, you'll see a wait circle keep spining, no respond any more.
Above descriptions is about bleprph, I'll do more tests on blehr and spp_server.

zzzh, 2023-07-12:
I tested more, blehr works well, spp_server have same issue as bleprph.

zzzh, 2023-07-14:
Hi, @ESPAbhinav ,

I got what you need, the logs are long so I save them to files as attached.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/12045871/sdkconfig.txt)
[bleprph_gatt_auth_fail.txt](https://github.com/espressif/esp-idf/files/12045868/bleprph_gatt_auth_fail.txt)
[bleprph_direct_write_0x55.txt](https://github.com/espressif/esp-idf/files/12045869/bleprph_direct_write_0x55.txt)

zzzh, 2023-07-19:
Hi, @ESPAbhinav,

I tried it with another phone, it works well.
Seems it truely is a particular phone issue.

Thank you so much for the help.

closed 2023-07-19 by zzzh

ESPAbhinav, 2023-07-19:
Hi, @zzzh,
Thank you for confirmation.