espressif/esp-idf #11843: possible websocket server bug

JanuNoo, 2023-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I built the "ws_echo_server" example and wanted to run it using qemu.
I configured the project as follows:
1) I checked "Example Connection Configuration->connect using ethernet"
2) I selected "OpenCores Ethernet MAC (EXPERIMENTAL)" in "Example Connection Configuration->Ethernet Type".
I tested it and works ok until I close the websocket connection. Then the log shows an endless loop showing the following message:
"http: httpd_queue_work: failed to queue work".
It seems that when connection is closed,  "httpd_sess_trigger_close" function is called, that then calls "httpd_sess_trigger_close_" function and that last one calls "httpd_queue_work" function that it is where it seems the problem is.
My question, is this problem happening when using OpenCores Ethernet? or maybe it is a qemu problem? can somebody try it in some real hardware?
Thanks!

nopnop2002, 2023-07-14:
This is the result when using W5500.

```
I (3499) ws_echo_server: Starting server on port: '80'
I (3509) ws_echo_server: Registering URI handlers
I (3509) main_task: Returned from app_main()
I (37599) ws_echo_server: Handshake done, the new connection was opened
I (37609) ws_echo_server: frame len is 12
I (37609) ws_echo_server: Got packet with message: Hello world!
I (37609) ws_echo_server: Packet type: 1
```

ESP-YJM, 2023-10-18:
@JanuNoo It seems control socket fd has been close and send failed. Could you please enable LWIP_DEBUG and SOCKETS_DEBUG to check the log. Do you change any example code?

ESP-YJM, 2024-03-06:
@JanuNoo Any update for this issue? I am closing this issue. If you have, feel free to open again. Thanks.

closed 2024-03-06 by ESP-YJM