espressif/esp-idf #11844: esp_http_client_connect sometimes causes panic and MCU restart for TLS connections

kotyara12, 2023-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50002.230601 (5.0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

PlatformIO, package Espressif 32 (6.3.2)

### Power Supply used.

External 5V

### What is the expected behavior?

**In the event of ANY https connection failure, there should be no device reboot due to panic**

### What is the actual behavior?

esp_http_client_connect sometimes panics and restarts the MCU for TLS connections. The situation repeats itself once every few days when trying to send data to a remote server using an HTTPS client.

After restarting, I receive a backtrace, according to which I calculated the location of the problem.
```
🔸 Device started

Firmware version: 20230704.067
Restart reason: EXCEPTION / PANIC
CPU0: Software reset CPU
CPU1: Software reset CPU
HEAP: 265.5kB : 103.8kB (39.1%) : 53.6kB (20.2%) 07/09/2023 07:38:48
TRACE: 0x400d42c0 0x400f0e72 0x40082b61 0x40083880 0x4009449d 0x4011d617 0x4011d4d4 0x4011641a 0x4011771a 0x4011791b 0x401159 ed 0x40115c00 0x40126ff5 0x40125a6c 0x40124971 0x401249d9 0x400ec5a1 0x400ec2a5 0x400ec375 0x40106e0e 0x401720e3 0x400eda97 0x400 edf05 0x400da20e 0x400da37b

07/10/2023 09:53:09
```

**To hell with the data, let them disappear, but the restart disrupts the normal operation of the mechanisms. Problems sending data to the server should not cause a restart!!!!!

Any ideas how to prevent panic?**

### Steps to reproduce.

Sources on GitHub. The problem is in this particular line:
https://github.com/kotyara12/reDataSend/blob/master/src/reDataSend.cpp#L319

### Debug Logs.

```plain
c:\PlatformIO\village_greenhouse\ota>bt_rav_1 0x400d42c0 0x400f0e72 0x40082b61 0x40083880 0x4009449d 0x4011d617 0x4011d4d4 0x4011641a 0x4011771a 0x4011791b 0x401159ed 0x40115c00 0x40126ff5 0x40125a6c 0x40124971 0x401249d9 0x400ec5a1 0x400ec2a5 0x400ec375 0x40106e0e 0x401720e3 0x400eda97 0x400edf05 0x400da20e 0x400da37b

c:\PlatformIO\village_greenhouse\ota>c:\Users\razzhivinav\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -pfiaC -e c:\PlatformIO\village_greenhouse\ota\firmware\firmware_1.elf 0x400d42c0 0x400f0e72 0x40082b61 0x40083880 0x4009449d 0x4011d617 0x4011d4d4 0x4011641a 0x4011771a 0x4011791b 0x401159ed 0x40115c00 0x40126ff5 0x40125a6c 0x40124971 0x401249d9 0x400ec5a1 0x400ec2a5 0x400ec375 0x40106e0e 0x401720e3 0x400eda97 0x400edf05 0x400da20e 0x400da37b
0x400d42c0: __wrap_esp_panic_handler at C:/PlatformIO/libs/system/reEsp32/src/reEsp32.cpp:489
0x400f0e72: panic_handler at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:192
0x40082b61: xt_unhandled_exception at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_system\port/panic_handler.c:223
0x40083880: _xt_user_exc at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\freertos\FreeRTOS-Kernel\portable\xtensa/xtensa_vectors.S:709
0x4009449d: ?? ??:0
0x4011d617: esp_mpi_read_result_hw_op at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\port\esp32/bignum.c:156
0x4011d4d4: mbedtls_mpi_mul_mpi at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\port/esp_bignum.c:537
0x4011641a: mbedtls_mpi_mul_mod at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecp.c:1026
0x4011771a: ecp_double_add_mxz at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecp.c:2410
 (inlined by) ecp_mul_mxz at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecp.c:2478
 (inlined by) ecp_mul_restartable_internal at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecp.c:2562
 (inlined by) ecp_mul_restartable_internal at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecp.c:2520
0x4011791b: mbedtls_ecp_mul_restartable at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecp.c:2597
0x401159ed: ecdh_gen_public_restartable at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecdh.c:82
 (inlined by) mbedtls_ecdh_gen_public at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecdh.c:96
0x40115c00: ecdh_make_public_internal at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecdh.c:518
 (inlined by) mbedtls_ecdh_make_public at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ecdh.c:552
0x40126ff5: ssl_write_client_key_exchange at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls12_client.c:2976
 (inlined by) mbedtls_ssl_handshake_client_step at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls12_client.c:3721
0x40125a6c: __wrap_mbedtls_ssl_handshake_client_step at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\port\dynamic/esp_ssl_cli.c:198 (discriminator 15)
0x40124971: mbedtls_ssl_handshake_step at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls.c:3730
 (inlined by) mbedtls_ssl_handshake_step at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls.c:3682
0x401249d9: mbedtls_ssl_handshake at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls.c:3795
 (inlined by) mbedtls_ssl_handshake at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\mbedtls\mbedtls\library/ssl_tls.c:3771
0x400ec5a1: esp_mbedtls_handshake at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp-tls/esp_tls_mbedtls.c:192
0x400ec2a5: esp_tls_handshake at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp-tls/esp_tls.c:85 (discriminator 15)
 (inlined by) esp_tls_low_level_conn at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp-tls/esp_tls.c:443 (discriminator 15)
0x400ec375: esp_tls_conn_new_sync at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp-tls/esp_tls.c:470
0x40106e0e: ssl_connect at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\tcp_transport/transport_ssl.c:109
0x401720e3: esp_transport_connect at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\tcp_transport/transport.c:123
0x400eda97: esp_http_client_connect at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_http_client/esp_http_client.c:1289
0x400edf05: esp_http_client_perform at C:\Users\razzhivinav\.platformio\packages\framework-espidf\components\esp_http_client/esp_http_client.c:1136
0x400da20e: dsCallDataApi(dataChannel_t*) at C:/PlatformIO/libs/clouds/reDataSend/src/reDataSend.cpp:319
 (inlined by) dsCallDataApi(dataChannel_t*) at C:/PlatformIO/libs/clouds/reDataSend/src/reDataSend.cpp:300
0x400da37b: dsTaskExec(void*) at C:/PlatformIO/libs/clouds/reDataSend/src/reDataSend.cpp:481
```

hmalpani, 2023-07-17:
Hello @kotyara12 
Thanks for reporting the issue.
Can you please share an application that can help reproduce this issue? Also, share the sdkconfig file with which you can reproduce this issue.

kotyara12, 2023-07-26:
It seems that the problem was that the stack of one of the tasks was overflowing and overwriting "foreign" memory. My apologies for disturbing

closed 2023-07-26 by kotyara12