espressif/esp-idf #11845: SPI W5500 Ethernet has unrecoverable errors which in turn prevents communications over WiFi

bensuffolk, 2023-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4.5

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board with WRover-E-N16R8 and W5500 Ethernet

### Power Supply used.

External 5V

### What is the expected behavior?

On V4.4.5 There SPI ethernet with a W5500 is very good, however once there is a fault found its never possible to recover from it, and the WiFI ethernet is then no longer accessible either.

I have a wired Ethernet network with a couple of devices connected. The ESP acting as a DHCP server and running an embedded MQTT server for these devices to connect to the ESP.

I also have a WiFi connection that connects to the internet to do SNTP. The Wired ethernet can bridge to the Internet over WiFi using NAT:
```
ip_napt_enable(eth_address.ip.addr, true);
```

The WiFi is also used to serve an HTTP server to allow clients to configure the settings for the system.

If there is an error on SPI ethernet it never recovers and more importantly the ESP is no longer responding to network requests from the WiFi. E.g. HTTP requests simple timeout.

I can force an SPI ethernet error by allocating a large block of memory from the ESP, and then attempting to connect a device on the wired ethernet.

Once there has been an error there is no recovery and the ESP gets in a loop repeatedly trying to read a PHY register and updating the link duplex spleed.

### What is the actual behavior?

I would expect the rest of the network stack not to stop working, and the ESP to continue to accept and process network requests via WiFi.

I would also expect to see some form of ethernet event posted so that my application can try to recover, by shutting down the ethernet and restarting it.

### Steps to reproduce.

Allocate enough memory to cause an Ethernet issue. This will be dependant on how much memory your application is already using for other things.

Connect a device to the SPI ethernet

Send and receive some ethernet packets of enough size to cause a "no mem for receive buffer"

### Debug Logs.

```plain
E (07:18:12.909) w5500.mac: no mem for receive buffer
E (07:18:13.045) w5500.mac: no mem for receive buffer
E (07:18:13.345) w5500.mac: no mem for receive buffer
E (07:18:13.646) w5500.mac: no mem for receive buffer
E (07:18:13.950) w5500.mac: w5500_write(74): spi transmit failed
E (07:18:13.951) w5500.mac: w5500_write_buffer(173): write TX buffer failed
E (07:18:13.952) w5500.mac: emac_w5500_transmit(528): write frame failed
E (07:18:14.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:14.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:14.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (07:18:16.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:16.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:16.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (07:18:18.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:18.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:18.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (07:18:20.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:20.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:20.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (07:18:22.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:22.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:22.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (07:18:24.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:24.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:24.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (07:18:26.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:26.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:26.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (07:18:28.833) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (07:18:28.833) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (07:18:28.837) w5500.phy: w5500_get_link(108): update link duplex speed failed
```
```

### More Information.

I tried to move my application to V5.02 and found that the ethernet fails in a similar loop (Reason unknown) within 2 - 5 hours of the application running. Whereas on V4.4.5 the network can be running for a month without any issues. Same Board, same SPI bus speed.

When on V5.0.2 there is also still no event posted to show a network error, and the WiFi still stops processing network connections.

kostaond, 2023-07-11:
@bensuffolk thank you for reporting potential issue.

Do you use PSRAM in your application? And could you please enable debugging logs in SPI master and try to reproduce the issue?

bensuffolk, 2023-07-11:
I do use PSRAM in my project, but nit any bank switching just the first 4M.

I will enable the debug logs whenI am back at my desk and provide them.

bensuffolk, 2023-07-11:
For completeness I am using the following GPIO for the W5500 :

```
 CS GPIO 15
 INT GPIO 4
 SCLK GPIO 14
 MOSI GPIO 13
 MISO GPIO 12
 RST GPIO 2
```

I'm also using the I2C bus with : 

```
SDA GPIO 21
SCL GPIO 22
```

I'm also using UART 1:

```
TX GPIO 21
RC GPIO 23
RTS GPIO 22
```

and UART  2:

```
TX GPIO 5
RX GPIO 19
RTS GPIO 18
```

With UART 0 being used as the console for debug etc during development.

And then 3 status LEDs on :

```
GPIO 27
GPIO 26
GPIO 33
```

bensuffolk, 2023-07-11:
I have some logs, but they are. in my opinion, not very revealing. Additional info I forgot to include above I have the SPI controlled for the W5500 as HSPI and the PSRAM controller as VSPI

```
I (31) boot: ESP-IDF 4.4.5 2nd stage bootloader
I (31) boot: compile time 15:50:22
I (31) boot: chip revision: v3.0
I (31) qio_mode: Enabling default flash chip QIO
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : QIO
I (43) boot.esp32: SPI Flash Size : 16MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00015000
I (66) boot:  1 otadata          OTA data         01 00 0001e000 00002000
I (73) boot:  2 ota_0            OTA app          00 10 00020000 005f0000
I (79) boot:  3 ota_1            OTA app          00 11 00610000 005f0000
I (86) boot:  4 spiffs           Unknown data     01 82 00c00000 00400000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=3d7d4h (251860) map
I (181) esp_image: segment 1: paddr=0005d7fc vaddr=3ffb0000 size=0281ch ( 10268) load
I (185) esp_image: segment 2: paddr=00060020 vaddr=400d0020 size=c1ab8h (793272) map
I (430) esp_image: segment 3: paddr=00121ae0 vaddr=3ffb281c size=00ac8h (  2760) load
I (431) esp_image: segment 4: paddr=001225b0 vaddr=40080000 size=19b2ch (105260) load
I (483) boot: Loaded app from partition at offset 0x20000
I (484) boot: Disabling RNG early entropy source...
I (493) psram: This chip is ESP32-D0WD
I (501) psram: PSRAM is in 2T mode
I (501) spiram: Found 32MBit SPI RAM device
I (501) spiram: SPI RAM mode: flash 80m sram 80m
I (501) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (507) cpu_start: Pro cpu up.
I (510) cpu_start: Starting app cpu, entry point is 0x400834f4
I (0) cpu_start: App cpu up.
I (1050) spiram: SPI SRAM memory test OK
I (1082) cpu_start: Pro cpu start user code
I (1082) cpu_start: cpu freq: 240000000
I (1082) cpu_start: Application information:
I (1093) cpu_start: Compile time:     Jul 11 2023 15:49:37
I (1098) cpu_start: ELF file SHA256:  dbd8b7f76117b848...
I (1103) cpu_start: ESP-IDF:          4.4.5
I (1107) cpu_start: Min chip rev:     v3.0
I (1111) cpu_start: Max chip rev:     v3.99 
I (1115) cpu_start: Chip rev:         v3.0
I (1119) heap_init: Initializing. RAM available for dynamic allocation:
I (1125) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1130) heap_init: At 3FFB6560 len 00029AA0 (166 KiB): DRAM
I (1135) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1141) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1146) heap_init: At 40099B2C len 000064D4 (25 KiB): IRAM
I (1152) spiram: Adding pool of 4013K of external SPI memory to heap allocator
I (1159) spi_flash: detected chip: generic
I (1162) spi_flash: flash io: qio
I (1166) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1178) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1440) wifi:wifi driver task: 3ffd1374, prio:23, stack:6656, core=0
I (1455) wifi:wifi firmware version: 0f80fa0
I (1456) wifi:wifi certification version: v7.0
I (1456) wifi:config NVS flash: enabled
I (1456) wifi:config nano formating: disabled
I (1460) wifi:Init data frame dynamic rx buffer num: 32
I (1465) wifi:Init management frame dynamic rx buffer num: 32
I (1471) wifi:Init management short buffer num: 32
I (1475) wifi:Init static tx buffer num: 16
I (1480) wifi:Init tx cache buffer num: 32
I (1483) wifi:Init static rx buffer size: 1600
I (1487) wifi:Init static rx buffer num: 10
I (1491) wifi:Init dynamic rx buffer num: 32
I (00:00:00.410) wifi_init: rx ba win: 6
[... 71 lines cut ...]
D (17:36:03.918) spi_master: Allocate TX buffer for DMA
D (17:36:03.921) spi_master: Allocate TX buffer for DMA
E (17:36:04.084) w5500.mac: no mem for receive buffer
E (17:36:04.123) w5500.mac: no mem for receive buffer
E (17:36:04.422) w5500.mac: no mem for receive buffer
E (17:36:04.716) w5500.mac: no mem for receive buffer
E (17:36:04.723) w5500.mac: no mem for receive buffer
D (17:36:05.138) spi_master: Allocate TX buffer for DMA
E (17:36:05.138) w5500.mac: w5500_write(74): spi transmit failed
E (17:36:05.138) w5500.mac: w5500_write_buffer(173): write TX buffer failed
E (17:36:05.144) w5500.mac: emac_w5500_transmit(528): write frame failed
E (17:36:06.791) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (17:36:06.792) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (17:36:06.795) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (17:36:08.791) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (17:36:08.791) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (17:36:08.795) w5500.phy: w5500_get_link(108): update link duplex speed failed

.. Repeats the following 3 lines forever.

E (17:36:10.791) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (17:36:10.791) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (17:36:10.795) w5500.phy: w5500_get_link(108): update link duplex speed failed

```

kostaond, 2023-07-11:
Here is my theory: I think the `malloc` allocates memory for TX packet in PSRAM which [is enabled by default](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#external-ram-config-malloc). When this happens, the SPI driver tries to allocate a new memory which is DMA capable (`spi_master: Allocate TX buffer for DMA`). However, all the memory was exhausted and so the SPI transaction fails. The problem probably is, the SPI driver [locked the bus](https://github.com/espressif/esp-idf/blob/release/v4.4/components/driver/spi_master.c#L965) but didn't unlock it when TX buffer allocation failed. Therefore all the following transactions fail. I'll try to discuss with colleagues responsible for SPI.

In the meantime, you can try to disable the PSRAM or to configure it to `SPIRAM_USE_CAPS_ALLOC`.

bensuffolk, 2023-07-11:
So I guess like you say the real issue is the locked SPI bus, although I don't understand why that will stop the WiFi network working.

I have no idea how the code of the LWIP network stack works, I would have expected a task per network interface, but maybe thats not the case and so no processing can happen whilst its trying to figure out the error in the SPI ethernet?

I'm not sure the memory thing in itself is fixable as I am purposefully starving the memory (I found this by mistake when I incorrectly allocated a large block of memory as __NOINIT_ATTR instead of EXT_RAM_NOINIT_ATTR), so it will always run out in this case.

But it highlighted the issue I have seen on rare occasion on V4.4.5 and very regularly on V5.0.2 that whenever there is an SPI ethernet error it just loops these error lines and locks up all networking.

What I'm looking for is a way to recover gracefully if the memory run out, or some other error happens. I'm not sure if the SPI bus being unlocked would let everything carry on again (if the cause of the error were resolved e.g. I freed up some memory, or whatever) or if I would need to restart the ethenert.

kostaond, 2023-07-12:
Do you free the memory? WiFi driver may also try to allocate TX buffer in DMA capable memory area, see [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#using-psram) for more information.

bensuffolk, 2023-07-12:
I have the following settings: 

```
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP
CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST
```

Which I think tries to get the majority of the WiFi memory using PSRAM. 

I have tried to esp_retsart() once the error is present, but that can't restart the ESP. I guess its blocking waiting for the network to close as part of the shutdown procedure. I think once there is an SPI error nothing can happen in the network stack at all.

bensuffolk, 2023-07-27:
> I'll try to discuss with colleagues responsible for SPI.

Any joy with getting them to look at this?

wanckl, 2023-07-28:
@bensuffolk Hello, sorry for replay late
@kostaond 's analyze is very understandable, I have a patch here based on this, could you have a try and check if issue is fixed, thank you very much.
[0001-fix-spi_master-polling_start-not-free-buslock-when-N.patch.txt](https://github.com/espressif/esp-idf/files/12189984/0001-fix-spi_master-polling_start-not-free-buslock-when-N.patch.txt)  (rename to .patch to use)

I'll very glade if you can replay if the patch is good, best wishes to you.

onetabsen, 2023-08-01:
Good evening!

Faced a similar problem. Configured Ethernet W5500 via SPI. The only solution was to disable the external PSRAM.

Attached code below:

```
bool setupW5500() {
  WiFi.begin();
  tcpip_adapter_set_default_eth_handlers();

  // Initialize TCP/IP network interface (should be called only once in application)
  ESP_ERROR_CHECK(esp_netif_init());
  esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
  esp_netif_t *eth_netif = esp_netif_new(&cfg);
  
  // Set default handlers to process TCP/IP stuffs
  ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));
  WiFi.disconnect();

  esp_eth_mac_t *eth_mac = NULL;
  esp_eth_phy_t *eth_phy = NULL;

  gpio_install_isr_service(0);

  spi_bus_config_t buscfg = {
    .mosi_io_num = 1,                      // Default 23 (16 mb == 1)
    .miso_io_num = 40,                      // Default 19 (16 mb == 40)
    .sclk_io_num = 39,                      // Default 18 (16 mb == 39)
    .quadwp_io_num = -1,
    .quadhd_io_num = -1,
  };
  ESP_ERROR_CHECK(spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO));

  spi_device_handle_t spi_handle = NULL;
  spi_device_interface_config_t devcfg = {
    .command_bits = 16, // Actually it's the address phase in W5500 SPI frame
    .address_bits = 8,  // Actually it's the control phase in W5500 SPI frame
    .mode = 0,
    .clock_speed_hz = 12 * 1000 * 1000,
    .spics_io_num = 37,                      // Default 18 (16 mb == 37)
    .queue_size = 20
  };
  ESP_ERROR_CHECK(spi_bus_add_device(SPI2_HOST, &devcfg, &spi_handle));
  /* w5500 ethernet driver is based on spi driver */
  eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
  w5500_config.int_gpio_num = 8;                      // Default 18 (16 mb == 8)

  eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
  eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
  phy_config.reset_gpio_num = 18;                      // Default 26 (16 mb == 18)

  eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
  if (eth_mac == NULL) {
    log_e("esp_eth_mac_new_esp32 failed");
    return false;
  }

  eth_phy = esp_eth_phy_new_w5500(&phy_config);
  if (eth_phy == NULL) {
    log_e("esp_eth_phy_new failed");
    return false;
  }

  esp_eth_config_t eth_config = ETH_DEFAULT_CONFIG(eth_mac, eth_phy);
  esp_eth_handle_t eth_handle = NULL;
  ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config, &eth_handle));

/* =================== test Dynamic Mac set (START) ===================*/

  uint8_t macArr[6] = {0x00, 0x08, 0xDC, 0x00, 0x00, 0x00};
  uint8_t empy_adr_count = 1;

  
  for(int i = 0; i < 3; i++){
    macArr[i+3] = EEPROM.read(192 + i);
    if(macArr[i+3] != 0 && macArr[i+3] != 255){
      empy_adr_count--;
    }
  }

//  DEBUG
//
//  Serial.print("\nMy MAC: ");
//for(int i = 0; i < 6; i++){
//  Serial.print(macArr[i]);
//  if(i != 5){Serial.print(".");
//  }else{Serial.println("\n");}
//}
  
/* =================== test Dynamic Mac set (END) ===================*/

  //Static MAC
  //uint8_t macArr[] = {0x02, 0x00, 0x00, 0x12, 0x34, 0x57};

  ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle, ETH_CMD_S_MAC_ADDR, macArr));

  /* attach Ethernet driver to TCP/IP stack */
  ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
  /* start Ethernet driver state machine */
  ESP_ERROR_CHECK(esp_eth_start(eth_handle));

  return true;
}
```

bensuffolk, 2023-08-01:
> @bensuffolk Hello, sorry for replay late @kostaond 's analyze is very understandable, I have a patch here based on this, could you have a try and check if issue is fixed, thank you very much. [...]

That patch file seems to be against idf 5.x but I'm using 4.4.5 so it won't apply.

wanckl, 2023-08-02:
@bensuffolk Sorry I ignored your version, here the 4.4 version of patch~
[0001-fix-spi_master-polling_start-not-free-buslock_v4.4.patch.txt](https://github.com/espressif/esp-idf/files/12235565/0001-fix-spi_master-polling_start-not-free-buslock_v4.4.patch.txt)

wanckl, 2023-08-14:
@bensuffolk Hi, sorry to bother you again, have you tried the patch work well with your issue?

bensuffolk, 2023-08-15:
Apologies for the delay, I have been snowed under with my other work. 

I am very pleased to say this patch has worked, the error no longer loops when I disconnect the remote connection. WiFi responds a expected.

Many thanks.

wanckl, 2023-08-21:
@bensuffolk :smiley: glade to hear that, thank you very much for your worthy report and test, I'll continuing doing this fix for other release versions, and very happy to invite you update to the latest version~

closed 2023-09-04 by espressif-bot

kostaond, 2023-09-21:
> @bensuffolk Sorry I ignored your version, here the 4.4 version of patch~
> [0001-fix-spi_master-polling_start-not-free-buslock_v4.4.patch.txt](https://github.com/espressif/esp-idf/files/12235565/0001-fix-spi_master-polling_start-not-free-buslock_v4.4.patch.txt)

Just note that the actual final fix is a little bit different, see https://github.com/espressif/esp-idf/commit/f7068bd937e82aefc8fdd1c85393915273b5e5ad