espressif/esp-idf #11846: map psram as iram

PickaxeHit, 2023-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Is there a way that the psram can be mapped as iram so that i can execute code from psram?
I have tried
```
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_FETCH_INSTRUCTIONS=y
CONFIG_SPIRAM_RODATA=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY=y
# CONFIG_ESP_SYSTEM_MEMPROT_FEATURE is not set
```
and app_main
```
heap_caps_print_heap_info(MALLOC_CAP_EXEC);
```

but it shows that the total heap size is 325992 bytes.
```
Heap summary for capabilities 0x00000001:
  At 0x3fc9a2f0 len 324640 free 304512 allocated 18188 min_free 304512
    largest_free_block 303104 alloc_blocks 42 free_blocks 2 total_blocks 44
  At 0x3fce9710 len 22308 free 21480 allocated 72 min_free 21480
    largest_free_block 20480 alloc_blocks 5 free_blocks 1 total_blocks 6
  Totals:
    free 325992 allocated 18260 min_free 325992 largest_free_block 303104
```internal
It seems that the psram is mapped to dram, and the iram is all from the internal sram.

igrr, 2023-07-10:
If you are using ESP32-S2 or ESP32-S3, please check https://github.com/espressif/esp-idf/tree/master/examples/system/xip_from_psram.

On ESP32, this is not possible.

PickaxeHit, 2023-07-10:
> If you are using ESP32-S2 or ESP32-S3, please check https://github.com/espressif/esp-idf/tree/master/examples/system/xip_from_psram.
> 
> On ESP32, this is not possible.

thanks for your help.
But the example seems that it "compiled" into psram, what i really want is to allocate memory dynamically (for example:ELF loader). Do i need to init psram manually?
ps: I'm using esp32s3.
[image]
how could i understand this? Can i think that psram can be mapped to iram for dynamic allocation of memory?

igrr, 2023-07-10:
I see, thanks for explaining what you are looking for.

This use case is not well supported out of the box, as IDF maps all of the PSRAM to heap for data access only. Consequently calling heap_caps_malloc with a combination of MALLOC_CAP_SPIRAM and MALLOC_CAP_EXEC doesn't work.

There are two ways to implement this:
1. Initialize PSRAM yourself, then implement a modified version of `esp_psram_extram_add_to_heap_allocator`. The modified version would map part of PSRAM on the data bus, part on the instruction bus. This will unfortunately require calling a bunch of private APIs, please see the headers of esp_psram component.
2. Let IDF map all of the PSRAM as data, then use esp_mm component APIs to additionally map some part of PSRAM to the instruction bus. Here is a piece of code to illustrate this idea: https://gist.github.com/igrr/ef5a3ad9f5fbf835f06c88b6b36defcc

Please let me know if you have success with either of these options. Based on your feedback we'd be open to adding such an example to IDF.

PickaxeHit, 2023-07-10:
Thanks for your guidance, I will try it later.

igrr, 2023-07-12:
These two buffers correspond to the same physical memory, just seen from two buses: data bus and instruction bus.

Yes, you could copy the code to instruction bus directly, just make sure to only use 32-bit store instructions in this case, since Xtensa instruction bus only supports 32-bit loads and stores. In that snippet I chose to copy the code over the data bus to sidestep this issue.

PickaxeHit, 2023-07-13:
When I copy the code to `exec_buf`(Use another memory copy function to ensure 4-byte memory alignment, at [here](https://gist.github.com/PickaxeHit/b86ff1c98b13cef400ea25512dafad9a))
`unaligned_copy(exec_buf, &test_func, code_size);`
the idf throw an exception:
```
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Setting breakpoint at 0x00000000 and returning...
```
but when it is `unaligned_copy(heap_buf, &test_func, code_size);`, it works.
What is causing this problem? and when the idf will throw this exception?
(During this period, I did not perform any operations on flash read and write.)

PickaxeHit, 2023-07-30:
This has a new question, I am very confused, why can it be executed directly even without remapping through the mmu?
A simple example of main.c
```c
#include <stdio.h>
#include <esp_heap_caps.h>
#include <esp_mmu_map.h>
#include <string.h>
#include <esp_cache.h>

int test(void)
{
    return 42;
}

static inline void *data_vaddr_to_instr_vaddr(void *vaddr)
{
    esp_paddr_t paddr = 0;
    mmu_target_t target;
    void *out_vaddr = NULL;
    esp_mmu_vaddr_to_paddr(vaddr, &paddr, &target);
    esp_mmu_paddr_to_vaddr(paddr, target, MMU_VADDR_INSTRUCTION, &out_vaddr);
    return out_vaddr;
}

void app_main(void)
{
    void*heap_buf=heap_caps_malloc(0x1000,MALLOC_CAP_SPIRAM);
    memcpy(heap_buf,&test,32);
    esp_cache_msync(heap_buf,32,0);
    void*exec_buf=data_vaddr_to_instr_vaddr(heap_buf);
    int result = ((int (*)(void))exec_buf)();
    printf("result:%d\n",result);
}
```
and the minimum sdkconfig:
```
CONFIG_ESPTOOLPY_FLASHSIZE_8MB=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
# CONFIG_ESP_SYSTEM_MEMPROT_FEATURE is not set
```
Is this a bug or a feature?

igrr, 2023-07-30:
On ESP32-S3, what you did happens to work because MMU simultaneously maps physical pages to both instruction and data regions. But in general, the code I posted is necessary to also work on ESP32-S2.

closed 2023-07-30 by PickaxeHit