espressif/esp-idf #11849: ESP32 Kaluga CAM sending video to webserver (very low framerate)

krupis, 2023-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I have been looking at Arduino Framework based ESP32 cam project that involves sending video to webserver:
https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/

My goal is to create something similar for esp-idf.

I have ESP32-S2 Kaluga development board and to get things going I will be using that since it comes together with a small ESP-LyraP-CAM that you can connect to the board.

**Program:**

- For the camera initialization, I used Kaluga official example from the component registry.

- For the webserver, I have copied some code from the example project below:
  https://esp32tutorials.com/esp32-cam-esp-idf-live-streaming-web-server/?unapproved=6521&moderation-hash=f5cc7d05f85bc47edecffbbe6bc956d5#comment-6521

The project compiles and runs. The issue is that I am only getting about 1fps. I am trying to understand what could be an issue since the project based on the Arduino is able to achieve much better results it must be something that I am missing.

You can access my project here:
https://github.com/krupis/ESP32-Bird-box

It should build and run on the ESP32-S2-Kaluga-1 Board. Once you flash the Kaluga board, you can connect to the webserver (The IP will be printed in the console):
[image]

My questions:

1. Does Kaluga development board has external RAM? From what I know, dealing with camera requires a lot of RAM and ESP32 internal RAM is not enough. I have been studying ESP32 Kaluga schematics and I see no trace of external RAM:
https://dl.espressif.com/dl/schematics/ESP32-S2-Kaluga-1_V1_3_SCH_20200526A.pdf

2. In the webserver ESP32 example project, the guy is using:
```
.pixel_format = PIXFORMAT_JPEG,
```

but I cannot get it to work. I am getting the following errors:

```
I (2347) esp32-cam-test: ESP32 CAM Web Server is up and running

W (2417) cam_hal: NO-SOI
W (2467) cam_hal: NO-SOI
W (2517) cam_hal: NO-SOI
W (2567) cam_hal: NO-SOI
W (2617) cam_hal: NO-SOI
W (2667) cam_hal: NO-SOI
W (2717) cam_hal: NO-SOI
W (2767) cam_hal: NO-SOI
W (2817) cam_hal: NO-SOI
W (2867) cam_hal: NO-SOI
W (2917) cam_hal: NO-SOI
W (2967) cam_hal: NO-SOI
```

so I am just using
```
.pixel_format = PIXFORMAT_RGB565,
```

Could that be the cause of low fps?

3. Are there any other options that I should enable in menuconfig to make the FPS better?

I appreciate any tips regarding ESP32-S3 Kaluga board and how to get the cam to work efficiently with webserver. Thanks in advance.

krupis, 2023-07-11:
Simillar issues regarding poor camera module and webserver are discussed here:
https://github.com/espressif/arduino-esp32/issues/4655

There does not seem to be a clear solution to this issue

WangYuxin-esp, 2025-04-03:
When the sensor is working in RGB565 mode, in order to obtain JPEG images, it will internally call the [fmt2jpg](https://github.com/espressif/esp32-camera/blob/master/conversions/include/img_converters.h#L72) interface to encode RGB565 data. Software coding requires a significant amount of time.  
When the sensor operates in JPEG mode, the frame rate should be increased. However, if the sensor sends data too quickly, the receiving end may not be able to receive the data in a timely manner, resulting in a `NO-SOI` error.  
As mentioned by @RealNee , reducing XCLK can lower the frame rate of the sensor. XCLK configurations of 10M and 20M are common.

Alvin1Zhang, 2025-07-07:
Thanks for reporting, will close due to inactivity, feel free to reopen with more updates. Thanks.

WangYuxin-esp, 2025-07-07:
If have new test results, please re-raise this issue in the esp32-camera repository.