espressif/esp-idf #11852: Display MPI3501 + Ili9486 + XPT2046 [lvgl 7 + esp-idf] - Display not showing, but touch works

Anderson-Chimuco, 2023-07-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi
I’m struggling to use the display MPI3501 with ESP32.
I tested with Bodmer TFT_eSPI library on Arduino IDE, and everything works.
Then I tested with an example from maker fans (Sunton 2.8 but I edit for a 3.5”)with the proper drivers from the lvgl_esp32_drivers. 

At first it worked, the display shows everything but the touch didn’t work, but I soon noticed that on the xpt2046.h the pins assignments were incorrect for my pin configuration. I change it to be the same than display SPI, and the problem started. The touch works because I can see the x y values on the monitor but the display do not shows nothing (sometimes garbage).

I’m using the same SPI Bus, both Half duplex. The display is RPI TOUCHSCREEN 3.5” MPI3501, with xpt2046 controller, using Ili9486, 20MHZ for SPI and 2.5Mhz.
 
 I opted for  the lvgl cause I want to build an interface with multiple tabs

Does anyone has an ideia of the cause of the problem? 
And how to solve ?
Thanks In advance

suda-morris, 2023-07-12:
Hi @Anderson-Chimuco you can ask it in the [esp32 forum](https://esp32.com/). 

> I change it to be the same than display SPI

So you're using the same SPI bus with a different CS? Are you sure the LCD and touch IC share the same SPI mode? Have you tried to slow down the SPI frequency for the LCD?

Alvin1Zhang, 2023-08-01:
Thanks for reporting, will close due to short of feedback. You could also create a post in our forum at https://esp32.com/. Thanks.

closed 2023-08-01 by Alvin1Zhang