espressif/esp-idf #11853: 在4.4.5分支中，深度休眠保持48脚电平时，保持不了

FxMain, 2023-07-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp32-s3-wroom-1

### Power Supply used.

External 5V

### What is the expected behavior?

      gpio_config_t campwdn_config={
	      .intr_type= GPIO_INTR_DISABLE,
	      .mode=GPIO_MODE_OUTPUT,
	      .pin_bit_mask=BIT64(48),
	      .pull_down_en=0,
	      .pull_up_en=1,
};
gpio_config(&campwdn_config);
gpio_set_level(48, 1);
gpio_hold_dis(48);
gpio_deep_sleep_hold_en(); 
esp_deep_sleep_start();

电平不会保持在高

### What is the actual behavior?

电平初始化

### Steps to reproduce.

      gpio_config_t campwdn_config={
	      .intr_type= GPIO_INTR_DISABLE,
	      .mode=GPIO_MODE_OUTPUT,
	      .pin_bit_mask=BIT64(48),
	      .pull_down_en=0,
	      .pull_up_en=1,
};
gpio_config(&campwdn_config);
gpio_set_level(48, 1);
gpio_hold_dis(48);
gpio_deep_sleep_hold_en(); 
esp_deep_sleep_start();

FxMain, 2023-07-12:
纠正下我提交的示例代码gpio_hold_dis(48);改为gpio_hold_en(48);
另外我在5.1和4.4分支测试了，都不行

10086loutianhao, 2023-07-26:
@FxMain hello, 这是由于vdd_sdio下电导致的，此时flash下电，对应的和flash有关的pad无法得到供电，hold就会失效
[image]
如图，这几个管脚在 deepsleep的时候都没法hold

closed 2024-04-23 by Sherry616