espressif/esp-idf #11857: [eth2ap example] ARP packet is incorrect (received 60 bytes, PC side is 42 bytes).

Sky-Soo-Ap, 2023-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

ESP32 received packets via ethernet is no change.
ESP32 received packets via Wi-Fi AP/Station is no change.

### What is the actual behavior?

PC side send ARP packet via ethernet is 42 bytes as below:
ff ff ff ff ff ff 08 97  98 7c 03 77 08 06 00 01  |.........|.w....|
08 00 06 04 00 01 08 97  98 7c 03 77 c0 a8 01 5b  |.........|.w...[|
00 00 00 00 00 00 c0 a8  01 01 00 00 00 00 00 00  |..........|

ESP32 side received ARP packet via ethernet is 60 bytes (it seems more 18 bytes 0x00 at back) as below:
ff ff ff ff ff ff 08 97  98 7c 03 77 08 06 00 01  |.........|.w....|
08 00 06 04 00 01 08 97  98 7c 03 77 c0 a8 01 5b  |.........|.w...[|
00 00 00 00 00 00 c0 a8  01 01 00 00 00 00 00 00  |................|
00 00 00 00 00 00 00 00  00 00 00 00              |............|

### Steps to reproduce.

1. Ethernet cable connect ESP32 and a PC;
2. Run eth2ap example;

xueyunfei998, 2023-08-09:
[image]

hi  @Sky-Soo-Ap 

ESP32 side received ARP packet via ethernet is 60 bytes (it seems more 18 bytes 0x00 at back) as below:

Does it mean that the arp data taken from the flow_control_queue is 18 bytes more?

Sky-Soo-Ap, 2023-08-09:
Hi [xueyunfei998](https://github.com/xueyunfei998),
I remembered that is forward packet from Wi-Fi to Ethernet;
ESP32 Wi-Fi station mode and a PC (Using Wi-Fi station mode) connect to same AP;
PC received ARP packet is 42 bytes;
But, ESP32 received ARP packet is 60 bytes as below function;
You can try it on v5.2-dev-1709-g4fc2e5cb95 and example “sta2eth”.
[image]

xueyunfei998, 2023-08-11:
> Hi [xueyunfei998](https://github.com/xueyunfei998), I remembered that is forward packet from Wi-Fi to Ethernet; ESP32 Wi-Fi station mode and a PC (Using Wi-Fi station mode) connect to same AP; PC [...]

hi  @Sky-Soo-Ap 

You mean that the arp received by wifi is 42 bytes, but it is sent to the pc through this interface  `pkt_wifi2eth `, and the bytes received by the pc are changed from 42 to 60 bytes?

Sky-Soo-Ap, 2023-08-14:
Hi [xueyunfei998](https://github.com/xueyunfei998), 
I remembered that the callback of Wi-Fi RX has already received 60 bytes ARP.

xueyunfei998, 2023-08-16:
hi @Sky-Soo-Ap 
The llc layer header of wifi has 12 bytes, which will be removed when it passes through this interface `pkt_wifi2eth`  .