espressif/esp-idf #11858: ESP32C6 light sleep current is high @ 330uA

arif-mansoori, 2023-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have made a bare-board esp32c6 pcb with no other components except the mandatory ones for strapping and power supply. I make the cpu go to light sleep. However the current drawn is 330uA, much larger than the stated one in the datasheet (35uA). Are there any specific actions before putting to light sleep?

igrr, 2023-07-13:
@arif-mansoori Could you please add, which ESP-IDF version you are using (you can check it using `git describe`) and also attach your `sdkconfig` file?

arif-mansoori, 2023-07-13:
Thanks Ivan.

esp-idf version v5.2-dev-1128-g03d4fa2869

sdkconfig is attached...
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/12036939/sdkconfig.txt)

esp-lis, 2023-07-13:
@arif-mansoori I recommend you to use the **$IDF_PATH/examples/wifi/power_save** example for testing, and make sure the following options are properly configured.

```
# Serial flasher config
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y

# Hardware Settings
CONFIG_ESP_SLEEP_GPIO_RESET_WORKAROUND=y
CONFIG_RTC_CLK_SRC_INT_RC32K=y

# Power Management
CONFIG_PM_ENABLE=y
CONFIG_PM_DFS_INIT_AUTO=y
CONFIG_PM_SLP_IRAM_OPT=y
CONFIG_PM_RTOS_IDLE_OPT=y
CONFIG_PM_SLP_DISABLE_GPIO=y
CONFIG_PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP=y
CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y

# FreeRTOS
CONFIG_FREERTOS_HZ=1000
CONFIG_FREERTOS_USE_TICKLESS_IDLE=y
CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP=3

# PHY
CONFIG_ESP_PHY_MAC_BB_PD=y

# Wi-Fi
CONFIG_ESP_WIFI_SLP_IRAM_OPT=y
CONFIG_ESP_WIFI_SLP_DEFAULT_MIN_ACTIVE_TIME=10
CONFIG_ESP_WIFI_SLP_DEFAULT_MAX_ACTIVE_TIME=60
CONFIG_ESP_WIFI_ENHANCED_LIGHT_SLEEP=y
CONFIG_ESP_WIFI_SLP_BEACON_LOST_OPT=y
CONFIG_ESP_WIFI_SLP_BEACON_LOST_TIMEOUT=10
CONFIG_ESP_WIFI_SLP_BEACON_LOST_THRESHOLD=3

# LWIP
# CONFIG_LWIP_ESP_GRATUITOUS_ARP is not set

# MBEDTLS
# remove after corresponding crypto module init/deinit supported
# CONFIG_MBEDTLS_HARDWARE_AES is not set
# CONFIG_MBEDTLS_HARDWARE_MPI is not set
# CONFIG_MBEDTLS_HARDWARE_SHA is not set
# CONFIG_MBEDTLS_HARDWARE_ECC is not set
```

If you are using a different example, please provide more detailed information.

arif-mansoori, 2023-07-13:
@esp-lis Thanks. I tried the config as you mentioned, but there is no change. All that my firmware is doing right now is 

while (true)
    {
        time(&now);
        boot_count++;
        ESP_LOGI(TAG, "boot count %d", (int)boot_count);
        prev_time = now;
#ifndef CONFIG_LOG_DEFAULT_LEVEL_NONE
        uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
#endif
	esp_sleep_enable_timer_wakeup(20000 * 1000);
	esp_light_sleep_start();
    }

arif-mansoori, 2023-07-13:
Deep sleep does bring down to 6.2uA

arif-mansoori, 2023-07-17:
Hello...any pointers in this direction? If the 35uA current during light sleep was validated, can I get the example firmware?

esp-lis, 2023-07-18:
> Hello...any pointers in this direction? If the 35uA current during light sleep was validated, can I get the example firmware?

Please add this function `esp_sleep_cpu_retention_init()` call.

arif-mansoori, 2023-07-18:
Thanks. Unfortunately there is no change. Current still remains same.

esp-lis, 2023-07-18:
> > Hello...any pointers in this direction? If the 35uA current during light sleep was validated, can I get the example firmware?
> 
> You are using the normal light sleep mode. You need to call `esp_sleep_cpu_retention_init()` to initialize the CPU retention in the **app_main()** function.

perhaps you can try return true in [here](https://github.com/espressif/esp-idf/issues/11858#issuecomment-1639369161)

arif-mansoori, 2023-07-18:
Sorry,  I did not understand. Where do I put the return?

esp-lis, 2023-07-18:
> @arif-mansoori I recommend you to use the **$IDF_PATH/examples/wifi/power_save** example for testing, and make sure the following options are properly configured.
>
> ```
> # Serial flasher config [...]

@arif-mansoori The power management strategy of light sleep without the Modem module (WiFi, BLE or 15.4) is currently under development and support. It is recommended to use the power save example for testing.

arif-mansoori, 2023-07-19:
> > Sorry, I did not understand. Where do I put the return?
> 
> > > > Hello...any pointers in this direction? If the 35uA current during light sleep was validated, can I get the example firmware?
> > > 
> > > 
> > > You are using the normal light sleep mode. You need to call `esp_sleep_cpu_retention_init()` to initialize the CPU retention in the **app_main()** function.
> > 
> > 
> > perhaps you can try return true in [here](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_modem.c#L288)
> 
> I've modified it.

Thanks! This "return" along with esp_sleep_cpu_retention_init() now gives me a light sleep current of 26uA. It now works!

arif-mansoori, 2023-07-19:
> @arif-mansoori The power management strategy of light sleep without the Modem module (WiFi, BLE or 15.4) is currently under development and support. It is recommended to use the power save example [...]

Do you have an ETA of when the power management strategy of light sleep without the Modem module will be released?

esp-lis, 2023-07-27:
> Do you have an ETA of when the power management strategy of light sleep without the Modem module will be released?

@arif-mansoori The issue will be resolved in the near future, and I will inform you here. Thank you!

AxelLin, 2023-08-23:
> @arif-mansoori The issue will be resolved in the near future, and I will inform you here. Thank you!

Will v5.1.1 include this fix?

AxelLin, 2023-09-14:
> > Do you have an ETA of when the power management strategy of light sleep without the Modem module will be released?
> 
> @arif-mansoori The issue will be resolved in the near future, and I will inform you here. Thank you!

@esp-lis  Any update about this?

AxelLin, 2023-09-25:
> > > Hello...any pointers in this direction? If the 35uA current during light sleep was validated, can I get the example firmware?
> > 
> > 
> > You are using the normal light sleep mode. You need to call `esp_sleep_cpu_retention_init()` to initialize the CPU retention in the **app_main()** function.

@esp-lis 
1. About cpu retentin usage
   When to call esp_sleep_cpu_retention_init() and esp_sleep_cpu_retention_deinit()?
   Is it correct to call esp_sleep_cpu_retention_init() before esp_light_sleep_start()
   and call esp_sleep_cpu_retention_deinit() after esp_light_sleep_start()?

2. I thought setting CONFIG_PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP=y is enough to reduce the power consumption.
   No idea why need to call esp_sleep_cpu_retention_init() for reducing power consumption,
   where can I find the document about this?
   Is it only necessary for C6?

> 
> perhaps you can try return true in [here](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_modem.c#L288)

   Seems only c6 defines SOC_PM_MODEM_RETENTION_BY_REGDMA=1.
   So this part is c6-only bug?

AxelLin, 2023-10-02:
> > > Do you have an ETA of when the power management strategy of light sleep without the Modem module will be released?
> > 
> > 
> > @arif-mansoori The issue will be resolved in the near future, and I will inform you here. Thank you!
> 
> @esp-lis Any update about this?

@esp-lis 
We are evaluating the hw for our new project.
However, without cpu retention for esp32c6, the esp32c6 takes more power than esp32c3 in light sleep.
You said this issue will be resolved in the near future 2 Months ago, but this is still not fixed yet.
Any update?

esp-lis, 2023-10-12:
> @esp-lis We are evaluating the hw for our new project. However, without cpu retention for esp32c6, the esp32c6 takes more power than esp32c3 in light sleep. You said this issue will be resolved in [...]

The ESP32-C6 CPU retention feature has been supported since the initial release. It can be enabled or disabled in auto light sleep mode through the **PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP** option (It is located at (Top) -> Component config -> Power Management -> **Power down CPU in light sleep**). For normal light sleep mode, you need to call [esp_sleep_cpu_retention_init](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_cpu.c#L743) during the application initialization stage to initialize the CPU retention functionality.

The current issue record is as follows:
For WiFi, BLE, and IEEE802.15.4, these three modules share the same power domain, known as the MODEM domain. To control the power state of this MODEM domain for these three modules (powering up or down during sleep), software management is required. We have already added the support and are preparing to merge it.

AxelLin, 2023-10-12:
> The ESP32-C6 CPU retention feature has been supported since the initial release. It can be enabled or disabled in auto light sleep mode through the **PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP** option (It is [...]
.
@esp-lis 
Could you clarify if calling esp_sleep_cpu_retention_init() is also requrired for **esp32c3** for or normal light sleep mode?
I don't find docuement mention this, I thought setting PM_POWER_DOWN_CPU_IN_LIGHT_SLEEP=y is enough.

AxelLin, 2023-10-23:
> perhaps you can try return true in [here](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_modem.c#L288)

If I made above change to return true in modem_domain_pd_allowed,
I got below issue after awake from lightsleep and join an AP:
```
E (23951) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (23951) task_wdt:  - IDLE (CPU 0)
E (23951) task_wdt: Tasks currently running:
E (23951) task_wdt: CPU 0: wifi
E (23951) task_wdt: Print CPU 0 (current core) registers
Stack dump detected
Core 00 register dump:
MEPC    : 0x40807440  RA      : 0x40806b5e  SP      : 0x40841a40  GP      : 0x408186c4
0x40807440: wait_freq_set_busy at ??:?

0x40806b5e: phy_wakeup_init_ at ??:?

TP      : 0x4080351c  T0      : 0x00000004  T1      : 0x4080d20a  T2      : 0x00000005
0x4080351c: npl_freertos_eventq_remove at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:352 (discriminator 2)

0x4080d20a: vPortSetupTimer at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/port_systick.c:81 (discriminator 1)

S0/FP   : 0x40819f50  S1      : 0x40820a74  A0      : 0x00000001  A1      : 0x00000000
A2      : 0x600af81c  A3      : 0xff7fffff  A4      : 0x600a0000  A5      : 0x00000000
A6      : 0xfe03ffff  A7      : 0x4081b42c  S2      : 0x40819f50  S3      : 0x00000000
S4      : 0x0001e7ff  S5      : 0x40880000  S6      : 0x40880000  S7      : 0x40880000
S8      : 0x40880000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000003  T4      : 0x00000009  T5      : 0x00000067  T6      : 0x0000000b
MSTATUS : 0x00000300  MTVEC   : 0x28010200  MCAUSE  : 0x00000000  MTVAL   : 0x4081de00
MHARTID : 0x40821000
E (28951) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (28951) task_wdt:  - IDLE (CPU 0)
E (28951) task_wdt: Tasks currently running:
E (28951) task_wdt: CPU 0: wifi
E (28951) task_wdt: Print CPU 0 (current core) registers
Core 00 register dump:
MEPC    : 0x40807440  RA      : 0x40806b5e  SP      : 0x40841a40  GP      : 0x408186c4
0x40807440: wait_freq_set_busy at ??:?

0x40806b5e: phy_wakeup_init_ at ??:?
```

esp-lis, 2023-10-23:
> If I made above change to return true in modem_domain_pd_allowed, I got below issue after awake from lightsleep and join an AP:
>
> ```
> E (23951) task_wdt: Task watchdog got triggered. The following [...]

The log indicates that the system hangs when the WiFi task enables RF. Does this issue occur 100% of the time? If it does, could you try disabling light sleep and test with modem sleep mode to see if the problem persists?

If it's not happening 100% of the time, please provide detailed software versions and configurations, and I will attempt to reproduce the issue locally. Thank you!

AxelLin, 2023-10-23:
@esp-lis 
I test on v5.1 branch.
Below code snip can reproduce the issue after device wake up from lightsleep and start wifi again.
**This issue only happen if you change modem_domain_pd_allowed() to return true.**

```
while (1) {
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        ESP_ERROR_CHECK(esp_wifi_init(&cfg)); 
        esp_wifi_start();
        vTaskDelay(pdMS_TO_TICKS(3000));
        esp_wifi_stop();
        esp_wifi_deinit();
        esp_sleep_enable_timer_wakeup(10000000);
        esp_light_sleep_start();
}
```
The log:
```
I (10619) LightSleep: Wakeup by timer
I (10649) pp: pp rom version: 5b8dcfa
I (10649) net80211: net80211 rom version: 5b8dcfa
I (10659) wifi:wifi driver task: 40834c48, prio:23, stack:6656, core=0
I (10659) wifi:wifi firmware version: dc6ffb6
I (10659) wifi:wifi certification version: v7.0
I (10659) wifi:config NVS flash: enabled
I (10659) wifi:config nano formating: disabled
I (10669) wifi:mac_version:HAL_MAC_ESP32AX_761,ut_version:N
I (10669) wifi:Init data frame dynamic rx buffer num: 32
I (10679) wifi:Init static rx mgmt buffer num: 5
I (10679) wifi:Init management short buffer num: 32
I (10689) wifi:Init dynamic tx buffer num: 32
I (10689) wifi:Init static tx FG buffer num: 2
I (10689) wifi:Init static rx buffer size: 1700
I (10699) wifi:Init static rx buffer num: 10
I (10699) wifi:Init dynamic rx buffer num: 32
I (10709) wifi_init: rx ba win: 6
I (10709) wifi_init: tcpip mbox: 32
I (10709) wifi_init: udp mbox: 6
I (10719) wifi_init: tcp mbox: 6
I (10719) wifi_init: tcp tx win: 5744
I (10729) wifi_init: tcp rx win: 5744
I (10729) wifi_init: tcp mss: 1440
I (10739) wifi_init: WiFi IRAM OP enabled
I (10739) wifi_init: WiFi RX IRAM OP enabled
E (15649) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (15649) task_wdt:  - IDLE (CPU 0)
E (15649) task_wdt: Tasks currently running:
E (15649) task_wdt: CPU 0: wifi
E (15649) task_wdt: Print CPU 0 (current core) registers
Stack dump detected
Core �8 register dump:
MEPC    : 0x40805f7e  RA      : 0x408056a0  SP      : 0x40834b50  GP      : 0x408170e4
0x40805f7e: wait_freq_set_busy at ??:?

0x408056a0: phy_wakeup_init_ at ??:?

TP      : 0x407fe824  T0      : 0x00000004  T1      : 0x4080b944  T2      : 0x00000005
0x4080b944: uxInitialiseStackTLS at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:164
 (inlined by) pxPortInitialiseStack at /home/axel/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:284

S0/FP   : 0x40818848  S1      : 0x4081f18c  A0      : 0x00000001  A1      : 0x00000000
A2      : 0x600af81c  A3      : 0xff7fffff  A4      : 0x600a0000  A5      : 0x00000000
A6      : 0xfe03ffff  A7      : 0x40819bd4  S2      : 0x40818848  S3      : 0x00000000
S4      : 0x0001e7ff  S5      : 0x40880000  S6      : 0x40880000  S7      : 0x40880000
S8      : 0x40880000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x00000003  T4      : 0x00000009  T5      : 0x00000067  T6      : 0x0000000b
MSTATUS : 0x00000300  MTVEC   : 0x28010200  MCAUSE  : 0x00000000  MTVAL   : 0x4081c518
MHARTID : 0x4081f000

Done
```

esp-lis, 2023-10-23:
> @esp-lis I test on v5.1 branch. Below code snip can reproduce the issue after device wake up from lightsleep and start wifi again. **This issue only happen if you change modem_domain_pd_allowed() to [...]

I tried to reproduce it locally using [release/v5.1](https://github.com/espressif/esp-idf/tree/release/v5.1) (c1c843f5), but it seems to be working fine. 

Below is the attachment of the sdkconfig for your reference. Thank you.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13070249/sdkconfig.txt)

esp-lis, 2023-10-24:
> Hi @esp-lis
> 
> After compare with your sdkconfig, I found this issue does not happen if CONFIG_ESP_PHY_MAC_BB_PD=y. You can set CONFIG_ESP_PHY_MAC_BB_PD=n to reproduce the issue.

Enabling the **CONFIG_ESP_PHY_MAC_BB_PD** option initializes WiFi MAC + BB's sleep retention function. esp_wifi_stop(), esp_wifi_deinit() in the example code is followed by a light sleep. I think it can still work normally without WiFi MAC + BB retention. I will report this issue to relevant developer, thank you.

AxelLin, 2023-11-05:
> > Do you have an ETA of when the power management strategy of light sleep without the Modem module will be released?
> 
> @arif-mansoori The issue will be resolved in the near future, and I will inform you here. Thank you!

@esp-lis 
It has been in reviewing state for more than 3 Months.
I didn't expect such long time to get the fix since you mentioned that this will be resolved in the near future on Jul 27.
Will this get fixed in v5.1 branch?

AxelLin, 2023-11-06:
> > Hello...any pointers in this direction? If the 35uA current during light sleep was validated, can I get the example firmware?
> 
> You are using the normal light sleep mode. You need to call `esp_sleep_cpu_retention_init()` to initialize the CPU retention in the **app_main()** function.

@esp-lis 
The sleep_enable_cpu_retention() is guarded by #if SOC_PM_SUPPORT_CPU_PD && SOC_PM_CPU_RETENTION_BY_RTCCNTL .
https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/include/esp_private/sleep_cpu.h#L43-L61

**ESP32C6 does not set SOC_PM_CPU_RETENTION_BY_RTCCNTL**, so I'm wondering if cpu retention really work for c6,
or I just misunderstand the purpose of this function?

AxelLin, 2023-11-07:
> perhaps you can try return true in [here](https://github.com/espressif/esp-idf/blob/master/components/esp_hw_support/sleep_modem.c#L288)

@esp-lis 
I notice another issue with above change.

With below configuration:
PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y

Use examples/system/light_sleep to reproduce the issue.
The modified code as below:
1. Use gpio 4 insteat of BOOT_BUTTON_NUM for wakup gpio (any rtc_gpio_is_valid_gpio() can trigger the issue.
2. Call esp_sleep_cpu_retention_init()
3. comment out timer/uart wakup, make it simple, just test gpio wakeup

I attached my modified code and sdkconfig below:
[light_sleep_test.diff.log](https://github.com/espressif/esp-idf/files/13274851/light_sleep_test.diff.log)
[sdkconfig.log](https://github.com/espressif/esp-idf/files/13274852/sdkconfig.log)

The symptom. (I add debug print to show wakeup_cause, it's always 0)
Returned from light sleep, reason: other, t=26040 ms, slept for 2 ms
E (24081) sleep: wakeup_cause=0
Entering light sleep
E (24097) sleep: wakeup_cause=0
Returned from light sleep, reason: other, t=26057 ms, slept for 2 ms
E (24097) sleep: wakeup_cause=0
Entering light sleep
E (24114) sleep: wakeup_cause=0
Returned from light sleep, reason: other, t=26075 ms, slept for 2 ms

AxelLin, 2023-12-10:
> Enabling the **CONFIG_ESP_PHY_MAC_BB_PD** option initializes WiFi MAC + BB's sleep retention function. esp_wifi_stop(), esp_wifi_deinit() in the example code is followed by a light sleep. I think it [...]

@esp-lis So how is the status for above fix? Should I create a new issue to trace this issue?

esp-lis, 2023-12-11:
> @esp-lis I test on v5.1 branch. Below code snip can reproduce the issue after device wake up from lightsleep and start wifi again. **This issue only happen if you change modem_domain_pd_allowed() to [...]

@AxelLin this issue has been fixed in [581824d1](https://github.com/espressif/esp-idf/commit/581824d1f21441d0c287c452d8c936bf2a8511e0)

AxelLin, 2024-02-22:
> > Do you have an ETA of when the power management strategy of light sleep without the Modem module will be released?
> 
> @arif-mansoori The issue will be resolved in the near future, and I will inform you here. Thank you!

@esp-lis 
How is the status now? Already wait for the fix for half-year.

esp-lis, 2024-04-02:
@AxelLin This feature has been merged into the latest master, pls test with the latest master branch/

AxelLin, 2024-04-02:
> @AxelLin This feature has been merged into the latest master, pls test with the latest master branch.

@esp-lis 
The master is imply broken to me.
I cannot test master now due to panic with https://github.com/espressif/esp-idf/issues/13288.

AxelLin, 2024-04-02:
> @AxelLin This feature has been merged into the latest master, pls test with the latest master branch.

BTW, do you mean now it's ok to remove the hack in https://github.com/espressif/esp-idf/issues/11858#issuecomment-1641568390 ?

esp-lis, 2024-04-02:
> > @AxelLin This feature has been merged into the latest master, pls test with the latest master branch.
> 
> BTW, do you mean now it's ok to remove the hack in [#11858 (comment)](https://github.com/espressif/esp-idf/issues/11858#issuecomment-1641568390) ?

I mean https://github.com/espressif/esp-idf/issues/11858#issuecomment-1640244734

AxelLin, 2024-04-03:
> > > @AxelLin This feature has been merged into the latest master, pls test with the latest master branch.
> > 
> > 
> > BTW, do you mean now it's ok to remove the hack in [#11858 (comment)](https://github.com/espressif/esp-idf/issues/11858#issuecomment-1641568390) ?
> 
> I mean [#11858 (comment)](https://github.com/espressif/esp-idf/issues/11858#issuecomment-1640244734)

Could you share which commit fixed this issue?
(I'd like to remove my local hack when the fix is in release branches).

esp-lis, 2024-04-07:
> Could you share which commit fixed this issue? (I'd like to remove my local hack when the fix is in release branches).

@AxelLin pls refer to [0c68b9b0](https://github.com/espressif/esp-idf/commit/0c68b9b06c7df83ba3edc1873cc3506e79cbc769).

closed 2024-08-16 by Sherry616