espressif/esp-idf #11859: Only update user application during OTA

rbruneau-altyor, 2023-07-13:
### Is your feature request related to a problem?

After coming across quite long OTA updates through BLE or custom protocols, I was wondering if there is any way in ESP-IDF to perform OTA update only on user application, meaning that when using BLE and Wi-Fi, we would not have to update the radio stack beside the application itself. This way, the binary image to download would be much smaller, making the process way faster.

As far as I known, partition table and ota partitions do not work this way, but if there is any way to keep the radio stack from one OTA to another, that would be great.

### Describe the solution you'd like.

May be a new partition table system that would allow to keep BLE/Wi-Fi stack apart from user application

igrr, 2023-07-13:
This is not possible out of the box in ESP-IDF, since the application is a statically linked binary. It is the combination of code and static data (.rodata, .data, .bss) from all the libraries which the application is built from, including the RTOS, drivers, networking and wireless stacks.

There are two alternatives you may consider:

* [esp-privilege-separation](https://github.com/espressif/esp-privilege-separation) project builds on top of ESP-IDF and allows splitting the monolithic application binary into two parts. At the moment, the project supports ESP32-C3 and ESP32-S3.
* Differential OTA updates can be significantly smaller if you only change the application code. Differential OTA updates are supported via [esp_delta_ota](https://components.espressif.com/components/espressif/esp_delta_ota) component.

rbruneau-altyor, 2023-07-13:
Hi @igrr, thank you for the information. I will definitely check the components you mentioned

closed 2023-07-13 by rbruneau-altyor