espressif/esp-idf #11862: DHCP Client is not stopped when netif is being deleted

ermacv, 2023-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

DHCP Client should be stopped before it is deleted when netif is being deleted.

### What is the actual behavior?

DHCP Client is just deleted with a function `dhcp_cleanup(esp_netif->lwip_netif);` that doesn't check the state of the DHCP Client.

### Steps to reproduce.

Configure WiFi STA
Connect to WiFi AP
Configure netif as DHCP Client for WiFi STA
Deconfigure netif
Deconfigure WiFi STA

### Debug Logs.

```plain
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.

Core  1 register dump:
PC      : 0x42083ccc  PS      : 0x00060130  A0      : 0x82083d90  A1      : 0x3fcc21e0  
0x42083ccc: dhcp_fine_tmr at /home/ermacv/esp/esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c:553

A2      : 0x3fcd8574  A3      : 0x400059f4  A4      : 0x00000004  A5      : 0x00060323  
0x400059f4: __call_ampdu_dispatch_upto in ROM

A6      : 0xb33fffff  A7      : 0xb33fffff  A8      : 0x8207aff6  A9      : 0x3fcc21c0  
A10     : 0x3fcd739c  A11     : 0x00000000  A12     : 0x00060320  A13     : 0x00000000  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000015  EXCCAUSE: 0x00000003  
EXCVADDR: 0x400059fe  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000  
0x400059fe: __call_ampdu_dispatch_upto in ROM

0x400570e8: memset in ROM

0x400570f3: memset in ROM

Backtrace: 0x42083cc9:0x3fcc21e0 0x42083d8d:0x3fcc2200 0x42082119:0x3fcc2220 0x4207ac99:0x3fcc2240
0x42083cc9: dhcp_fine_tmr at /home/ermacv/esp/esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c:551

0x42083d8d: dhcp_fine_timeout_cb at /home/ermacv/esp/esp-idf/components/lwip/lwip/src/core/ipv4/dhcp.c:316

0x42082119: sys_check_timeouts at /home/ermacv/esp/esp-idf/components/lwip/lwip/src/core/timeouts.c:401

0x4207ac99: tcpip_timeouts_mbox_fetch at /home/ermacv/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:98
 (inlined by) tcpip_thread at /home/ermacv/esp/esp-idf/components/lwip/lwip/src/api/tcpip.c:142

ELF file SHA256: 3cec974d01adcd83

E (13900) esp_apptrace: Trace destination 1 not supported!
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382130
0x40382130: spinlock_release at /home/ermacv/esp/esp-idf/components/esp_hw_support/include/spinlock.h:160
 (inlined by) vPortExitCritical at /home/ermacv/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:523

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3ad8,len:0x381c
load:0x403c9700,len:0xe78
load:0x403cc700,len:0x54f0
entry 0x403c9980
```

### More Information.

Currently I use my own API that splits netif and WiFi as much as possible but I don't think that it is related to my code.
I also found that the following function is called when netif is deleted:
```c
static void esp_netif_lwip_remove(esp_netif_t *esp_netif)
{
    if (esp_netif->lwip_netif) {
        if (netif_is_up(esp_netif->lwip_netif)) {
            netif_set_down(esp_netif->lwip_netif);
        }
        netif_remove(esp_netif->lwip_netif);
#if ESP_GRATUITOUS_ARP
        if (esp_netif->flags & ESP_NETIF_FLAG_GARP) {
            netif_unset_garp_flag(esp_netif->lwip_netif);
        }
#endif
#if ESP_MLDV6_REPORT && LWIP_IPV6
        if (esp_netif->flags & ESP_NETIF_FLAG_MLDV6_REPORT) {
            netif_unset_mldv6_flag(esp_netif->lwip_netif);
        }
#endif
        if (esp_netif->flags & ESP_NETIF_DHCP_CLIENT) {
#if CONFIG_LWIP_IPV4
            dhcp_cleanup(esp_netif->lwip_netif);
#endif
        }

    }
}
```

As you can see the only DHCP client related part just deletes the DHCP Client (`dhcp_cleanup(esp_netif->lwip_netif);`) that could be started and waiting for internal timer callback. And when callback happens then it uses invalid DHCP structure fields (NULL) that leads to CPU exception.

I also added `dhcp_release_and_stop(esp_netif->lwip_netif);` function before `dhcp_cleanup(esp_netif->lwip_netif);` and I don't see problems anymore. But I didn't performed a lot of tests to check different scenarios.

ermacv, 2023-07-14:
I performed some extra tests and again I can reproduce the bug even with `dhcp_release_and_stop` function.

I checked that DHCP Client fine timer was added in v5.1 release and I've just started migration for my project from v4.4.5 to v5.1 and I haven't seen this behavior on v4.4.5.
I also disabled `ESP_LWIP_DHCP_FINE_TIMERS_ONDEMAND` option and I can't verify the bug anymore. So it seems that the behavior is related to fine timers.

espressif-abhikroy, 2024-05-22:
This issue is addressed in the following commits.

ESP-IDF
Fix: Added check to ensure safe restart of dhcp fine timer
V5.2: 9a2fb1a4a5d8a5805e09231dbee52a3e38794df5
V5.1: 59a06a8ad21c8a58a09b26d9629fbe2a14a05391
V5.0: 73d238cdfb411c169b75efe300924b09570f4b19
 
lwip/dhcp: Fixed ondemand fine timers leakage bug
V4.3: 84d93a31dd7ced4c610ef9200e1042a78d91354d
V4.4: c781fda86f8205abb12fc7343beb17acf2eec3e3

ermacv, 2024-06-03:
Hello! The issue was related to my code (wrong usage of LwIP and WiFi/Eth components). Issue could be closed.

closed 2024-06-03 by ermacv

LauravL0623, 2024-06-03:
> Hello! The issue was related to my code (wrong usage of LwIP and WiFi/Eth components). Issue could be closed.

I'm currently facing the same issue; I'm seeing DHCP messages on Wireshark even after calling `dhcp_release_and_stop()`. Can you elaborate on what you did wrong? It might help me and others like me in the future :)