espressif/esp-idf #11864: ESP32-C3: I2C master hardware issue with the WRITE op

rojer, 2023-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The WRITE op over-fetches the data: if told to send N bytes, it will fetch N+1 bytes from the TX FIFO.

Consider a simple sequence of commands:
 * 00003000 RSTART
 * 00000902 WRITE 2, ACK_CHECK_EN
 * 00001000 STOP
With 2 bytes in the FIFO at the beginning (fifo_st.tx_fifo_raddr = 0, fifo_st.tx_fifo_waddr = 2, sr.tx_fifo_cnt = 2), this results in 2 bytes sent to the bus, as expected.
However, a third byte is fetched from the FIFO into the internal shift register, as evidenced by the FIFO underflow at the end: fifo_st.tx_fifo_raddr = 3, fifo_st.tx_fifo_waddr = 2, sr.tx_fifo_cnt = 63

So, if two bytes are sent on the wire, what's the problem, you might ask?
Well, the problem arises in a slightly more complicated scenario, involving END.
Consider two transactions:

 * 00003000 RSTART
 * 00000901 WRITE 1, ACK_CHECK_EN
 * 00000901 WRITE 1, ACK_CHECK_EN (a)
 * 00002000 END

 * 00003000 RSTART
 * 00000901 WRITE 1, ACK_CHECK_EN (b)
 * 00001c01 READ 1, NAK
 * 00001000 STOP

This is a 1-byte register read, split into two transactions separated by END (first writes addr and reg addr, second writes addr and reads value), with bus held in between.
As with the case above, the first transaction ends with FIFO underflow and when it comes time to execute the second transaction, WRITE (b) will send whatever was in the FIFO at the time of WRITE (a).
This is not the case if END is replaced with STOP - it seems to reset the internal state.

With the previous version of the hardware we had a bug with the END operation (https://github.com/espressif/esp-idf/issues/606). With the newer revision (date = 20070201) that one has been fixed but it seems we have a new one.
Third time the charm? :)

igrr, 2023-07-13:
@rojer Just to clarify, are you using ESP-IDF I2C driver or writing your own? If you are using ESP-IDF, please mention the version you are using.

If you are writing your own driver, have you checked if ESP-IDF has the same issue?

rojer, 2023-07-13:
i am writing my own, working directly with hardware. if i get some free time, i might give idf's driver a go but at this point i'm pretty sure it's abnormal hardware behavior.

igrr, 2023-07-13:
Right, the question is — if IDF driver has the same issue then we can assign this issue to someone to investigate and try to find a workaround. If this is a chip bug, and IDF driver doesn't suffer from this issue, then there is not much we can do in IDF project about it.

rojer, 2023-07-15:
ok, so i repro'd this behavior using IDF's driver:

```
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, 0x40 << 1 | WRITE_BIT, ACK_CHECK_EN);
    i2c_master_write_byte(cmd, 0x03, ACK_CHECK_EN);
    // i2c_master_stop(cmd);
    ret = i2c_master_cmd_begin(i2c_num, cmd, 1000 / portTICK_PERIOD_MS);
    volatile i2c_dev_t *dev = &I2C0;
    ESP_LOGI(TAG, "ret %d FIFO tx %u/%u cnt %u", ret, dev->fifo_st.tx_fifo_raddr, dev->fifo_st.tx_fifo_waddr, dev->sr.tx_fifo_cnt);
    if (ret != ESP_OK) {
        return ret;
    }
    i2c_cmd_link_delete(cmd);
```

... but only with FIFO protection disabled (`I2C0.fifo_conf.fifo_prt_en = false`). so i guess there's still a hardware issue but the interest is largely theoretical now.

mythbuster5, 2025-01-13:
As you said you reproduce the issue with `fifo_prt_en` as false. But when this reg set to false the fifo is not work as fifo anymore.. So please set it as true when you write data.

closed 2025-01-13 by mythbuster5