espressif/esp-idf #11866: esp-idf-components-lovyanGFX

nadeem268, 2023-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.0

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I  am using a code examples usb_camera _lcd  with esp32s3 mcu  , and i have added LovyanGFX library for LCD for usb camera live streaming on ili9488 lcd  , in my main.cpp file i have included the library 
#define LGFX_USE_V1
#include "LovyanGFX.h"

### What is the actual behavior?

When i compile the code i got  one error like this
 /esp-iot-solution-usb-add_usb_solutions1/examples/usb/host/usb_camera_wifi_transfer/components/LovyanGFX/src/lgfx/v1/platforms/esp32/common.hpp:28:10: fatal error: driver/gpio.h: No such file or directory
   28 | #include <driver/gpio.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
[879/991] Building CXX object esp-idf/LovyanGFX/CMak...idf_LovyanGFX.dir/src/lgfx/v1/misc/pixelcopy.cpp.obj
ninja: build stopped: subcommand failed.

### Steps to reproduce.

1. Step
2. Step
3. Step
...
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_CP_DMA_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_RISCV_COPROC_SUPPORTED=y
CONFIG_SOC_USB_OTG_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_CACHE_SUPPORT_WRAP=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_PSRAM_DMA_CAPABLE=y
CONFIG_SOC_XT_WDT_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
[... 1740 lines cut ...]
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32S2_TIME_SYSCALL_USE_RTC_SYSTIMER=y
CONFIG_ESP32S2_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32S2_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32S2_TIME_SYSCALL_USE_SYSTIMER is not set
# CONFIG_ESP32S2_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32S2_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32S2_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

suda-morris, 2023-07-14:
@nadeem268 You may want to read this migration guide first: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#specify-component-requirements-explicitly

and then add the "driver" as a dependency of LovyanGFX in your component CMake

Alvin1Zhang, 2023-08-01:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates.

closed 2023-08-01 by Alvin1Zhang

orewaBrt, 2024-01-25:
Hello everyone. Sorry for bothering. I'm new to this. Just tried to build project I found online but I keep getting these errors. 
OS is Win 11, VScode ext ESP-IDF. WT32 SC01 Plus is device.

errors :

warning: "REG_SPI_BASE"
warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   80 |         dev->channel[channel].in.int_ena.val |= mask;
[1023/1154] Building CXX object esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/platforms/esp32/common.cpp.objFAILED: esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/platforms/esp32/common.cpp.obj

Here is the log. After esp32s3 image has been created: 

Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated C:/Users/balsh/OneDrive/Desktop/espidf-project/ESP32-TUX/build/bootloader/bootloader.bin
[96/96] cmd.exe /C "cd /D C:\Users\balsh\OneDrive\Desktop\espidf-proje...rive/Desktop/espidf-project/ESP32-TUX/build/bootloader/bootloader.bin 
Bootloader binary size 0x5040 bytes. 0x2fc0 bytes (37%) free.
[935/1154] Building CXX object esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp.objIn file included from C:/Users/balsh/esp/esp-idf/components/esp_hw_support/include/esp_memory_utils.h:12,
                 from C:/Users/balsh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:79,
                 from C:/Users/balsh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:58,
                 from C:/Users/balsh/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:70,
                 from C:/Users/balsh/esp/esp-idf/components/driver/include/esp_private/spi_common_internal.h:13,
                 from C:/Users/balsh/OneDrive/Desktop/espidf-project/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.hpp:30,
                 from C:/Users/balsh/OneDrive/Desktop/espidf-project/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:26:
C:/Users/balsh/esp/esp-idf/components/soc/esp32s3/include/soc/soc.h:37: warning: "REG_SPI_BASE" redefined
   37 | #define REG_SPI_BASE(i)         (((i)>=2) ? (DR_REG_SPI2_BASE + (i-2) * 0x1000) : (0))    // GPSPI2 and GPSPI3
      |
C:/Users/balsh/OneDrive/Desktop/espidf-project/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32/Bus_SPI.cpp:23: note: this is the location of the previous definition
   23 |  #define REG_SPI_BASE(i)   (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
      |
[939/1154] Building CXX object esp-idf/LovyanGFX/CMakeFiles/__idf_LovyanGFX.dir/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp.objIn file included from C:/Users/balsh/OneDrive/Desktop/espidf-project/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp:29:
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/gdma_ll.h: In function 'void gdma_ll_rx_enable_interrupt(gdma_dev_t*, uint32_t, uint32_t, bool)':
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/gdma_ll.h:80:46: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   80 |         dev->channel[channel].in.int_ena.val |= mask;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/gdma_ll.h:82:46: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
   82 |         dev->channel[channel].in.int_ena.val &= ~mask;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from C:/Users/balsh/OneDrive/Desktop/espidf-project/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp:29:
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/gdma_ll.h: In function 'void gdma_ll_tx_enable_interrupt(gdma_dev_t*, uint32_t, uint32_t, bool)':
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/gdma_ll.h:328:47: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  328 |         dev->channel[channel].out.int_ena.val |= mask;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/gdma_ll.h:330:47: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  330 |         dev->channel[channel].out.int_ena.val &= ~mask;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
In file included from C:/Users/balsh/OneDrive/Desktop/espidf-project/ESP32-TUX/components/LovyanGFX/src/lgfx/v1/platforms/esp32s3/Bus_RGB.cpp:32:
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/lcd_ll.h: In function 'void lcd_ll_enable_interrupt(lcd_cam_dev_t*, uint32_t, bool)':
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/lcd_ll.h:557:33: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  557 |         dev->lc_dma_int_ena.val |= mask & 0x03;
      |         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
C:/Users/balsh/esp/esp-idf/components/hal/esp32s3/include/hal/lcd_ll.h:559:33: warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
  559 |         dev->lc_dma_int_ena.val &= ~(mask & 0x03);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~
[... cut ...]
  643 |       (&dev->comd0)[index].val = cmd_val;
      |              ^~~~~
      |              comd
[1040/1154] Building C object esp-idf/lvgl/CMakeFiles/__idf_lvgl.dir/src/extra/libs/rlottie/lv_rlottie.c.objninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\balsh\OneDrive\Desktop\espidf-project\ESP32-TUX\build\log\idf_py_stderr_output_8772 and C:\Users\balsh\OneDrive\Desktop\espidf-project\ESP32-TUX\build\log\idf_py_stdout_output_8772