espressif/esp-idf #11870: ESP32-S3 - incorrect behaviour of EXT1 Wake Up All Low

MarcinInne, 2023-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1,  ESP32-DevKitC

### Power Supply used.

USB

### What is the expected behavior?

Base on documentation [Sleep Modes](https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32/api-reference/system/sleep_modes.html#external-wakeup-ext1l), ESP32-S3 or ESP32 with multi GPIOs pins configure to wake up using **External Wakeup (ext1)** with option **ESP_EXT1_WAKEUP_ALL_LOW** should wake up only when all configured GPIOs have low state.

### What is the actual behavior?

ESP32-S3 don't  behavive correclty as descirbe documentation [Sleep Modes](https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32/api-reference/system/sleep_modes.html#external-wakeup-ext1l) and multi GPIOs pins configure to wake up using **External Wakeup (ext1)** with option **ESP_EXT1_WAKEUP_ALL_LOW**, wake up when only one of pins have low state.
ESP32 don't have this problem and behave correctly.

### Steps to reproduce.

1. Compile below code for target ESP32-S3
```
#include "driver/gpio.h"
#include "esp_sleep.h"
#include "esp_system.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "sdkconfig.h"

#if CONFIG_IDF_TARGET_ESP32
    #define WAKE_UP_GPIO_1	GPIO_NUM_32
    #define WAKE_UP_GPIO_2	GPIO_NUM_33
    #define WAKE_UP_GPIO_3	GPIO_NUM_14
#endif

#if CONFIG_IDF_TARGET_ESP32S3
    #define WAKE_UP_GPIO_1	    GPIO_NUM_4
    #define WAKE_UP_GPIO_2	    GPIO_NUM_6
    #define WAKE_UP_GPIO_3	    GPIO_NUM_7
#endif

static const char *TAG = "app";

void print_gpio_status(gpio_num_t gpio_num)
{
    gpio_set_direction(gpio_num, GPIO_MODE_INPUT);
    gpio_set_pull_mode(gpio_num, GPIO_FLOATING);
    ESP_LOGI(TAG, "GPIO %d level %d", gpio_num, gpio_get_level(gpio_num));
}

void app_main(void)
{
    esp_sleep_wakeup_cause_t wake_up_cause = esp_sleep_get_wakeup_cause();
    ESP_LOGI(TAG, "Wake Up Couse %d", wake_up_cause);

    if(ESP_SLEEP_WAKEUP_EXT1 == wake_up_cause) {
        uint64_t ext1_wake_up_status = esp_sleep_get_ext1_wakeup_status();
        ESP_LOGI(TAG, "Wake Up EXT1 status 0x%08llx", ext1_wake_up_status);
    }

    print_gpio_status(WAKE_UP_GPIO_1);
    print_gpio_status(WAKE_UP_GPIO_2);
    print_gpio_status(WAKE_UP_GPIO_3);

    uint64_t ext1_wake_up_status = 0;
    ext1_wake_up_status |= (1ULL << WAKE_UP_GPIO_1);
    ext1_wake_up_status |= (1ULL << WAKE_UP_GPIO_2);
    ext1_wake_up_status |= (1ULL << WAKE_UP_GPIO_3);

    ESP_LOGI(TAG, "Set Wake Up EXT1 status 0x%08llx", ext1_wake_up_status);
    esp_err_t err = esp_sleep_enable_ext1_wakeup(ext1_wake_up_status, ESP_EXT1_WAKEUP_ALL_LOW);
    if (err) {
        ESP_LOGE(TAG, "Set Wake Up EXT1 error 0x%x", err);
    }

    vTaskDelay( pdMS_TO_TICKS(5000));

    ESP_LOGI(TAG, "deep sleep start");
    esp_deep_sleep_start();
}
```
2. Set GPIOs 4, 6, 7 to high state
3. Flash ESP32-S3
4. Set one of GPIOs (4 or 6 or 7) to low state
5. ESP3232 wake up, but it shouldn't.  ESP32S3 should only wake up only when all GPIOs (4, 6, 7) are low state
6. Repeat all above steps on ESP32 and it behave correctly. Only wake up when all GPIOs (32, 33, 14) are low state.

### Debug Logs.

```plain
I (0) cpu_start: App cpu up.
I (193) cpu_start: Pro cpu start user code
I (194) cpu_start: cpu freq: 160000000 Hz
I (194) cpu_start: Application information:
I (197) cpu_start: Project name:     test_deep_sleep
I (202) cpu_start: App version:      1
I (207) cpu_start: Compile time:     Jul 13 2023 17:37:05
I (213) cpu_start: ELF file SHA256:  0b443dd94c5e5576...
I (219) cpu_start: ESP-IDF:          v5.1
I (223) cpu_start: Min chip rev:     v0.0
I (228) cpu_start: Max chip rev:     v0.99 
I (233) cpu_start: Chip rev:         v0.1
I (238) heap_init: Initializing. RAM available for dynamic allocation:
I (245) heap_init: At 3FC95500 len 00054210 (336 KiB): DRAM
I (251) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (258) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (264) heap_init: At 600FE070 len 00001F90 (7 KiB): RTCRAM
I (271) spi_flash: detected chip: generic
I (275) spi_flash: flash io: dio
I (292) sleep: Configure to isolate all GPIO pins in sleep state
I (299) sleep: Enable automatic switching of GPIO sleep configuration
I (306) app_start: Starting scheduler on CPU0
I (311) app_start: Starting scheduler on CPU1
I (311) main_task: Started on CPU0
I (321) main_task: Calling app_main()
I (321) app: Wake Up Couse 3
I (331) app: Wake Up EXT1 status 0x00000040
I (331) app: GPIO 4 level 1
I (341) app: GPIO 6 level 0
I (341) app: GPIO 7 level 1
I (341) app: Set Wake Up EXT1 status 0x000000d0
I (5351) app: deep sleep start
```

### More Information.

Is this the software or hardware issue (faulty unit) , or was there a hardware modification between ESP32S3 and ESP32. Documentation says that deep sleep and wake up should behave the same for all ESP32 family.

10086loutianhao, 2023-07-20:
@MarcinInne  hello, this is not a hardware bug, but rather an error in our EXT1 example and the official documentation. In reality, except for the ESP32, other chips such as ESP32-S2, ESP32-S3, ESP32-C6, and ESP32-H2 only support ANY LOW wake-up, not ALL LOW.

MarcinInne, 2023-09-05:
Thanks for information
Please update information about EXT1 Wake Up in official documentation for all ESP32 family, beside original ESP32

closed 2023-09-05 by MarcinInne