espressif/esp-idf #11872: Can't use separately UART0 and USB SERIAL JTAG with ESP32H2

RJSDevel, 2023-07-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dirty

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom ESP32H2

### Power Supply used.

USB

### What is the expected behavior?

Separated uart0 and USB SERIAL JTAG input/output

### What is the actual behavior?

I'm trying to use USB SERIAL JTAG as uart bridge and use uart0 for logs, when write logs vie ESP_LOGI, the logs go through USB SERIAL JTAG too, but if I write some data to USB SERIAL JTAG, it doesn't print to uart.

I tryed play with 
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG
CONFIG_ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG
and even CONFIG_ESP_CONSOLE_NONE doesn't work well, logs whatever print

### Steps to reproduce.

`
#include "hal/usb_serial_jtag_ll.h"
#include "driver/usb_serial_jtag.h"

usb_serial_jtag_driver_config_t usb_serial_config = USB_SERIAL_JTAG_DRIVER_CONFIG_DEFAULT();
ESP_ERROR_CHECK(usb_serial_jtag_driver_install(&usb_serial_config));
xTaskCreate(jtag_uart_task, "jtag_uart_task", 2048, NULL, 2 | portPRIVILEGE_BIT, NULL);

static void jtag_uart_task(void *args) {
	uint8_t *data = (uint8_t *) malloc(MT_RPC_DATA_MAX);
    while (true) {
    	int len = usb_serial_jtag_read_bytes(data, MT_RPC_DATA_MAX, 0 / portTICK_PERIOD_MS);
    	if(len > 0) {
    		MT_ProcessIncoming(data);
    	}
    }
}
`

Write - usb_serial_jtag_write_bytes(pRspMsg, sizeof(pRspMsg),  0);

mythbuster5, 2023-07-14:
Sorry, I'm not very clear. So, what you mean is when you want to print via uart0, the data should not print via usb_serial_jtag right?

RJSDevel, 2023-07-14:
Yes, I wanna print logs via uart0 and transfer data vie USB JTAG serial

RJSDevel, 2023-07-14:
Currently USB and UART work separatly, but usb print when chip stars, it was'nt obviously

closed 2023-07-14 by RJSDevel

mythbuster5, 2023-07-17:
@RJSDevel May I know what did you do? I'm afraid other people have similar problem, thanks!

RJSDevel, 2023-07-17:
The PC application thought that my esp32 device is in bad state, and just rebooted it, and when the esp32 started that again wrote logs to the PC program. I just resolve esp32 troubles. But haven't found the way to disable USB_SERIAL_JTAG logs on start of the chip, and I passed it. USB_SERIAL_JTAG and UART0 logs are very simmilar and I was confused why UART0 writes to USB_SERIAL_JTAG