espressif/esp-idf #11874: idf.py command on Windows 10 opens a new Python window

krupis, 2023-07-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello. I have recently purchased ESP32-S3-EYE
and following instructions on how to run the examples:
https://github.com/espressif/esp-who/#run-examples

I have downloaded esp-who project.

Step 1 says that I need to open the terminal and go to any example folder in the project:
[image]

I cannot move past this step as I am facing multiple issues.

**Issue 1**

idf.py command is acting weird on windows 10.

When I try to execute command ```idf.py set-target esp32s3```, instead of executing the set-traget command, Python editor opens automatically. Why would that happen and how to prevent that?

[image]

**Issue 2**

I have been playing around and the only way to work around Issue 1 (preventing python editor from opening), I have to open the project using vs-code with esp-idf extension and then click open ESP-IDF terminal:
[image]

now here, I can execute ```idf.py set-target esp32s3``` command without the python editor opening but I get another issue that I cant seem to be able to solve.

```
PS C:\Users\petrikas.lu\Desktop\WORK\ESP32\esp-who\examples\motion_detection\lcd> idf.py set-target esp32s3
C:\Users\petrikas.lu\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory 'c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\motion_detection\lcd\build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.
PS C:\Users\petrikas.lu\Desktop\WORK\ESP32\esp-who\examples\motion_detection\lcd>
```

I cant really understand how am I meant to clean it. What do I need to delete and why it cannot be automatically done?

Appreciate any help!

igrr, 2023-07-14:
> idf.py command is acting weird on windows 10.

In this case, are you opening a regular Windows command prompt, or "IDF Command Prompt" shortcut created by the installer? This command will work inside IDF Command Prompt (or Powershell, if you prefer that), since it will have all the required variables set up.

> Directory 'c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\motion_detection\lcd\build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.

You can simply do what the message says, and delete the build directory. Then run `idf.py set-target esp32s3` again.

krupis, 2023-07-14:
Thanks for a very quick response.

> In this case, are you opening a regular Windows command prompt, or "IDF Command Prompt" shortcut created by the installer? This command will work inside IDF Command Prompt (or Powershell, if you prefer that), since it will have all the required variables set up.
> 

I have tried to open a regular Command Prompt as well as ESP-IDF 5.0 CMD and both work the same. They both open python editor when I execute idf.py command.
I think I had this issue ages ago, and I had found  the solution. The solution was to provide full python and idf.py paths but I cant really reproduce this solution nor find the forum thread where this was discussed. 

>
> You can simply do what the message says, and delete the build directory. Then run idf.py set-target esp32s3 again.
> 

I have tried the following:
```
PS C:\Users\petrikas.lu\Desktop\WORK\ESP32\esp-who\examples\motion_detection\lcd> rm build

Confirm
The item at C:\Users\petrikas.lu\Desktop\WORK\ESP32\esp-who\examples\motion_detection\lcd\build has children and the Recurse parameter was not specified. If you continue, 
all children will be removed with the item. Are you sure you want to continue?
[Y] Yes  [A] Yes to All  [N] No  [L] No to All  [S] Suspend  [?] Help (default is "Y"): Y
PS C:\Users\petrikas.lu\Desktop\WORK\ESP32\esp-who\examples\motion_detection\lcd> idf.py set-target esp32s3
C:\Users\petrikas.lu\esp\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\motion_detection\lcd\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\motion_detection\lcd\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\motion_detection\lcd"...
CMake Error at C:/Users/petrikas.lu/esp/esp-idf/tools/cmake/targets.cmake:19 (message):
  IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.   
  To change the target, clear the build directory and sdkconfig file, and     
  build the project again
Call Stack (most recent call first):
  C:/Users/petrikas.lu/esp/esp-idf/tools/cmake/project.cmake:7 (__target_init)
  CMakeLists.txt:7 (include)
```

There still seem to be an issue with the sdkconfig and build files.

igrr, 2023-07-14:
For the first issue (idf.py), please open IDF Command prompt and run: `doskey /MACROS:ALL`, and paste the output here.

> IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.

For the second issue, the error message suggests that there is still a CMakeCache.txt file left in the build directory. After deleting the build directory, could you please check (for example, in Windows Explorer) that the build directory was really deleted?

Also, could you check if you can `idf.py set-target esp32s3` in some other project, for example inside the `blink` example in ESP-IDF?

krupis, 2023-07-14:
> For the first issue (idf.py), please open IDF Command prompt and run: `doskey /MACROS:ALL`, and paste the output here.
> 
> > IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.

I have opened ESP-IDF Command Prompt and executed ```doskey /MACROS:ALL```. It does not return anything.

[image]

> 
> For the second issue, the error message suggests that there is still a CMakeCache.txt file left in the build directory. After deleting the build directory, could you please check (for example, in Windows Explorer) that the build directory was really deleted?
> 

I have manually deleted the build directory and ensured through windows explorer that there is no build folder. After that, I tried to execute ```idf.py set-target esp32s3``` command but it still complains:

```
PS C:\Users\petrikas.lu\Desktop\WORK\ESP32\esp-who\examples\color_detection\lcd> idf.py set-target esp32s3
C:\Users\petrikas.lu\esp5\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\color_detection\lcd\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\color_detection\lcd\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 c:\users\petrikas.lu\desktop\work\esp32\esp-who\examples\color_detection\lcd"...
CMake Error at C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/targets.cmake:19 (message):
  IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.    
  To change the target, clear the build directory and sdkconfig file, and      
  build the project again
Call Stack (most recent call first):
  C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/project.cmake:7 (__target_init)
  CMakeLists.txt:7 (include)
```

> Also, could you check if you can `idf.py set-target esp32s3` in some other project, for example inside the `blink` example in ESP-IDF?

I opened clean(previously has not been compiled ever) example project "hello world" in vscode and opened ESP-IDF terminal in vscode. I ran ```idf.py set-target esp32s3``` and tried to set the target using command ```idf.py set-target esp32s3```. It does not seem to work:

```
PS C:\Users\petrikas.lu\esp5\esp-idf\examples\get-started\hello_world> idf.py set-target esp32s3
C:\Users\petrikas.lu\esp5\esp-idf\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\Users\petrikas.lu\esp5\esp-idf\examples\get-started\hello_world\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory C:\Users\petrikas.lu\esp5\esp-idf\examples\get-started\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 C:\Users\petrikas.lu\esp5\esp-idf\examples\get-started\hello_world"...
CMake Error at C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/targets.cmake:19 (message):
  IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.
  To change the target, clear the build directory and sdkconfig file, and
  build the project again
Call Stack (most recent call first):
  C:/Users/petrikas.lu/esp5/esp-idf/tools/cmake/project.cmake:7 (__target_init)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the C:\Users\petrikas.lu\esp5\esp-idf\examples\get-started\hello_world\build\log\idf_py_stderr_output_20132 and C:\Users\petrikas.lu\esp5\esp-idf\:\Users\petrikas.lu\esp5\esp-idf\examples\get-started\hello_world\build\log\idf_py_stdout_output_20132
PS C:\Users\petrikas.lu\esp5\esp-idf\examples\get-started\hello_world>
```

However, it works without any issues if I change target using the below method:
1. Click on vscode currently selected target (in my case its esp32)
[image]

2. Select desired target:

[image]

[... 148 lines cut ...]
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/Users/petrikas.lu/.espressif5/python_env/idf5.0_py3.11_env/Scripts/python.exe (found version "3.11.2") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "hello_world" version: v5.0.3
-- Adding linker script C:/Users/petrikas.lu/esp5/esp-idf/examples/get-started/hello_world/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/petrikas.lu/esp5/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script C:/Users/petrikas.lu/esp5/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
[... cut ...] mponents/unity C:/Users/petrikas.lu/esp5/esp-idf/components/usb C:/Users/petrikas.lu/esp5/esp-idf/components/vfs C:/Users/petrikas.lu/esp5/esp-idf/components/wear_levelling C:/Users/petrikas.lu/esp5/esp-idf/components/wifi_provisioning C:/Users/petrikas.lu/esp5/esp-idf/components/wpa_supplicant C:/Users/petrikas.lu/esp5/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/petrikas.lu/esp5/esp-idf/examples/get-started/hello_world/build
```

igrr, 2023-07-14:
I see! Yes, of course, VS Code extension will set the target selected in the UI automatically. What you finally did (selected the target in VS Code) is the correct approach there.

> As you can see as soon as I opened the ESP-IDF Command Prompt it prints The system cannot find the path specified.. Perhaps that is one of the issues?

Yes, probably this means that the script which runs first when this shortcut is opened got moved somewhere or deleted. If you still want to use the command prompt I would recommend running the installer again so that it can recreate the shortcuts. Alternatively, you can use the terminal within VS Code, as that seemingly is working fine.

krupis, 2023-07-14:
@igrr 

> Alternatively, you can use the terminal within VS Code, as that seemingly is working fine.

That is exactly why I am confused since selecting target using terminal within VS code does not work. 
**For example, if I open vscode terminal:**

[image]

and run ```idf.py set-target esp32``` command, it will not go through (the project is clean and there is no build folder):

[image]

**But I can just simply select the target by the following method:**
[image]

[image]

and it will work just fine.

I dont understand why it works one way but not another

krupis, 2023-07-14:
> Since now we are talking about reinstalling your ESP-IDF, I'd like to suggest to install the latest bugfix version of ESP-IDF v4.4.* because the esp-who documentation suggests this - I can just guess but it looks like ESP-IDF v5.0 is not yet supported (?).

Good point. Thanks il install v.4.4 for ESP-WHO

krupis, 2023-07-14:
I am still confused regarding setting target using vscode Terminal. 
 
How come ```idf.py set-target esp32``` works without any issues:
```
idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'd:\programming\esp32\esp-eye-test\esp-who\examples\color_detection\lcd\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory d:\programming\esp32\esp-eye-test\esp-who\examples\color_detection\lcd\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=1 d:\programming\esp32\esp-eye-test\esp-who\examples\color_detection\lcd"...
-- Found Git: C:/Users/Lukas/.espressif4/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/Lukas/.espressif4/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/Lukas/.espressif4/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/Lukas/.espressif4/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Processing 1 dependencies:
[1/1] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: C:/Users/Lukas/esp4/esp-idf/components/freemodbus.
-- Project sdkconfig file D:/Programming/ESP32/esp-eye-test/esp-who/examples/color_detection/lcd/sdkconfig
Loading defaults file D:/Programming/ESP32/esp-eye-test/esp-who/examples/color_detection/lcd/sdkconfig.defaults...
-- Found Python3: C:/Users/Lukas/.espressif4/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "color_detection_lcd" version: v0.9.3-170-g5497ff2
-- Adding linker script D:/Programming/ESP32/esp-eye-test/esp-who/examples/color_detection/lcd/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/Users/Lukas/esp4/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- DEBUG: Use esp-modbus component folder: C:/Users/Lukas/esp4/esp-idf/components/freemodbus.
-- Components: app_trace app_update asio bootloader bootloader_support bt bus cbor cmock coap console cxx driver efuse esp-code-scanner esp-dl esp-sr esp-tls esp32 esp32-camera esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs fb_gfx freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns modules mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread screen sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/Lukas/esp4/esp-idf/components/app_trace C:/Users/Lukas/esp4/esp-idf/components/app_update C:/Users/Lukas/esp4/esp-idf/components/asio C:/Users/Lukas/esp4/esp-idf/components/bootloader C:/Users/Lukas/esp4/esp-idf/components/bootloader_support C:/Users/Lukas/esp4/esp-idf/components/bt D:/Programming/ESP32/esp-eye-test/esp-who/components/bus C:/Users/Lukas/esp4/esp-idf/components/cbor C:/Users/Lukas/esp4/esp-idf/components/cmock C:/Users/Lukas/esp4/esp-idf/components/coap C:/Users/Lukas/esp4/esp-idf/components/console C:/Users/Lukas/esp4/esp-idf/components/cxx C:/Users/Lukas/esp4/esp-idf/components/driver C:/Users/Lukas/esp4/esp-idf/components/efuse D:/Programming/ESP32/esp-eye-test/esp-who/components/esp-code-scanner D:/Programming/ESP32/esp-eye-test/esp-who/components/esp-dl D:/Programming/ESP32/esp-eye-test/esp-who/components/esp-sr C:/Users/Lukas/esp4/esp-idf/com [... cut ...]
CMake Error at C:/Users/Lukas/esp4/esp-idf/tools/cmake/targets.cmake:19 (message):
  IDF_TARGET in CMake cache does not match IDF_TARGET environment variable.
  To change the target, clear the build directory and sdkconfig file, and
  build the project again
Call Stack (most recent call first):
  C:/Users/Lukas/esp4/esp-idf/tools/cmake/project.cmake:7 (__target_init)
  CMakeLists.txt:7 (include)
```

I can set the target to esp32 and build the esp-who example project. But since I use ESP32-S3 board, I need to change target to esp32s3 but it wont let me. That is so strange!

igrr, 2023-07-14:
> I am still confused regarding setting target using vscode Terminal.

> How come idf.py set-target esp32 works without any issues:

This happens because the target selector in VS Code extension is set to "esp32", which determines the target setting in the environment. When you try to set the target to a different value in the command line inside VS Code, these two settings are in conflict.

Since you are using VS Code extension to build the project, setting the target in VS Code rather than in the command line is the correct approach.

krupis, 2023-07-14:
> This happens because the target selector in VS Code extension is set to "esp32", which determines the target setting in the environment. When you try to set the target to a different value in the [...]

Ahhh it makes sense to me now. Thanks for explaining

closed 2023-07-17 by krupis