espressif/esp-idf #11875: error in ULP macro I_WR_REG when trying to use I_GPIO_HOLD_DIS in the wake stub

0x0fe, 2023-07-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

expected that  I_GPIO_HOLD_DIS can be used from the wake stub.

### What is the actual behavior?

In function 'void esp_wake_deep_sleep()':
```
C:/Users/xxxxl/.platformio/packages/framework-espidf/components/ulp/include/esp32/ulp.h:347:5: error: expected primary-expression before '.' token
     .wr_reg = {\  

C:/Users/xxxx/.platformio/packages/framework-espidf/components/ulp/include/esp32/ulp.h:375:39: note: in expansion of macro 'I_WR_REG'
 #define I_WR_REG_BIT(reg, shift, val) I_WR_REG(reg, shift, shift, val)   
```

### Steps to reproduce.

call` I_GPIO_HOLD_DIS((gpio_num_t)16);` in the default wake stub.

### Build or installation Logs.

```plain
C:/Users/xxxxl/.platformio/packages/framework-espidf/components/ulp/include/esp32/ulp.h:347:5: error: expected primary-expression before '.' token
     .wr_reg = {\  

C:/Users/xxxx/.platformio/packages/framework-espidf/components/ulp/include/esp32/ulp.h:375:39: note: in expansion of macro 'I_WR_REG'
 #define I_WR_REG_BIT(reg, shift, val) I_WR_REG(reg, shift, shift, val)   
```
```

### More Information.

So, i use ESP32D0WDR2 which has internal 2M PSRAM, just before when entering deep sleep I hold GPIO16 high to avoid a current leak.
```
  pinMode(16,OUTPUT);
  digitalWrite(16,1);
  gpio_hold_en((gpio_num_t) 16);
```
However, when waking from sleep the PSRAM fails to initilize (cannot read ID), which makes sense, the GPIO16 is still held high, and for some reason the PSRAM init function do not disable the HOLD. 
However, calling 
```
  gpio_hold_dis((gpio_num_t) 16);
  gpio_deep_sleep_hold_dis();
  rtc_gpio_force_hold_dis_all();
```
At the very beginning of the firmware does not solve the problem, i assume the reason is simply that the PSRAM is initilized either in early boot or even in bootloader, so disabling the HOLD afterwards cannot solve the issue.

Since i use the wake stub for other purposes i figured i should disable the HOLD here, and found this macro, sadly the macro wont compile, i assume there is a trick perhaps related to the scope of some structures used in ulp.h, but i could not find anything.

ESP-Marius, 2023-07-17:
I think you might be getting some things mixed up here.

`ulp.h` contains macros intended to be used from the ULP co-processor, while the wake-up stub is code that is ran from the main cpu (but ran at a early stage in the boot up process)

ESP-Marius, 2023-10-11:
Closing this for now, but feel free to reopen if there is still anything unresolved here.

closed 2023-10-11 by ESP-Marius