espressif/esp-idf #11879: I'm using the a2dp_sink from ESP_IDF, How to use esp_avrc_feature_flag_t function?

leonardokucarz, 2023-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm already made a pre-configuration for this example work fine, but my objective is bring this data from cell phone:
ESP_AVRC_FEAT_FLAG_COVER_ART_GET_IMAGE_PROP = 0x0080     //  Cover Art GetImageProperties
ESP_AVRC_FEAT_FLAG_COVER_ART_GET_IMAGE = 0x0100     //  capa Art GetImage */
ESP_AVRC_FEAT_FLAG_COVER_ART_GET_LINKED_THUMBNAIL = 0x0200      //  Cover Art GetLinkedThumbnail
Every help is welcome!

leonardokucarz, 2023-07-17:
Good morning @xiongweichao.

So this this function exists on ESP_IDF but isn't usefull yet?
I'm just trying to understand.

graciously.

xiongweichao, 2023-07-17:
@leonardokucarz  ESP-IDF does not support OBEX temporarily, so this function is not supported.

leonardokucarz, 2023-07-17:
@xiongweichao thank you for this explanation.

drabaioli, 2024-02-03:
Did anything change regarding the support of cover art?

xiongweichao, 2024-02-04:
@drabaioli I'm very sorry that this feature is still not supported.

esp-lrh, 2024-09-18:
Hi, @leonardokucarz @drabaioli 
AVRCP cover art feature is now supported as of commit https://github.com/espressif/esp-idf/commit/cda2846558d10cd8a8190e69a2facfb4e57bec54 in the esp-idf master branch. This feature will be included in the upcoming 5.4 release. For usage instructions, you can refer to `examples/bluetooth/bluedroid/classic_bt/a2dp_sink` demo.

`esp_avrc_feature_flag_t` is a set of bit masks defined according to the Bluetooth AVRCP specification, used to determine whether the remote device supports specific features:
https://github.com/espressif/esp-idf/blob/3c99557eeea4e0945e77aabac672fbef52294d54/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/bt_app_av.c#L473-L477

If you have any other questions, please feel free to ask.

closed 2024-12-31 by Alvin1Zhang