espressif/esp-idf #11880: Cannot run `install.sh` when v5.1 is checked out

dot4qu, 2023-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Clone the repo, run `./install.sh`, then `source export.sh` to set up the environment. 

I successfully have the `v5.0` tag working. This problem only appears when I try to run `./install.sh` or `source export.sh` after checking out `v5.1`.

This is the output of `source export.sh` in a new terminal window on `v5.0`, when it is working successfully:
```
[1635] <name>@<Names>-MacBook-Pro ~/Developer/<project>/esp-idf ((HEAD detached at v5.0))  source export.sh
Setting IDF_PATH to '/Users/<name>/Developer/<project>/esp-idf'
Detecting the Python interpreter
Checking "python3" ...
Python 3.9.6
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool openocd-esp32 found in PATH: 0.11.0.
Checking if Python packages are up to date...
Constraint file: /Users/<name>/.espressif/espidf.constraints.v5.0.txt
Requirement files:
 - /Users/<name>/Developer/<project>/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /Users/<name>/.espressif/python_env/idf5.0_py3.9_env/bin/python
/Users/<name>/Developer/<project>/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements are satisfied.
Updated PATH variable:
  /Users/<name>/Developer/<project>/esp-idf/components/esptool_py/esptool:/Users/<name>/Developer/<project>/esp-idf/components/espcoredump:/Users/<name>/Developer/<project>/esp-idf/components/partition_table:/Users/<name>/Developer/<project>/esp-idf/components/app_update:/Users/<name>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/Users/<name>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/Users/<name>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/Users/<name>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/Users/<name>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/Users/<name>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/Users/<name>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Users/<name>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/Users/<name>/.espressif/python_env/idf5.0_py3.9_env/bin:/Users/<name>/Developer/<project>/esp-idf/tools:/opt/homebrew/bin/mosh-server:/opt/homebrew/bin/mosh-server:/opt/homebrew/opt/llvm/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/sbin:/usr/local/opt/llvm/bin:/Users/<name>/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/opt/homebrew/bin/mosh-server:/opt/homebrew/opt/llvm/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/sbin:/usr/local/opt/llvm/bin:/Users/<name>/.cargo/bin:/opt/homebrew/opt/fzf/bin
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```

### What is the actual behavior?

 There is a python error trying to build/parse a regex for a version string somewhere in the esp-idf initialisation code.
```
[1647] <project>@<Names>-MacBook-Pro ~/Developer/<project>/esp-idf ((HEAD detached at v5.1))  ./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.9.6
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Current system platform: macos-arm64
Updating /Users/<project>/.espressif/idf-env.json
Selected targets are: esp32s2, esp32c6, esp32, esp32c2, esp32s3, esp32c3, esp32h2
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@12.1_20221002 (already installed)
Skipping riscv32-esp-elf-gdb@12.1_20221002 (already installed)
Skipping xtensa-esp32-elf@esp-12.2.0_20230208 (already installed)
Skipping xtensa-esp32s2-elf@esp-12.2.0_20230208 (already installed)
Skipping xtensa-esp32s3-elf@esp-12.2.0_20230208 (already installed)
Skipping riscv32-esp-elf@esp-12.2.0_20230208 (already installed)
Skipping esp32ulp-elf@2.35_20220830 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20230419 (already installed)
Traceback (most recent call last):
  File "/Users/<project>/Developer/<project>/esp-idf/tools/idf_tools.py", line 2660, in <module>
    main(sys.argv[1:])
  File "/Users/<project>/Developer/<project>/esp-idf/tools/idf_tools.py", line 2652, in main
    action_func(args)
  File "/Users/<project>/Developer/<project>/esp-idf/tools/idf_tools.py", line 1898, in action_install
    tool_obj.find_installed_versions()
  File "/Users/<project>/Developer/<project>/esp-idf/tools/idf_tools.py", line 726, in find_installed_versions
    ver_str = self.get_version()
  File "/Users/<project>/Developer/<project>/esp-idf/tools/idf_tools.py", line 678, in get_version
    return re.sub(self._current_options.version_regex, self._current_options.version_regex_replace, match.group(0))  # type: ignore
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/re.py", line 210, in sub
    return _compile(pattern, flags).sub(repl, string, count)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/re.py", line 327, in _subx
    template = _compile_repl(template, pattern)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/re.py", line 318, in _compile_repl
    return sre_parse.parse_template(repl, pattern)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/sre_parse.py", line 1036, in parse_template
    addgroup(int(this[1:]), len(this) - 1)
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/sre_parse.py", line 980, in addgroup
    raise s.error("invalid group reference %d" % index, pos)
re.error: invalid group reference 1 at position 1
```

### Steps to reproduce.

Clone the repo, checkout the `v5.1` tag, then run `./install.sh`.

### More Information.

I've tried with different versions of python, I've also tried with the homebrew python and the system python.

dobairoland, 2023-07-17:
Hi @dot4qu. I believe this will be something specific to your system. Thank you for doing the bisecting. The linked commit could make the install script fail if you have multiple ESP-IDF copies and executing the script from one ESP-IDF copy but IDF_PATH points to the other one.

Could you please check that you run `install.sh` in a clean terminal where you haven't run `source export.sh` before (you can check the output of `echo $PATH`)? Also before running `install.sh` could you please check the output of `git status` and `echo $IDF_PATH`? Thank you!

second-string, 2023-07-20:
Output of `git status`:
```
HEAD losgelöst bei v5.1
nichts zu committen, Arbeitsverzeichnis unverändert
```
Sorry for the german, it means HEAD detached at v5.1, nothing to commit, working directory unchanged.

The tests above were run in a fresh shell. There is no `IDF_PATH` automatically set from any *_profile or *rc files, so `echo $IDF_PATH` outputs nothing. There is nothing containing idf in my PATH or in my env.

Are you sure it's a problem with conflicting IDF versions? It doesn't look like it's exiting gracefully when it's finding problems with conflicting installations - the stack trace I posted is an exception in the python code when it's trying to parse some version number (`self._current_options.version_regex_replace`).

second-string, 2023-07-20:
It has something to do with the esp-rom-elfs tool. That's the last tool that's being processed in the `for tool_spec in tools_spec:` block in the `action_install` func in idf_tools.py. 

Looking in my `~/.espressif/tools/` directory, I see that all of the other tools are downloaded except for esp-rom-elfs. Is there a dependency somewhere missing in the install script that should be downloading this, just like all of the xtensa compilers, openocd, etc are installed?

Stepping through the code I see that the IDFToolOptions object for the esp-rom-elfs command is empty:
```> /Users/<name>/Developer/<project>/esp-idf/tools/idf_tools.py(662)get_version()
-> cmd = self._current_options.version_cmd  # type: ignore
(Pdb) 
> /Users/<name>/Developer/<project>/esp-idf/tools/idf_tools.py(663)get_version()
-> if executable_path:
(Pdb) p cmd
['']
(Pdb) p self._current_options
IDFToolOptions(version_cmd=[''], version_regex='', version_regex_replace='\\1', is_executable=False, export_paths=[['']], export_vars=OrderedDict([('ESP_ROM_ELF_DIR', '${TOOL_PATH}/')]), install='always', info_url='https://github.com/espressif/esp-rom-elfs', license='Apache-2.0', strip_container_dirs=0, supported_targets=['all'])
(Pdb) 
```

Then looking at the tool definitions in `esp-idf/tools/tools.json`, I see that the cmd is also empty there: https://github.com/espressif/esp-idf/blob/v5.1/tools/tools.json#L987

Once that `get_version` functions executes the empty version cmd, there's a totally random string from my shell stored. That's why the regex parse is failing, because it's trying to pull a version number out of:
```
'\r\n\nMOSH CONNECT 60001 CyVYmpF+IxpbwwA2/YhjPA\n\nmosh-server (mosh 1.4.0) [build mosh 1.4.0]\nCopyright 2012 Keith Winstein <mosh-devel@mit.edu>\nLicense GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law.\n\n[mosh-server detached, pid = 54309]\n'
```

I have no clue where that is even coming from, as I have mosh installed but had not run it in this shell at all. So how is `install.sh` working for anyone when there's no cmd for the esp-rom-elfs tool?

dobairoland, 2023-07-21:
> Are you sure it's a problem with conflicting IDF versions?

No, I'm not sure. You point out later that there is no version command in the tools file. Yes, that is intentional. And support for this was added to `idf_tools.py`. But if an older `idf_tools.py` is used with that new `tools.json` then it will fail. That is why I thought this might be the case. However, it could be something else we haven't faced before.

dobairoland, 2023-07-21:
> Looking in my ~/.espressif/tools/ directory, I see that all of the other tools are downloaded except for esp-rom-elfs.

So there is no `esp-rom-elfs` directory? I have several versions with ELF files in it (this is not a runable tool). 
```
`--> ls ~/.espressif/tools/esp-rom-elfs/202*
/home/user/.espressif/tools/esp-rom-elfs/20220823:
esp32c2_rev0_rom.elf  esp32c3_rev0_rom.elf  esp32c3_rev3_rom.elf  esp32_rev0_rom.elf  esp32_rev3_rom.elf  esp32s2_rev0_rom.elf  esp32s3_rev0_rom.elf

/home/user/.espressif/tools/esp-rom-elfs/20230113:
esp32c2_rev0_rom.elf  esp32c3_rev0_rom.elf  esp32c3_rev3_rom.elf  esp32c6_rev0_rom.elf  esp32h2_rev0_rom.elf  esp32_rev0_rom.elf  esp32_rev3_rom.elf  esp32s2_rev0_rom.elf  esp32s3_rev0_rom.elf

/home/user/.espressif/tools/esp-rom-elfs/20230320:
esp32c2_rev100_rom.elf  esp32c3_rev0_rom.elf  esp32c3_rev3_rom.elf  esp32c6_rev0_rom.elf  esp32h2_rev0_rom.elf  esp32_rev0_rom.elf  esp32_rev300_rom.elf  esp32s2_rev0_rom.elf  esp32s3_rev0_rom.elf
```

dobairoland, 2023-07-21:
```
IDFToolOptions(version_cmd=[''], version_regex='', version_regex_replace='\\1', is_executable=False,  ....
```
This part is fine. It is not empty. `is_executable=False` because it is a library and not executable tool.

```
File "/Users/<project>/Developer/<project>/esp-idf/tools/idf_tools.py", line 726, in find_installed_versions
    ver_str = self.get_version()
  File "/Users/<project>/Developer/<project>/esp-idf/tools/idf_tools.py", line 678, in get_version
    return re.sub(self._current_options.version_regex, self._current_options.version_regex_replace, match.group(0))  # type: ignore
```
I don't understand this part. It should not get to `return`. `self.get_version()` should fail and the exception caught leaving the function because there the item is not installed.

dobairoland, 2023-07-21:
@Lapshin PTAL. Maybe I've missed something.

fhrbata, 2023-07-21:
@dobairoland yes, I agree. It seems strange. My understanding is that the `subprocess.run()` should raise EPERM. This should be caught by the OSError in `get_version()` and later skipped in `find_installed_versions()` based on the `ToolNotFound` exception.  Just a dumb idea and I for sure may be missing something: What does this
`python -c "import subprocess; subprocess.run([''])"` return in the same environment. Isn't it possible that something(mosh or maybe something from history) get actally started with this and returns without error. I know it's strange and I cannot see how this could happen, but based on the discussion it seems that it might be somehow possible :). Anyway this is just me thinking out loud. Thank you!

second-string, 2023-07-22:
y i k e s, something is happening where executing that blank `subprocess.run([''])` is trying to launch mosh-server. Even launching a shell with `/bin/bash --noprofile` it's still happening. 

It doesn't matter if I use the built-in python 3.9 from Xcode tools or python 3.10 from homebrew.

Dig a little more digging and found that if mosh-server is anywhere on my PATH (even if it's not the first thing on the path), it gets executed with a blank `subprocess.run` call:

```
Last login: Sat Jul 22 09:03:23 on ttys000
[1001] <name>@<Names>-MacBook-Pro ~/Developer  echo $PATH
/opt/homebrew/opt/llvm/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/sbin:/usr/local/opt/llvm/bin:/Users/<name>/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/opt/homebrew/opt/fzf/bin
[1002] <name>@<Names>-MacBook-Pro ~/Developer  python3 -c "import subprocess; subprocess.run([''])" 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 505, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/Applications/Xcode.app/Contents/Developer/Library/Frameworks/Python3.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
PermissionError: [Errno 13] Permission denied: ''
[1003] <name>@<Names>-MacBook-Pro ~/Developer  
[1003] <name>@<Names>-MacBook-Pro ~/Developer  
[1003] <name>@<Names>-MacBook-Pro ~/Developer  
[1003] <name>@<Names>-MacBook-Pro ~/Developer  export PATH="/opt/homebrew/bin/mosh-server:$PATH"   
[1004] <name>@<Names>-MacBook-Pro ~/Developer  python3 -c "import subprocess; subprocess.run([''])"

MOSH CONNECT 60001 B5YQlWkLYWdWhAY9jJ/CdQ

mosh-server (mosh 1.4.0) [build mosh 1.4.0]
Copyright 2012 Keith Winstein <mosh-devel@mit.edu>
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

[mosh-server detached, pid = 69322]
[1005] <name>@<Names>-MacBook-Pro ~/Developer  
```

Once I remove mosh-server from my path, the download of esp-rom-elfs and the rest of the install.sh and source.sh scripts work fine. Super weird, I'll go ahead and file an issue on the mosh repo.

Thank you all for your help! I'll leave this issue as-is in case you want to add anything more, but feel free to close it.

dobairoland, 2023-07-22:
Thanks for the info. I agree that is very weird.

closed 2023-07-22 by dobairoland

fhrbata, 2023-07-23:
@dobairoland maybe we can consider to not call the `get_version()` if there is no command for it. Just my $0.02

CrazyAndy, 2023-08-02:
modify the {IDF_PATH}/tools/idf_tools.py  at row 666 as follows
`try:
            if(cmd[0]==''):
                return None
            version_cmd_result = run_cmd_check_output(cmd, None, extra_paths)
        except OSError:`
avoid the exception of get_version of  esp-rom-elfs