espressif/esp-idf #11881: Ability to disable timestamps in logging

Devilbinder, 2023-07-15:
### Is your feature request related to a problem?

I am getting task stack overflows after my device has run for a few hours. The source is the timestamp in the `ESP_LOGx` output string. Took me some hours to find out that it is the timestamp that is causing the stack overflow.

### Describe the solution you'd like.

Option in menuconfig to disable timestamps for logging.

The only option available is `CONFIG_LOG_TIMESTAMP_SOURCE_RTOS` and `CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM` thus it forces you to use a growing string size.

### Describe alternatives you've considered.

Using `printf` instead of `ESP_LOGx` as a work around.

### Additional context.

ESP32S3
IDF V5.0.2

My timestamp source is CONFIG_LOG_TIMESTAMP_SOURCE_RTOS

Build system overrides if I remove CONFIG_LOG_TIMESTAMP_SOURCE_RTOS by hand.

The function `esp_log_timestamp` returns a `uint32_t` (4294967295  max) this can result in a string of size of 10+1 if there is no system reboot. My application is intended to run 24/7.

This can cause unforeseen bugs in a production environment. It can take 11.5 days for the string to reach 10 characters in length and a bug to present itself.

nopnop2002, 2023-07-18:
You are free to customize the logging format.

Please refer to this.

https://github.com/nopnop2002/esp-idf-time-logging

KonstantinKondrashov, 2023-07-19:
Hi @Devilbinder!
> The function esp_log_timestamp returns a uint32_t (4294967295 max) this can result in a string of size of 10+1 if there is no system reboot. My application is intended to run 24/7.

The `esp_log_timestamp()` does return the timestamp as uint32_t (4294967295 max). The timestamp is in milliseconds. 
4294967295ms/1000/86400 = 49 days. (not 11.5 days). I agree that 49 days might be not enough if the device is never rebooted. I will work on that to increase it.

What exactly is your case? Please give your sdkconfig file.
If you use CONFIG_LOG_TIMESTAMP_SOURCE_RTOS option the timestamp is handled as uint32_t  it is not converted to a string.
If you use CONFIG_LOG_TIMESTAMP_SOURCE_SYSTEM option then `gettimeofday()` is used to print the time in the format `(00:01:30.000)`. When you use this option it might lead that the stack usage is higher, maybe this leads to the overflow issue. If so then you need to increase the stack for your tasks.

> It can take 11.5 days for the string to reach 10 characters in length and a bug to present itself.

We use a buffer of 18 chars.
```c
char *esp_log_system_timestamp(void)
{
    static char buffer[18] = {0};
...
```

Devilbinder, 2023-07-19:
Hi @KonstantinKondrashov,

My use case is to make the logging predictable over a long period of time. My application is never intended to reboot without manual intervention nor will it have the luxury of OTA. The code must be able to run for the lifetime of the device days, months or years. I intend to OTP the production devices.

Removing the timestamp should save some memory. Adding the leading zeros `%10d` would also work at the cost of memory and exaction time. Both of these should make the logging string predictable. I would prefer removal since it solves my problem at compile time and not runtime also has the benefit of reducing the code footprint.

I am doing QA testing on my code. I am using the serial logging as a watchdog. My application and watchdog simply does not care if the timestamp is there or not. It is a nice to have when debugging the code. 

My concern is that the code has a `error` or `warn` log somewhere that has never printed and will crash the application. 

My logging timestamps are set to  `CONFIG_LOG_TIMESTAMP_SOURCE_RTOS` (default)

Stack size is `2048`. The high watermark is `48`. I do not view increasing the task stack size a solution, merely delays the fault. I did increase it anyways to `4098`.

The bug (in my code) is present regardless if the log level is set to `info` or `warn`. The `printf` stamens replace a `warn` log. Bug is gone if I use `printf`

Code that is the source of the overflow. Simple code. Press and hold a button to delete the config files.

``` C
static QueueHandle_t button_evt_queue = NULL;

void IRAM_ATTR freset(void *parm){
	uint32_t gpio_num = (uint32_t)parm;
	xQueueSendFromISR(button_evt_queue, &gpio_num, NULL);
	gpio_intr_disable((gpio_num_t)gpio_num);
}

void button_task(void* param)
{
	button_evt_queue = xQueueCreate(10, sizeof(uint32_t));
	DIn fReset(BTN);
	fReset.Init(GPIO_PULLUP_ONLY);
	fReset.SetISR(freset, GPIO_INTR_NEGEDGE);

	uint32_t i = 0;
	uint32_t io_num;

	for(;;) {
               //added after stack overflow was detected
		ESP_LOGI(TAG, "TASK: %s, Free stack: %lu",ThisThread::getName(), ThisThread::GetMinFreeStack());

		if(xQueueReceive(button_evt_queue, &io_num, portMAX_DELAY) == pdFALSE){
			continue;
		}

		while(xQueueReceive(button_evt_queue, &io_num, 20) != pdFALSE){
			continue;
		}

		for(i = 0; i < FACTORY_RESET_CNT;i++){
			ThisThread::sleep(1000);

			if(gpio_get_level((gpio_num_t)io_num) != 0){
				gpio_intr_enable(BTN);
				break;
			}
	                //used to be ESP_LOGW(....)
			printf("Factory reset in %ld\n", FACTORY_RESET_CNT-i); // stack overflow here if logging macro is used after some time
		}

		if(i >= 9){
                        //used to be ESP_LOGW(....)
			printf("Factory reset system\n");
			DeviceConfig::FactoryResetFiles();
			//added after stack overflow was detected
			ESP_LOGI(TAG, "TASK: %s, Free stack: %lu",ThisThread::getName(), ThisThread::GetMinFreeStack()); // stack overflows if info level is used
			esp_restart();
			gpio_intr_enable(BTN);
		}

	}
}
```

Printout at overflow:
```
(logging prepend) Factory reset in 10
***ERROR*** TASK:xxxx Stack Overflow
```
This is from memory

> 4294967295ms/1000/86400 = 49 days. (not 11.5 days)

Agreed, was early morning

> If you use CONFIG_LOG_TIMESTAMP_SOURCE_RTOS option the timestamp is handled as uint32_t it is not converted to a string.

Not possible, the RTOS ticks must be converted to a UTF-8 string. You would print gibberish to the console otherwise unless there is a converter in `idf,py monitor` that I am not aware of. I have not looked at the raw output strings in some time.

> We use a buffer of 18 chars.

Not this then. The stack is clearly increasing over time. I may have misinterpreted the function of `esp_log_timestamp`. 

It is extremely convenient to be able to disable and enable parts of the logging framework. Thus my request.

**Please note** that I do not have a debugger so I am unable to dig into the brain.

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/12094766/sdkconfig.zip)

KonstantinKondrashov, 2023-07-19:
FYI, https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#stack-overflow

> Stack size is 2048. The high watermark is 48. I do not view increasing the task stack size a solution, merely delays the fault. I did increase it anyways to 4098.

By default, the ESP_LOGI macro in IDF uses the vprintf function to format and print log messages. The stack size required for this operation can depend on the length and complexity of the log message being printed. Additionally, if the log message contains format specifiers (e.g., %d, %s), the stack usage may increase.
The default stack size for a task in IDF is typically 2048 bytes. However, this may not be sufficient if you have other tasks running concurrently or if the log messages are large and complex. It's generally recommended to provide additional stack space to accommodate any unforeseen increases in stack usage. You can try to mesure it using ESP_LOGI(TAG, "%s stack use", uxTaskGetStackHighWaterMark(NULL));

If you want to disable timestamp you can redefine [esp_log_timestamp()](https://github.com/espressif/esp-idf/blob/master/components/log/log_freertos.c#L99-L110) and return 0. This is the easiest and fastest way to do it. See how to do it [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#wrappers-to-redefine-or-extend-existing-functions).

> Option in menuconfig to disable timestamps for logging.

Regarding this, we will consider implementing this feature. Thanks.

Could you create an example that can be run without changes? Please give logs that you get. I do not think that the issue some where in the log.
Thanks.

Devilbinder, 2023-07-19:
> By default, the ESP_LOGI macro in IDF uses the vprintf function to format and print log messages. The stack size required for this operation can depend on the length and complexity of the log [...]

This refers the user defined strings not IDF defined strings. It should not be the users responsibility to manage the stack for an internally created string. You cannot expect the user to have intrinsic knowledge of how the SDK works.

Users reasonability to manage.
`ESP_LOGI(TAG,"some counter: %d",counter++);`

IDFs reasonability to manage.
```C
#define LOG_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%" PRIu32 ") %s: " format LOG_RESET_COLOR "\n"
#define LOG_SYSTEM_TIME_FORMAT(letter, format)  LOG_COLOR_ ## letter #letter " (%s) %s: " format LOG_RESET_COLOR "\n"
```
https://github.com/espressif/esp-idf/blob/a2d76ad38ad4be46bdc7ce2becab45fbe71ecc48/components/log/include/esp_log.h#L308C115-L309C1

In other words if the user creates a string of arbitrary length and there is not enough stack for the string then the user should allocate more space for it.
If the user creates a string of a fixed length string the `uxTaskGetStackHighWaterMark` should not decrease.

This is my opinion.

 > If you want to disable timestamp you can redefine [esp_log_timestamp()](https://github.com/espressif/esp-idf/blob/master/components/log/log_freertos.c#L99-L110) and return 0. This is the easiest and fastest way to do it. See how to do it [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#wrappers-to-redefine-or-extend-existing-functions).
 
 Thanks this is better than using `printf`.
 
 Thank you for considering the feature.

Devilbinder, 2023-07-19:
If anyone is interested here is a gist of how to override the timestamps
https://gist.github.com/Devilbinder/fe7ebdeb8505265562741fb80ed86243

KonstantinKondrashov, 2023-07-20:
@Devilbinder 

> xTaskCreate(log_panic_task,"panic", 1792 + (64) + (32) + (16), NULL, 5, NULL);

1904 is a small size for task that uses printf APIs. Please have a look at the recommendations about setting it here https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#internal-stack-sizes, https://www.esp32.com/viewtopic.php?t=4295

I close the issue because it is not related to the log issue anymore. Please follow the recommendation of setting the proper stack size. Feel free to open a new issue if any issues with the setting of stack size.

Option in menuconfig to disable timestamps for logging. Will be implemented in a further release (5.3 or above).
Thanks.

closed 2023-07-20 by KonstantinKondrashov