espressif/esp-idf #11884: esp_netif_sntp errors

dannybackx, 2023-07-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect 5.1 documentation examples to compile without error.
I see that ip_event_to_renew is initialized to 0 while it should be an enum.
No clue which of the values to take, the first one is IP_EVENT_STA_GOT_IP and that doesn't seem appropriate.

Note that the rest of the error message below appears to indicate a second bug.

### What is the actual behavior?

/home/danny/src/esp-idf/esp32s3-4827s043/minimal/main/App.cpp: In static member function 'static void App::connect(void*, esp_event_base_t, int32_t, void*)':
/home/danny/src/github/esp32/esp-idf-v5.1/components/esp_netif/include/esp_netif_sntp.h:53:34: error: invalid conversion from 'int' to 'ip_event_t' [-fpermissive]
   53 |             .ip_event_to_renew = 0,                 \
      |                                  ^
      |                                  |
      |                                  int
/home/danny/src/github/esp32/esp-idf-v5.1/components/esp_netif/include/esp_netif_sntp.h:63:13: note: in expansion of macro 'ESP_NETIF_SNTP_DEFAULT_CONFIG_MULTIPLE'
   63 |             ESP_NETIF_SNTP_DEFAULT_CONFIG_MULTIPLE(1, {server})
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/danny/src/esp-idf/esp32s3-4827s043/minimal/main/App.cpp:87:28: note: in expansion of macro 'ESP_NETIF_SNTP_DEFAULT_CONFIG'
   87 |   esp_sntp_config_t sntp = ESP_NETIF_SNTP_DEFAULT_CONFIG(CONFIG_NTP_SERVER_0);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

### Steps to reproduce.

from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html?highlight=sntp_init#_CPPv49sntp_initv

it says use this :
esp_sntp_config_t config = ESP_NETIF_SNTP_DEFAULT_CONFIG("pool.ntp.org");
esp_netif_sntp_init(&config);

david-cermak, 2023-07-28:
Duplicate of https://github.com/espressif/esp-idf/issues/11825#issuecomment-1655069735

Sorry for the trouble, will fix.

dannybackx, 2023-09-12:
Note : still present in esp-idf-v5.1.1 :-(

AxelLin, 2023-10-03:
fixed by ef451cab0fec03b24d71552054c8489c2bd6217b

david-cermak, 2023-10-26:
Fixed in https://github.com/espressif/esp-idf/commit/597810bf96f90d93a98cc8641f90e5bd09a75ff0

closed 2023-10-26 by david-cermak