espressif/esp-idf #11885: esp 32c2 Run example: gatt_server reports errors controller rom commit: [3314f9d], controller lib commit: [628b93c].

LHYYSY, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 c2 -12

### Power Supply used.

USB

### What is the expected behavior?

Run the gatt server routine。

### What is the actual behavior?

esp 32c2 Run example: gatt_server reports errors controller rom commit: [3314f9d], controller lib commit: [628b93c].

### Steps to reproduce.

I (21) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (21) boot: compile time 12:12:25
I (21) boot: chip revision: v1.0
I (23) boot.esp32c2: MMU Page Size  : 32K
I (28) boot.esp32c2: SPI Speed      : 60MHz
I (33) boot.esp32c2: SPI Mode       : DIO
I (38) boot.esp32c2: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length  
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c078020 size=26c78h (158840) map
I (129) esp_image: segment 1: paddr=00036ca0 vaddr=3fcae020 size=01378h (  4984) load
I (131) esp_image: segment 2: paddr=00038020 vaddr=42000020 size=7622ch (483884) map
I (243) esp_image: segment 3: paddr=000ae254 vaddr=3fcaf398 size=006e0h (  1760) load
I (244) esp_image: segment 4: paddr=000ae93c vaddr=40380000 size=0e01ch ( 57372) load
I (268) boot: Loaded app from partition at offset 0x10000
I (268) boot: Disabling RNG early entropy source...
I (280) cpu_start: Pro cpu up.
I (288) cpu_start: Pro cpu start user code
I (288) cpu_start: cpu freq: 120000000 Hz
I (288) cpu_start: Application information:
I (291) cpu_start: Project name:     gatt_server_demos
I (297) cpu_start: App version:      1
I (302) cpu_start: Compile time:     Jul 15 2023 12:11:23
I (308) cpu_start: ELF file SHA256:  27c8aeec5fb27459...
I (314) cpu_start: ESP-IDF:          v5.0.1-dirty
I (319) cpu_start: Min chip rev:     v1.0
I (324) cpu_start: Max chip rev:     v1.99 
I (329) cpu_start: Chip rev:         v1.0
I (333) heap_init: Initializing. RAM available for dynamic allocation:
I (340) heap_init: At 3FCB3E20 len 00028D50 (163 KiB): D/IRAM
I (347) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (355) spi_flash: detected chip: generic
I (358) spi_flash: flash io: dio
I (363) coexist: coexist rom version de8c800
I (367) cpu_start: Starting scheduler.
I (392) phy_init: phy_version 210,e134770,Oct 20 2022,15:26:13
controller lib commit: [628b93c]
controller rom commit: [3314f9d]
I (412) system_api: Base MAC address is not set
I (412) system_api: read default base MAC address from EFUSE
I (442) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (442) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (442) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (452) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (462) GATTS_DEMO: the gatts demo char length = 3

I (462) GATTS_DEMO: prf_char[0] =11

I (472) GATTS_DEMO: prf_char[1] =22

I (472) GATTS_DEMO: prf_char[2] =33

I (472) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (482) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (492) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (502) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (502) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (512) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

### Debug Logs.

```plain
I (21) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (21) boot: compile time 12:12:25
I (21) boot: chip revision: v1.0
I (23) boot.esp32c2: MMU Page Size  : 32K
I (28) boot.esp32c2: SPI Speed      : 60MHz
I (33) boot.esp32c2: SPI Mode       : DIO
I (38) boot.esp32c2: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length  
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=00010020 vaddr=3c078020 size=26c78h (158840) map
I (129) esp_image: segment 1: paddr=00036ca0 vaddr=3fcae020 size=01378h (  4984) load
I (131) esp_image: segment 2: paddr=00038020 vaddr=42000020 size=7622ch (483884) map
I (243) esp_image: segment 3: paddr=000ae254 vaddr=3fcaf398 size=006e0h (  1760) load
I (244) esp_image: segment 4: paddr=000ae93c vaddr=40380000 size=0e01ch ( 57372) load
I (268) boot: Loaded app from partition at offset 0x10000
I (268) boot: Disabling RNG early entropy source...
I (280) cpu_start: Pro cpu up.
I (288) cpu_start: Pro cpu start user code
I (288) cpu_start: cpu freq: 120000000 Hz
I (288) cpu_start: Application information:
I (291) cpu_start: Project name:     gatt_server_demos
I (297) cpu_start: App version:      1
I (302) cpu_start: Compile time:     Jul 15 2023 12:11:23
I (308) cpu_start: ELF file SHA256:  27c8aeec5fb27459...
I (314) cpu_start: ESP-IDF:          v5.0.1-dirty
I (319) cpu_start: Min chip rev:     v1.0
I (324) cpu_start: Max chip rev:     v1.99 
I (329) cpu_start: Chip rev:         v1.0
I (333) heap_init: Initializing. RAM available for dynamic allocation:
I (340) heap_init: At 3FCB3E20 len 00028D50 (163 KiB): D/IRAM
I (347) heap_init: At 3FCDCB70 len 0000294C (10 KiB): STACK/DRAM
I (355) spi_flash: detected chip: generic
I (358) spi_flash: flash io: dio
I (363) coexist: coexist rom version de8c800
I (367) cpu_start: Starting scheduler.
I (392) phy_init: phy_version 210,e134770,Oct 20 2022,15:26:13
controller lib commit: [628b93c]
controller rom commit: [3314f9d]
I (412) system_api: Base MAC address is not set
I (412) system_api: read default base MAC address from EFUSE
I (442) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 0

I (442) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 40

I (442) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 40

I (452) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40

I (462) GATTS_DEMO: the gatts demo char length = 3

I (462) GATTS_DEMO: prf_char[0] =11

I (472) GATTS_DEMO: prf_char[1] =22

I (472) GATTS_DEMO: prf_char[2] =33

I (472) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (482) GATTS_DEMO: REGISTER_APP_EVT, status 0, app_id 1

I (492) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (502) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (502) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (512) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44
```

zhaoweiliang2021, 2023-07-19:
@LHYYSY  Based on the provided log information, the functionality appears to be normal, and no abnormal issues have been identified. Please help indicate the specific log points during the time period you consider to be abnormal, to facilitate further investigation.

LHYYSY, 2023-07-22:
thank you very much for your reply, this issue has been resolved

closed 2023-07-22 by LHYYSY