espressif/esp-idf #11886: esp32c3 examples\wifi\getting_started\station build fail

goog, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

build success

### What is the actual behavior?

```bash
t/riscv -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/wear_levelling/include -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/sdmmc/include -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/fatfs/diskio -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/fatfs/vfs -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/fatfs/src -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/idf_test/include -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/idf_test/include/esp32c3 -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/ieee802154/include -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/json/cJSON -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/mqtt/esp-mqtt/include -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/spiffs/include -IC:/Espressif/frameworks/esp-idf-v5.0.2/components/wifi_provisioning/include -march=rv32imc  -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fmacro-prefix-map=F:/esp-idf/examples/wifi/getting_started/station=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v5.0.2=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0.2-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/station_example_main.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\station_example_main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/station_example_main.c.obj -c F:/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c
F:/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c: In function 'wifi_init_sta':
F:/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:131:14: error: 'wifi_sta_config_t' has no member named 'sae_h2e_identifier'
  131 |             .sae_h2e_identifier = EXAMPLE_H2E_IDENTIFIER,
      |              ^~~~~~~~~~~~~~~~~~
F:/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:131:35: warning: initialization of 'unsigned char' from 'char *' makes integer from pointer without a cast [-Wint-conversion]
  131 |             .sae_h2e_identifier = EXAMPLE_H2E_IDENTIFIER,
      |                                   ^~~~~~~~~~~~~~~~~~~~~~
F:/esp-idf/examples/wifi/getting_started/station/main/station_example_main.c:131:35: note: (near initialization for 'wifi_config.sta.failure_retry_cnt')
[0/1] Re-running CMake...-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file F:/esp-idf/examples/wifi/getting_started/station/sdkconfig
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- App "bootloader" version: v5.0.2-dirty
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/main/ld/esp32c3/bootloader.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/main/ld/esp32c3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.0.2/components/efuse C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_common C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_system C:/Espressif/frameworks/esp-idf-v5.0.2/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.0.2/components/freertos C:/Espressif/frameworks/esp-idf-v5.0.2/components/hal C:/Espressif/frameworks/esp-idf-v5.0.2/components/log C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/main C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/components/micro-ecc C:/Espressif/frameworks/esp-idf-v5.0.2/components/newlib C:/Espressif/frameworks/esp-idf-v5.0.2/components/partition_table C:/Espressif/frameworks/esp-idf-v5.0.2/components/riscv C:/Espressif/frameworks/esp-idf-v5.0.2/components/soc C:/Espressif/frameworks/esp-idf-v5.0.2/components/spi_flash
-- Configuring done
-- Generating done
-- Build files have been written to: F:/esp-idf/examples/wifi/getting_started/station/build/bootloader
[1/1] cmd.exe /C "cd /D F:\esp-idf\examples\wifi\getting_started\station\build\bootloader\esp-idf\esptool_py && C:\Es...s.py --offset 0x8000 bootloader 0x0 F:/esp-idf/examples/wifi/getting_started/station/build/bootloader/bootloader.bin"Bootloader binary size 0x4ee0 bytes. 0x3120 bytes (38%) free.
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the F:\esp-idf\examples\wifi\getting_started\station\build\log\idf_py_stderr_output_924 and F:\esp-idf\examples\wifi\getting_started\station\build\log\idf_py_stdout_output_924
PS F:\esp-idf\examples\wifi\getting_started\station> idf.py --version
C:\Espressif\frameworks\esp-idf-v5.0.2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
ESP-IDF v5.0.2-dirty
```

### Steps to reproduce.

1. idf.py build

Xiehanxin, 2023-07-17:
hi, @goog , from the log I think you use the ESP32-C3, IDF v5.0.2, station example. 
And you got a build error because the _sae_h2e_identifier_ is not defined in V5.0.2. I want to know if you add this by yourself? This parameter is defind in later version, so you can try latest version and build.

igrr, 2023-07-17:
Or alternatively, if you want to use v5.0.2, please use the example from that version of IDF, not from the master branch of the repository.

goog, 2023-07-18:
@igrr i get it. i was using the example from master branch

closed 2023-07-18 by goog