espressif/esp-idf #11887: build softap_sta example fail on win10 linux ok

goog, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.2-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

build ok

### What is the actual behavior?

add REQUIRES  esp_netif does not work

### Steps to reproduce.

1.  idf.py set-target esp32c3  
2.  idf.py build  

ps. 我现在的最新的windows离线安装包的

### Build or installation Logs.

```plain
cro-prefix-map=C:/Espressif/frameworks/esp-idf-v5.0.2=/IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v5.0.2-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/softap_sta.c.obj -MF esp-idf\main\CMakeFiles\__idf_main.dir\softap_sta.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/softap_sta.c.obj -c C:/Espressif/frameworks/esp-idf-v5.0.2/soft/main/softap_sta.c
C:/Espressif/frameworks/esp-idf-v5.0.2/soft/main/softap_sta.c: In function 'app_main':
C:/Espressif/frameworks/esp-idf-v5.0.2/soft/main/softap_sta.c:238:9: error: implicit declaration of function 'esp_netif_napt_enable' [-Werror=implicit-function-declaration]
  238 |     if (esp_netif_napt_enable(esp_netif_ap) != ESP_OK) {
      |         ^~~~~~~~~~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
[811/820] Performing configure step for 'bootloader'-- Found Git: C:/Espressif/tools/idf-git/2.39.2/cmd/git.exe (found version "2.39.2.windows.1")
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-v5.0.2/soft/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- App "bootloader" version: v5.0.2-dirty
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/main/ld/esp32c3/bootloader.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/main/ld/esp32c3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.0.2/components/efuse C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_common C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.0.2/components/esp_system C:/Espressif/frameworks/esp-idf-v5.0.2/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.0.2/components/freertos C:/Espressif/frameworks/esp-idf-v5.0.2/components/hal C:/Espressif/frameworks/esp-idf-v5.0.2/components/log C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/main C:/Espressif/frameworks/esp-idf-v5.0.2/components/bootloader/subproject/components/micro-ecc C:/Espressif/frameworks/esp-idf-v5.0.2/components/newlib C:/Espressif/frameworks/esp-idf-v5.0.2/components/partition_table C:/Espressif/frameworks/esp-idf-v5.0.2/components/riscv C:/Espressif/frameworks/esp-idf-v5.0.2/components/soc C:/Espressif/frameworks/esp-idf-v5.0.2/components/spi_flash
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Espressif/frameworks/esp-idf-v5.0.2/soft/build/bootloader
ninja: build stopped: subcommand failed.
HINT: Please check that the function name is correct. Also it is possible that you've forgot to import esp_netif_napt_enable library(s) in header file or add the necessary REQURIES component. Try to add missing libraries to your project header file or check idf_component_register(REQUIRES ...) section in your component CmakeList.txt file. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the function has been removed, renamed or replaced by an alternative function - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.0.2\soft\build\log\idf_py_stderr_output_3428 and C:\Espressif\frameworks\esp-idf-v5.0.2\soft\build\log\idf_py_stdout_output_3428
PS C:\Espressif\frameworks\esp-idf-v5.0.2\soft> idf.py --version
C:\Espressif\frameworks\esp-idf-v5.0.2\tools\check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
ESP-IDF v5.0.2-dirty
PS C:\Espressif\frameworks\esp-idf-v5.0.2\soft> git describe --tags
v5.0.2
PS C:\Espressif\frameworks\esp-idf-v5.0.2\soft>
```

espressif-abhikroy, 2023-07-24:
Hi,
esp_netif_napt_enable() can be found in the latest lwip version 2.1.3-esp. To address the issue you are facing, please follow these steps:
1. Navigate to the directory using the command: ```cd ${IDF_PATH}/components/lwip/lwip```
2. Switch to the version 2.1.3-esp using the command: ```git checkout 2.1.3-esp```
Once these steps are completed, attempt to recompile the softap_sta example.

espressif-abhikroy, 2023-07-25:
Hi,
The softap_sta example is designed to work on IDF release v5.1 and later versions. However, if you're unable to upgrade to IDF v5.1, you can try an alternative approach by upgrading ```lwip``` to version ```2.1.3-esp```. Along with this, make the following changes to the lines in the softap_sta.c file:

Old code:
```c
if (esp_netif_napt_enable(esp_netif_ap) != ESP_OK) {
    ESP_LOGE(TAG_STA, "NAPT not enabled on the netif: %p", esp_netif_ap);
}
```

New code:
```c
struct netif *lwip_netif = esp_netif_get_netif_impl(esp_netif_ap);
if (!ip_napt_enable_netif(lwip_netif, 1)) {
    ESP_LOGE(TAG_STA, "NAPT not enabled on the netif: %p", esp_netif_ap);
}
```
By making these changes, the softap_sta example should function properly even if you are unable to upgrade to IDF v5.1.

goog, 2023-07-26:
:star2:

closed 2023-07-26 by goog

JeremyPasquiet, 2023-10-06:
thank you so much ! Worked for me