espressif/esp-idf #11888: Cannot enable flash encryption after secure-boot v2

FnxQT, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-dirty

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Rev.3 - Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I enabled secure boot v2 and it worked well. Now I would like to enable flash encryption with an external key so that FW are already encrypted when they're distributed to the devices.

### What is the actual behavior?

The "espefuse.py" tool does not let me burn the flash encryption key. It says :

```
espefuse.py v4.5.1
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.

Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.

Detecting chip type... ESP32

=== Run "burn_key" command ===
Burn keys to blocks:
 - BLOCK1 -> [Key bytes]
        Reversing the byte order
        Disabling read to key block

A fatal error occurred: This efuse cannot be read-disabled due the to RD_DIS field is already write-disabled
```

However when running the summary command, nothing is written to the BLOCK1 used for the flash encryption key :

```
espefuse.py v4.5.1
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.
.
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.

Detecting chip type... ESP32

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1107 R/- (0b00001)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 56 R/W (0x38)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 257 R/W (0x0101)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 0 R/- (0x0)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme                
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address                               
   = 78:21:84:a5:c4:78 (CRC 0xad OK) R/W 
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 173 R/W (0xad)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = True R/W (0b1)
WAFER_VERSION_MINOR (BLOCK0):                      WAFER VERSION MINOR                                = 0 R/W (0b00)
CHIP_PACKAGE (BLOCK0):                             Chip package identifier                            = 0 R/W (0b000)
CHIP_PACKAGE_4BIT (BLOCK0):                        Chip package identifier #4bit                      = 0 R/W (0b0)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = 0 R/W (0x00)
WAFER_VERSION_MAJOR (BLOCK0):                      calc WAFER VERSION MAJOR from CHIP_VER_REV1 and CH = 3 R/W (0b011)
                                                   IP_VER_REV2 and apb_ctl_date (read only)          
PKG_VERSION (BLOCK0):                              calc Chip package = CHIP_PACKAGE_4BIT << 3 + CHIP_ = 0 R/W (0x0)
                                                   PACKAGE (read only)                               

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = True R/W (0b1)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                                   Flash encryption key                              
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 
BLOCK2 (BLOCK2):                                   Secure boot key                                   
   = Secure boot key bytes R/- 
BLOCK3 (BLOCK3):                                   Variable Block 3                                  
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Flash voltage (VDD_SDIO) determined by GPIO12 on reset
```

### Steps to reproduce.

1. Generate a secure boot key : espsecure.py generate_signing_key --version 2 sb_private_key.pem
2. Export the public key : openssl rsa -in sb_private_key.pem -pubout > sb_public_key.pem
3. Burn the digest in efuse : espefuse.py burn_key_digest sb_public_key.pem
4. Generate a flash encryption key digest : espsecure.py generate_flash_encryption_key --keylen 256 flash_enc_key.bin
5. Flash the  digest : espefuse.py --port rfc2217://host.docker.internal:4005?ign_set_control burn_key flash_encryption flash_enc_key.bin

I get the error at step number 5 :

```
espefuse.py v4.5.1
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.

Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Device PID identification is only supported on COM and /dev/ serial ports.

Detecting chip type... ESP32

=== Run "burn_key" command ===
Burn keys to blocks:
 - BLOCK1 -> [Key bytes]
        Reversing the byte order
        Disabling read to key block

A fatal error occurred: This efuse cannot be read-disabled due the to RD_DIS field is already write-disabled
```

### More Information.

Do I make something wrong ? Is there another way to generate keys and flash ? I found the documentation about secure boot v2 and flash encryption a bit lacking because I did not find any information about espefuse.py burn_key_digest sb_public_key.pem which seems to be the command to be used for flashing the secure_boot v2 key digest.

AdityaHPatwardhan, 2023-07-17:
Hi @FnxQT this is a Secure Boot limitation.
It has been listed in [Secure boot limitations](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html#restrictions-after-secure-boot-is-enabled)

The reason is as follows:
1) The Secure Boot digest needs to be kept as software readable and must not be read protected. If that digest is accidentally write protected then secure boot would not longer work on that device and would render the device useless.

To avoid this scenario, when Secure Boot is enabled, the ability to read protect any key is also disabled.
But the Flash Encryption feature requires the FE key present in the eFuse to be read protected otherwise Flash Encryption is not safe.

For this reason, Once Secure Boot is enabled on a device, Flash Encryption cannot be enabled on the same device afterwards. ( Unless Secure Boot is first enabled in Insecure Mode).

Due to this you are not able to burn the flash encryption key with default command which also enables write protection for the key.

I think you would still be able to test the FE feature on your device (though it would not be secure as the key is not read protected). By following the steps in [Enabling flash encryption externally](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-flash-encryption-externally)

with changing the key burning command from
```espefuse.py --port rfc2217://host.docker.internal:4005?ign_set_control burn_key flash_encryption flash_enc_key.bin```
to
```espefuse.py --port rfc2217://host.docker.internal:4005?ign_set_control burn_key flash_encryption flash_enc_key.bin --no-protect-key```

FnxQT, 2023-07-17:
Hi @AdityaHPatwardhan, 

Thank you for your help. Ok I understand my fault. However the message in Secure Boot limitations is not very clear. It should be written something like : Flash Encryption cannot be enabled after Secure Boot v2 so enable it before...

However, something even stranger happened. By using espefuse.py locally, (not through rfc2217) I could actually burn the key without the --no-protect-key option.

But, it is still not working. Upon booting I get this error:

`E (1412) flash_encrypt: Flash encryption eFuse bit was not enabled in bootloader but CONFIG_SECURE_FLASH_ENC_ENABLED is on`

I tried to burn one bit of FLASH_CRYPT_CNT but it had no effect. I got an error message directly from the bootloader :

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x6c376833
invalid header: 0x6c376833
invalid header: 0x6c376833
invalid header: 0x6c376833
invalid header: 0x6c376833
invalid header: 0x6c376833
invalid header: 0x6c376833
invalid header:�ets Jul 29 2019 12:21:46
```

Is all of this related to the fact I did not enable flash encryption before secure boot ?

AdityaHPatwardhan, 2023-07-17:
You need to write the encrypted binaries on the flash of the esp32.
Please follow step 5 of the same [document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-flash-encryption-externally)

FnxQT, 2023-07-17:
Ok thank you. This document is very clear I did not find it previously. 

Last question, the workflow seem to be a bit strange right now. Secure Boot v2 is based on RSA-PSS using an RSA:3072 key. However, using this key I cannot generate the bootloader secure digest as it only accepts AES keys:

```
espsecure.py digest_secure_bootloader --keyfile sign/sb_private_key.pem --output bootloader_reflash_
digest.bin build/bootloader/bootloader-signed.bin
espsecure.py v4.5.1

A fatal error occurred: Key file contains wrong length (2459 bytes), 16, 24, 32 or 64 expected.
```

How to generate this digest using an RSA key ? Is there a document like the one you showed but for Secure Boot V2 ? It seems to me that this digest is only used for reflashing the bootloader but maybe I misunderstood something.

Thank you a lot again :)

FnxQT, 2023-07-17:
Well I bricked a second device. I think the documentation is very unclear. Here is what I did :

```
#Burn required efuse
espefuse.py --port rfc2217://host.docker.internal:4008?ign_set_control burn_key flash_encryption flash_enc_key.bin
espefuse.py --port rfc2217://host.docker.internal:4008?ign_set_control burn_key_digest sb_public_key.pem
espefuse.py --port rfc2217://host.docker.internal:4010?ign_set_control --chip esp32 burn_efuse FLASH_CRYPT_CNT 127
espefuse.py --port rfc2217://host.docker.internal:4008?ign_set_control --chip esp32 burn_efuse FLASH_CRYPT_CONFIG 0xF
espefuse.py --port rfc2217://host.docker.internal:4008?ign_set_control burn_efuse ABS_DONE_1 1

#Build the app
idf.py clean
idf.py build

#Encrypt every partitions
espsecure.py encrypt_flash_data --keyfile sign/flash_enc_key.bin --address 0x1000 --output build/bootloader/bootloader-enc.bin build/bootloader/bootloader.bin
espsecure.py encrypt_flash_data --keyfile sign/flash_enc_key.bin --address 0xc000 --output build/partition_table/partition-table-enc.bin build/partition_table/partition-table.bin
espsecure.py encrypt_flash_data --keyfile sign/flash_enc_key.bin --address 0xd000 --output build/ota_data_initial-enc.bin build/ota_data_initial.bin
espsecure.py encrypt_flash_data --keyfile sign/flash_enc_key.bin --address 0x10000 --output build/app-enc.bin build/app.bin

#Flash bootloader
esptool.py --chip esp32 --port rfc2217://host.docker.internal:4010?ign_set_control -b 460800 --before=default_reset --after=no_reset write_flash --flash_mode dio --flash_freq 40m --flash_size keep 0x1000 /build/bootloader/bootloader-enc.bin

#Flash other partitions
esptool.py --port rfc2217://host.docker.internal:4010?ign_set_control -b 460800 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size keep --flash_freq 40m 0xc000 build/partition_table/partition-table-enc.bin 0xd000 build/ota_data_initial-enc.bin 0x10000 build/app-enc.bin
```

I end with an error message saving that the signature block of the bootloader is not found:

`No signature block magic byte found at signature sector (found 0xaa not 0xe7). Image not V2 signed?`

But when I check it with espsecure.py it is ok :

```
espsecure.py v4.5.1
Signature block 0 is valid (RSA).
Public key digest for block 0: 84 53 80 e3 92 5d 50 22 9f ec 26 79 e0 8c a1 7e 96 ba 5b c3 5f a6 08 d5 87 4d ac e1 5c a2 1f 16
Signature block 1 absent/invalid. Skipping checking next blocks.
```

However, if I check the encrypted bootloader it is not ok :

```
espsecure.py v4.5.1
Signature block 0 absent/invalid. Skipping checking next blocks.
```
In the document you provided they encrypt the bootloader at step 5.

For instance here is my security config :

```
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
CONFIG_SECURE_BOOT_V1_SUPPORTED=y
CONFIG_SECURE_BOOT_V2_PREFERRED=y
CONFIG_SECURE_BOOT_V2_RSA_ENABLED=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
# CONFIG_SECURE_BOOT_V1_ENABLED is not set
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="sign/sb_private_key.pem"
# CONFIG_SECURE_BOOTLOADER_MODE=SECURE_BOOTLOADER_REFLASHABLE
# CONFIG_SECURE_BOOTLOADER_REFLASHABLE=y
# CONFIG_SECURE_BOOT_INSECURE=y
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_MODE=SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT
# CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
CONFIG_SECURE_INSECURE_ALLOW_DL_MODE=y
# end of Security features
```

I'm at a loss, what do I do wrong ?

AdityaHPatwardhan, 2023-07-18:
Hi @FnxQT 

> How to generate this digest using an RSA key ? Is there a document like the one you showed but for Secure Boot V2 ? It seems to me that this digest is only used for reflashing the bootloader but maybe I misunderstood something.

Are you using the documentation for secure boot V1 by any chance when secure boot v2 is enabled for your device.
You can find the Secure Boot V2 documentation at [this link](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html) 

I dont think you need to generate the digest once again. digest needs to be generated only when you want to burn it into the eFuse, as you have already done that I think simply signing the data using the steps mentioned [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html#signing-using-espsecure-py) should be sufficient.

Just to answer your question, To generate a digest for the secure boot v2 there is a different command.
you need to use `digest_sbv2_public_key` command to generate the secure boot digest.

I dont think the first device is bricked as of yet. It should still work

AdityaHPatwardhan, 2023-07-18:
For the second device, can you share the `efuse summary`?

FnxQT, 2023-07-18:
Hi @AdityaHPatwardhan,

Effectively, I read the v2 documentation but I understood something wrong. I though Secure Boot v2 still a reflashable mode like in v1 where a digest of bootloader + private key could be generated to reflash the bootloader.

For the first device I could try to reflash the encrypted partition table, otadata and app.bin.

Still, should ALL partitions be encrypted before putting them on the device ? Also the bootloader ? 

Here the efuse summary of the second device :

```
Detecting chip type... ESP32

=== Run "summary" command ===
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1156 R/W (0b01000)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 56 R/W (0x38)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 384 R/W (0x0180)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 1 R/W (0x1)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme                
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address                               
   = 78:21:84:a5:c4:68 (CRC 0x30 OK) R/W 
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 48 R/W (0x30)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = True R/W (0b1)
WAFER_VERSION_MINOR (BLOCK0):                      WAFER VERSION MINOR                                = 0 R/W (0b00)
CHIP_PACKAGE (BLOCK0):                             Chip package identifier                            = 0 R/W (0b000)
CHIP_PACKAGE_4BIT (BLOCK0):                        Chip package identifier #4bit                      = 0 R/W (0b0)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = 0 R/W (0x00)
WAFER_VERSION_MAJOR (BLOCK0):                      calc WAFER VERSION MAJOR from CHIP_VER_REV1 and CH = 3 R/W (0b011)
                                                   IP_VER_REV2 and apb_ctl_date (read only)          
PKG_VERSION (BLOCK0):                              calc Chip package = CHIP_PACKAGE_4BIT << 3 + CHIP_ = 0 R/W (0x0)
                                                   PACKAGE (read only)                               

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 1 R/W (0b0000001)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = True R/W (0b1)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = False R/W (0b0)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                                   Flash encryption key                              
[... cut ...]
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W 

Flash voltage (VDD_SDIO) determined by GPIO12 on reset 
```

And the exact message it prints :

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:2, clock div:2
secure boot v2 enabled
No signature block magic byte found at signature sector (found 0xaa not 0xe7). Image not V2 signed?
secure boot verification failed
ets Jul 29 2019 12:21:46
```

AdityaHPatwardhan, 2023-07-18:
For the second device, 
As Flash Encryption and Secure Boot V2 both are enabled in eFuse.
You will have to make sure that
1) Bootloader is signed with the correct key.
2) Secure Boot v2 is enabled in menuconfig while building the bootloader
3) Application is signed with the correct key.

As a part of flash encryption.
The following binaries must be encrypted -
1) bootloader.bin 2) partition-table.bin 3) application.bin

Also, about the issue where the signature of the encrypted binary could not be verified. Yes as the binary is encrypted, espsecure.py cannot decrypt the binary and parse the signature so it would just exit with failure. You can only check the signature of a non-encrypted binary (which you have done in previously)

AdityaHPatwardhan, 2023-07-18:
@FnxQT For testing security features on esp32, you can also use the https://github.com/espressif/qemu  project.
You can emulate the esp32 chip using qemu without worrying about bricking the chip.

FnxQT, 2023-07-18:
@AdityaHPatwardhan 

Nice, the qemu thing is really interesting. However my FW still produces an error :

```
Adding SPI flash device
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff00b8,len:11492
load:0x40078000,len:23948
load:0x40080400,len:3496
entry 0x400805f0
W (3513) secure_boot: UART ROM Download mode kept enabled - SECURITY COMPROMISED
W (3947) flash_encrypt: Not disabling UART bootloader encryption
**E (24116) esp_image: image at 0x380000 has invalid magic byte (nothing flashed here?)
ets Jul 29 2019 12:21:46**

rst:0x3 (SW_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:2, clock div:2
secure boot v2 enabled
secure boot verification succeeded
load:0x3fff00b8 len:0x2ce4
load:0x40078000 len:0x5d8c
load:0x40080400 len:0xda8
entry 0x400805f0
I (11123) quad_psram: This chip is ESP32-D0WD
I (11154) esp_psram: Found 4MB PSRAM device
I (11155) esp_psram: Speed: 40MHz
I (11156) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
I (11193) cpu_start: Pro cpu up.
I (11196) cpu_start: Starting app cpu, entry point is 0x40081648
I (31133) cpu_start: App cpu up.
I (11407) esp_psram: SPI SRAM memory test OK
D (11431) clk: RTC_SLOW_CLK calibration value: 3495245
I (2230) cpu_start: Pro cpu start user code
I (2231) cpu_start: cpu freq: 240000000 Hz
I (2233) cpu_start: Application information:
I (2234) cpu_start: Project name:     DINGZ-GW
I (2235) cpu_start: App version:      v5.0.2-dirty
I (2236) cpu_start: Compile time:     Jul 18 2023 06:39:58
I (2238) cpu_start: ELF file SHA256:  5defc22551c051a5...
I (2240) cpu_start: ESP-IDF:          v5.0.2-dirty
I (2242) cpu_start: Min chip rev:     v3.0
I (2252) cpu_start: Max chip rev:     v3.99
I (2255) cpu_start: Chip rev:         v3.0
D (2265) memory_layout: Checking 8 reserved memory ranges:
D (2267) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
D (2270) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (2271) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb6688
D (2272) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (2274) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (2275) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (2277) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (2278) memory_layout: Reserved memory range 0x40080000 - 0x40090858
D (2280) memory_layout: Building list of available memory regions:
D (2282) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (2284) memory_layout: Available memory region 0x3ffb6688 - 0x3ffb8000
D (2285) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
D (2286) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
D (2287) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
D (2289) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
D (2290) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
D (2291) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
D (2292) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
[... 55 lines cut ...]
D (2689) cpu_start: calling init function: 0x40105c58
D (2704) cpu_start: calling init function: 0x400d47bc
D (2706) cpu_start: calling init function: 0x400d3958
D (2707) cpu_start: calling init function: 0x400d7988 on core: 0
D (2720) intr_alloc: Connected src 17 to int 3 (cpu 0)
D (2723) cpu_start: calling init function: 0x4016a118 on core: 0
D (2737) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (2739) cpu_start: Starting scheduler on PRO CPU.
D (26) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (29) cpu_start: Starting scheduler on APP CPU.
D (2831) heap_init: New heap initialised at 0x3ffe0440
D (2834) heap_init: New heap initialised at 0x3ffe4350
I (2835) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
D (2840) esp_psram: Allocating block of size 32768 bytes
D (2849) intr_alloc: Connected src 16 to int 12 (cpu 0)
D (2855) MAIN: FW Version: 1.0.17
I (5172) nvs: NVS key partition empty, generating keys
```

The line ```
E (24116) esp_image: image at 0x380000 has invalid magic byte (nothing flashed here?)
ets Jul 29 2019 12:21:46
```

Is it normal or should I worry about ? I'm using the qemu emulatation and my emulated device boots correctly. But produces this error at first.

FnxQT, 2023-07-18:
Another strange is that my emulated device does not display :
```

I (1022) boot: Loaded app from partition at offset 0x10000
I (1022) boot: Checking secure boot...
I (1022) secure_boot: bootloader secure boot is already enabled, continuing..
I (1023) boot: Checking flash encryption...
I (1023) flash_encrypt: flash encryption is enabled (0 plaintext flashes left)
```

On another device using secure boot and flash encryption (that I did not develop) this is outputed. Is it normal that mine does not output anything about this ?

FnxQT, 2023-07-18:
I figured out that this was because I did not include the second app partition in the merge_bin command. Now it is integrated and all partitions are encrypted correctly.

Is it necessary to put both partitions at the beginning ?

FnxQT, 2023-07-18:
Thank you. Now it all works well. Last question, if I happened to make a mistake in the firmware and for example the device crashes in loop. Is there a way to reflash it or not at all now that flash encryption and secure boot v2 are enabled ?

AdityaHPatwardhan, 2023-07-19:
Hi @FnxQT If there's nothing else then can you close the issue?
Thanks,
Aditya

FnxQT, 2023-07-21:
Thank you for your help.

closed 2023-07-21 by FnxQT