espressif/esp-idf #11890: [regression] 4.4.5 ESP32S3 BLE authentication failure from IOS devices

KonssnoK, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Between the following 2 commits
8b94183c9cb47ede8f02df5598d8b9d68c754860
and
3cec3a0026098d1b027f2103ec154a15baf97318

something changed in the BLE component.

The result of these changes make any IOS device fail to authenticate with error
```
GATT_TRACE_ERROR( "GATT_INSUF_AUTHENTICATION\n");
```

using esp provisioning 2.0.14 -> BLE -> encrypted communication selected

KonssnoK, 2023-07-17:
I see no relevant changes in protocomm and wifi_provisioning

KonssnoK, 2023-07-17:
The issue on IOS is triggered when sending `proto-ver` from function `getDeviceVersionInfo`

KonssnoK, 2023-07-17:
- use IOS with esp BLE provisioning 2.0.14 (from the store)
- use https://github.com/espressif/esp-idf/tree/master/examples/provisioning
- compile with ESP-IDF at commit https://github.com/espressif/esp-idf/commit/8b94183c9cb47ede8f02df5598d8b9d68c754860
- connect using the app -> it will fail
- rollback esp-idf
- compile with ESP-IDF at commit https://github.com/espressif/esp-idf/commit/3cec3a0026098d1b027f2103ec154a15baf97318
- connect using the app -> it will succeed

KonssnoK, 2023-07-17:
@zhp0406 answer above 👍🏼

KonssnoK, 2023-07-17:
Log for failure
```
I (01:01:10.785) processor: Button pressed for 5243 ms!
E (94187) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
W (94264) BT_SMP: Non bonding: No keys will be exchanged
W (96662) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
E (96887) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
I (01:02:00.294) offline: object saved to PSRAM, free 2095777

```
Log for success

```
E (42982) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
W (43058) BT_SMP: Non bonding: No keys will be exchanged
W (44706) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
W (46695) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
E (48734) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
W (48810) BT_SMP: Non bonding: No keys will be exchanged
W (50009) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
I (10:48:26.372) stats: iteration_time_us=999974 Name=last% 
```

KonssnoK, 2023-07-26:
@zhp0406 did you manage to replicate?

esp-cjh, 2023-07-26:
Hi, @KonssnoK 
Could you please revert this commit (6c10a7f60d5241dedbbda1a89d77f6726d829d46)  and try again?

KonssnoK, 2023-07-27:
> Hi, @KonssnoK Could you please revert this commit ([6c10a7f](https://github.com/espressif/esp-idf/commit/6c10a7f60d5241dedbbda1a89d77f6726d829d46)) and try again?

Hi @esp-cjh , i confirm that commit is the issue, by rolling back the entire module to a17e093350f31d09b8bcbfaad40b5b35535ea97b i can pair with IOS

zhp0406, 2023-08-01:
@KonssnoK 
I noticed that the esp-idf/examples/provisioning directory includes several examples. Which example did you use? I utilized the 'esp-idf/examples/provisioning/wifi_prov_mgr' example, but I couldn't reproduce the issue you mentioned. The example seems to be functioning correctly. Below is the log I collected during my testing:

```
I (46973) app: Received Wi-Fi credentials
        SSID     : iQOO 11
        Password : 11112222
I (47983) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (48673) wifi:state: init -> auth (b0)
I (48743) wifi:state: auth -> assoc (0)
I (48793) wifi:state: assoc -> run (10)
I (48813) wifi:connected with iQOO 11, aid = 4, channel 1, BW20, bssid = 62:b2:8c:3c:a6:f4
I (48813) wifi:security: WPA2-PSK, phy: bgn, rssi: -43
I (48833) wifi:pm start, type: 1

I (48833) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 14, mt_pti: 25000, mt_time: 10000
I (48863) wifi:<ba-add>idx:0 (ifx:0, 62:b2:8c:3c:a6:f4), tid:0, ssn:1, winSize:64
I (48873) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (49833) app: Connected with IP Address:192.168.141.49
```

Can you provide me with more information, please? Thank you.

KonssnoK, 2023-08-03:
@zhp0406 i'm sorry but how are you succeeding?
With the provisioning example i cannot even reach the pairing point

i use:
- esp provisioning 2.1.0
- cc71308e2fdce1d6d27fc52d39943f5d0fb83f35 
- esp-idf\examples\provisioning\wifi_prov_mgr
- ios 15.7

If i move back to 7eba5f80027e1648775b46f889cb4d9519afc965 it's the same

```

I (2027) BT_GATT: GATT_GetConnIdIfConnected status=0

W (2027) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x13
E (2037) BT_HCI: CC evt: op=0x2022, status=0x2
I (2037) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (2047) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
W (2057) BT_L2CAP: L2CA_UpdateBleConnParams - unknown BD_ADDR 74a70b337452
E (2057) BT_APPL: Update connection parameters failed!
I (2067) BT_APPL: BTA_DmSetBleAdvParamsAll: 256, 256

I (2067) BT_APPL: adv_type = 0, addr_type_own = 0, chnl_map = 7, adv_fil_pol = 0
```

with android it all works fine.

What test did you do?

zhp0406, 2023-08-03:
@KonssnoK 
Sure, thank you. Could you please provide me with the complete log and did you make any other changes to the code?

KonssnoK, 2023-08-03:
Working provisioning app
```
I (678) wifi:mode : sta (7c:df:a1:e0:74:40)
I (678) wifi:enable tsf
I (678) BLE_INIT: BT controller compile version [963cad4]
I (678) BLE_INIT: Bluetooth MAC: 7c:df:a1:e0:74:42

I (688) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (688) BT_BTM: BTM_SEC_REG[0]: id 50, is_orig 1, psm 0x001f, proto_id 0, chan_id 0

I (698) BT_BTM:                : sec: 0x80, service name [] (up to 21 chars saved)

I (708) BT_BTM: btm_sec_set_security_level : sec: 0x0

I (708) BT_BTM: BTM_SEC_REG[0]: id 50, is_orig 0, psm 0x001f, proto_id 0, chan_id 0

I (718) BT_BTM:                : sec: 0x80, service name [] (up to 21 chars saved)

I (728) BT_GATT: GATT_Register
I (738) BT_GATT: allocated gatt_if=1

I (738) BT_GATT: GATT_StartIf gatt_if=1
I (738) BT_GATT: GATTS_CreateService

I (748) BT_GATT: GATTS_StartService
I (748) BT_GATT: GATT_Register
I (758) BT_GATT: allocated gatt_if=2

I (758) BT_GATT: GATT_StartIf gatt_if=2
I (768) BT_GATT: GATTS_CreateService

I (768) BT_GATT: GATTS_StartService
I (788) BT_BTM: BTM_SetPageScanType

I (788) BT_BTM: BTM_SetInquiryScanType

I (788) BT_GATT: GATT_Register
I (788) BT_GATT: allocated gatt_if=3

I (788) BT_GATT: GATT_StartIf gatt_if=3
I (798) BT_GATT: GATTS_CreateService

I (808) BT_GATT: GATTS_StartService
I (808) BT_APPL: BTA_DmSetBleAdvParamsAll: 256, 256

I (808) BT_APPL: adv_type = 0, addr_type_own = 0, chnl_map = 7, adv_fil_pol = 0

I (818) wifi_prov_mgr: Provisioning started with service name : PROV_E07440 
I (828) app: Provisioning started
I (828) app: If QR code is not visible, copy paste the below URL in a browser.
https://espressif.github.io/esp-jumpstart/qrcode.html?data={"ver":"v1","name":"PROV_E07440","pop":"abcd1234","transport":"ble"}
I (472788) BT_BTM: BTM_InqDbRead: bd addr [4e283631e018]

I (472848) BT_GATT: GATT_GetConnIdIfConnected status=1

I (472848) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (472868) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (472868) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (472908) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (472948) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (472978) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (473078) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (473148) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (473228) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (473298) BT_GATT: GATTS_SendRsp: conn_id: 1  trans_id: 1  Status: 0x0000

I (473298) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (473378) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
I (473448) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 4e283631e018
[... 132 lines cut ...]

I (142149) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 464c485c042d
I (142221) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (142223) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 6  Status: 0x0000

I (142224) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 464c485c042d
E (142296) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
I (142296) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 464c485c042d
W (142374) BT_SMP: Non bonding: No keys will be exchanged
I (142376) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 464c485c042d
I (168708) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 464c485c042d
I (168714) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 464c485c042d
I (168807) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 464c485c042d
I (168884) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0006  BDA: 464c485c042d
W (168957) BT_SMP: FOR LE SC LTK IS USED INSTEAD OF STK
E (169183) BT_GATT: gatts_write_attr_perm_check - GATT_INSUF_ENCRYPTION
```

we didn't touch the BLE, the whole code of esp-idf is shared beetween the 2 applications.

We had the following differences in sdkconfig
- CONFIG_BT_CTRL_DFT_TX_POWER_LEVEL_P6
- CONFIG_BT_CTRL_DFT_TX_POWER_LEVEL_EFF=10
- logging to Warning
- CONFIG_BT_MULTI_CONNECTION_ENBALE=n

KonssnoK, 2023-08-03:
@zhp0406 
another thing that i see
- with the provisioning app it goes directly to the pop page -> no request of pairing
- with out FW it stops and asks to pair (the error is already displayed at this step

zhp0406, 2023-08-03:
@KonssnoK 
You are correct. When I use the provisioning app, there doesn't seem to be any request for pairing. It goes directly to the pop-up page. So, how can I trigger the pairing process?

KonssnoK, 2023-08-03:
@zhp0406 i double confirm.
once removed the option to force the encryption, also our FW started working with IOS.

So we can go back to the initial discussion, but now we know how to trigger it also on the example.

`CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION=y`

```
I (01:00:23.554) main: Free internal heap size: 23775 bytes

I (37597) BT_BTM: BTM_InqDbRead: bd addr [6e6186966ecc]

I (37690) BT_GATT: GATT_GetConnIdIfConnected status=1

I (37692) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0
I (37720) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (37723) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (37780) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (37840) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (37900) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (37975) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38035) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38095) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38155) BT_GATT: GATTS_SendRsp: conn_id: 1  trans_id: 1  Status: 0x0000

I (38156) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38215) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38275) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38335) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38395) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38455) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38515) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38575) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38635) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38695) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (38697) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 2  Status: 0x0000

I (38698) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38755) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (38757) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 3  Status: 0x0000

I (38758) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38815) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (38817) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 4  Status: 0x0000

I (38818) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38875) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (38877) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 5  Status: 0x0000

I (38878) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38935) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (38937) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 6  Status: 0x0000

I (38938) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (38995) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (38998) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 7  Status: 0x0000

I (38999) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (39055) BT_GATT: GATT_GetConnectionInfor conn_id=3
I (39057) BT_GATT: GATTS_SendRsp: conn_id: 3  trans_id: 8  Status: 0x0000

I (39058) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (40795) BT_GATT: GATTS_SendRsp: conn_id: 2  trans_id: 9  Status: 0x0000

I (40796) BT_L2CAP: L2CA_SendFixedChnlData()  CID: 0x0004  BDA: 6e6186966ecc
I (01:01:00.348) offline: object saved to PSRAM, free 2096481
I (01:01:04.345) processor: send status... (60994ms)

```

zhp0406, 2023-08-09:
@KonssnoK 
Hi，could you please try it again with the following config：
(Top) → Component config → Bluetooth → Bluetooth → Host

Please choose (X) NimBLE - BLE only

Also, please ensure that you set:
CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION

Thank you once again.

KonssnoK, 2023-08-11:
@zhp0406 i confirm that selecting Nimble - BLE only removes the issue.

What are the next steps?

zhp0406, 2023-08-14:
@KonssnoK 
If Classic Bluetooth isn't in use, opt for Nimble - BLE only. We are currently addressing certain issues with Bluedroid (dual-mode supported) and working on solutions.

KonssnoK, 2023-08-14:
> @KonssnoK If Classic Bluetooth isn't in use, opt for Nimble - BLE only. We are currently addressing certain issues with Bluedroid (dual-mode supported) and working on solutions.

I can't exclude the fact that classic bluetooth isn't used.
We support all phones to connect to our devices, so i guess old ones will use it 🤔 .

Do you have any forecast on how long will it be before a fix is released?
Thanks

zhp0406, 2023-08-14:
@KonssnoK 
After updating the app on your end to the latest version V2.1.0(4668cda), is the issue still present? or encountered other problems?

KonssnoK, 2023-08-14:
@zhp0406 
sorry i don't understand:
the issue can be resolved on app side?

We didn't do further checks yet

zhp0406, 2023-08-14:
@KonssnoK 
I have now identified the probable cause of the issue: the encryption process has not been completed yet, but the read-write operations were performed on the APP side, which resulted in an error(GATT_INSUF_AUTHENTICATION). I have already reported this issue to my colleagues responsible for the APP.
and I will inform you once the issue has been resolved.
To ensure uninterrupted use, I am offering you a workaround patch. please have a test with it.
[0001-fix-bt-Workaround-for-wifi_prov_mgr.zip](https://github.com/espressif/esp-idf/files/12334441/0001-fix-bt-Workaround-for-wifi_prov_mgr.zip)

KonssnoK, 2023-08-14:
@zhp0406 thanks.
I guess then it's an app issue, I will have to make my colleagues aware of it so that it doesn't happen with our app side too.

I'll try the patch

zhp0406, 2023-08-15:
@KonssnoK 
you are right,it's an app issue.and I will let my colleagues fix it ASAP.

KonssnoK, 2023-08-17:
@zhp0406 apparently my colleagues' IOS app has the same issue connecting to the device.

If you can share the fix once it's known,
thanks.

zhp0406, 2023-08-17:
@KonssnoK 
ok,Is the "work around patch" that I give you works properly?Did you test it?

KonssnoK, 2023-08-17:
> @KonssnoK ok,Is the "work around patch" that I give you works properly?Did you test it?

@zhp0406 
yes the patch works properly, but i guess that is expected, the whole encryption part is cut from the system.

As additional detail, our app is in javascript, not swift, and we still have the issue

KonssnoK, 2023-08-18:
@zhp0406 we fear the issue is intrinsic in the IOS operating system and think that the only way to solve it is by changing the device code.

IOS does not allow to specify when to bond (android does)

We propose the following solution.

[ios_encryption_protocomm.patch](https://github.com/espressif/esp-idf/files/12379494/ios_encryption_protocomm.patch)

What do you think @zhp0406 ?

Does it make sense?

KonssnoK, 2023-08-23:
@zhp0406 is there any news? we are going to temporarily ship the attached patch in case we don't receive updates.
Thanks

zhp0406, 2023-08-23:
@KonssnoK 
I'm sorry, my colleague is still working on it, and I'm also continuously monitoring the issue on my end.

KonssnoK, 2023-08-23:
@zhp0406 did you have time to check the patch i submitted?

zhp0406, 2023-08-24:
@KonssnoK 
received,I will check it ASAP.

KonssnoK, 2023-08-25:
@zhp0406 another test to do!
use CONFIG_WIFI_PROV_BLE_BONDING set, which is not set by default.

I was able to connect with this on IOS 15, now we have to try IOS16 and IOS17

zhp0406, 2023-08-28:
@KonssnoK 
(Top) → Component config → Wi-Fi Provisioning Manager 
[*] Enable BLE bonding
[*] Force Link Encryption during characteristic Read / Write

Once `Enable BLE bonding` is turned on, normal pairing can be completed. Does this fulfill your requirement?

It's important to note that bonding will save the key, so you won't need to pair again next time. If you want to clear the pairing information, you need to execute the following command:

```
bashCopy code
idf.py erase-flash -p /dev/ttyUSB3
```

Replace `/dev/ttyUSB3` with the actual COM ID as per your situation.

Additionally, you'll also need to remove the bonding on iOS.

KonssnoK, 2023-08-28:
@zhp0406 
enabling bonding has solved the issue.
you might want to put it as mandatory if you want to select "force encryption", otherwise you will have issues with IOS.

closed 2023-08-28 by KonssnoK

zhp0406, 2023-08-29:
@KonssnoK 
You are right, we will further optimize it.
Thank you again for reporting the issue.