espressif/esp-idf #11891: bluetooth related crash

SimonPVS, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf v4.3.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board with esp32 

### Power Supply used.

External 3.3V

### What is the expected behavior?

Working bluetooth without any crashes

### What is the actual behavior?

After a random time, the esp crashes and restarts. 
Error from terminal:
`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008c348  PS      : 0x00060c31  A0      : 0x8008c058  A1      : 0x3ffbe540
0x4008c348: resume_dev_in_isr at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:282
 (inlined by) bg_exit_core at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:511
 (inlined by) spi_bus_lock_bg_exit at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:725

A2      : 0x3ffd3a10  A3      : 0x00000000  A4      : 0x3ffbe564  A5      : 0x3ffdf814
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x000fffff
A10     : 0x01004010  A11     : 0x3ffbe564  A12     : 0x01004000  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00060723  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4008c345:0x3ffbe540 0x4008c055:0x3ffbe560 0x40082a65:0x3ffbe5a0 0x4016b937:0x3ffbd8c0 0x400d385d:0x3ffbd8e0 0x40091feb:0x3ffbd900
0x4008c345: bg_exit_core at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:511
 (inlined by) spi_bus_lock_bg_exit at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:725

0x4008c055: spi_intr at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_master.c:674

0x40082a65: _xt_lowint1 at C:/Users/simon/esp/esp-idf-v4.3.1/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4016b937: allocAdjustedBits at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/external/sbc/decoder/srce/bitalloc.c:327

0x400d385d: esp_register_freertos_idle_hook_for_cpu at C:/Users/simon/esp/esp-idf-v4.3.1/components/esp_common/src/freertos_hooks.c:67

0x40091feb: prvIdleTask at C:/Users/simon/esp/esp-idf-v4.3.1/components/freertos/tasks.c:3839`

### Steps to reproduce.

Start from the ESP-IDF bluetooth source example, and build it with esp-idf4.3.1.
After some time (can take hours) the esp will crash and restart

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008c348  PS      : 0x00060c31  A0      : 0x8008c058  A1      : 0x3ffbe540
0x4008c348: resume_dev_in_isr at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:282
 (inlined by) bg_exit_core at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:511
 (inlined by) spi_bus_lock_bg_exit at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:725

A2      : 0x3ffd3a10  A3      : 0x00000000  A4      : 0x3ffbe564  A5      : 0x3ffdf814
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x000fffff
A10     : 0x01004010  A11     : 0x3ffbe564  A12     : 0x01004000  A13     : 0x00000000
A14     : 0x00000001  A15     : 0x00060723  SAR     : 0x0000000b  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace:0x4008c345:0x3ffbe540 0x4008c055:0x3ffbe560 0x40082a65:0x3ffbe5a0 0x4016b937:0x3ffbd8c0 0x400d385d:0x3ffbd8e0 0x40091feb:0x3ffbd900
0x4008c345: bg_exit_core at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:511
 (inlined by) spi_bus_lock_bg_exit at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_bus_lock.c:725

0x4008c055: spi_intr at C:/Users/simon/esp/esp-idf-v4.3.1/components/driver/spi_master.c:674

0x40082a65: _xt_lowint1 at C:/Users/simon/esp/esp-idf-v4.3.1/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x4016b937: allocAdjustedBits at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/external/sbc/decoder/srce/bitalloc.c:327

0x400d385d: esp_register_freertos_idle_hook_for_cpu at C:/Users/simon/esp/esp-idf-v4.3.1/components/esp_common/src/freertos_hooks.c:67

0x40091feb: prvIdleTask at C:/Users/simon/esp/esp-idf-v4.3.1/components/freertos/tasks.c:3839
```

### More Information.

I have tried to update to a higher idf version. 
But other problems occured (sometimes i could connect with bluetooth to the esp, but the connection was broken directly after)

So i would rather stay on idf version 4.3.1 to not introduce new unkown errors

xiongweichao, 2023-07-18:
Hi @SimonPVS , 

1. Could you please describe the usage scenario?

2. Could you please provide the idf commit id?
 >I have tried to update to a higher idf version.
 >But other problems occured (sometimes i could connect with bluetooth to the esp, but the connection was broken directly after)

Thanks

SimonPVS, 2023-07-18:
1. I am developing an audio wall panel with a bluetooth sink. It is of great importance that the bluetooth is stable for lots of hours. I am using an esp32 (dual core), and started from the provided bluetooth sink example. 

2. I am using the stable release v4.3.1 now. Tests occured with v4.3.5 but I had other issues there. 

Other info:
- I am not using a large number of print statements
- The bluetooth task: "bt_app_task_handler" has prio: 'configMAX_PRIORITIES - 3'
- The I2S task: "bt_i2s_task_handler" has the same prio: 'configMAX_PRIORITIES - 3'

I have just found some similar issues: 
- https://github.com/espressif/esp-idf/issues/7805
- https://github.com/espressif/esp-idf/issues/6517

xiongweichao, 2023-07-18:
@SimonPVS The bluetooth task priority is changed to [10](https://github.com/espressif/esp-idf/blob/a2d76ad38ad4be46bdc7ce2becab45fbe71ecc48/examples/bluetooth/bluedroid/classic_bt/a2dp_sink/main/bt_app_core.c#L187).

xiongweichao, 2023-07-18:
Run Bluetooth tasks (including [cotroller task](https://github.com/espressif/esp-idf/blob/a2d76ad38ad4be46bdc7ce2becab45fbe71ecc48/components/bt/controller/esp32/Kconfig.in#L152) and [host task](https://github.com/espressif/esp-idf/blob/a2d76ad38ad4be46bdc7ce2becab45fbe71ecc48/components/bt/host/bluedroid/Kconfig.in#L8)) on the same core, and other tasks on another core.

SimonPVS, 2023-07-26:
@xiongweichao, i changed the bluetooth task prio to 10, but the problem still occured (another failure message this time).
Any other tips?

the error message:
`Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400fecf7  PS      : 0x00060530  A0      : 0x800fe07c  A1      : 0x3ffe5260
0x400fecf7: reassemble_and_dispatch at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/packet_fragmenter.c:175
 (inlined by) reassemble_and_dispatch at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/packet_fragmenter.c:136

A2      : 0x3ffbd1c8  A3      : 0x0000000c  A4      : 0x0000957a  A5      : 0x00000080
A6      : 0x00001100  A7      : 0x00000000  A8      : 0x800fecf4  A9      : 0x3ffe5210
A10     : 0x00000000  A11     : 0x00009582  A12     : 0x00000000  A13     : 0x00000018
A14     : 0x00000020  A15     : 0x00000024  SAR     : 0x00000016  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000002  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

Backtrace:0x400fecf4:0x3ffe5260 0x400fe079:0x3ffe5290 0x40116a6e:0x3ffe52b0 0x4011010b:0x3ffe52e0 0x4011697c:0x3ffe5300 0x401109f3:0x3ffe5320
0x400fecf4: reassemble_and_dispatch at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/packet_fragmenter.c:174
 (inlined by) reassemble_and_dispatch at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/packet_fragmenter.c:136

0x400fe079: hal_says_packet_ready at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/hci_layer.c:413

0x40116a6e: hci_hal_h4_hdl_rx_packet at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/hci_hal_h4.c:296
 (inlined by) event_uart_has_bytes at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/hci_hal_h4.c:304

0x4011010b: fixed_queue_process at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/common/osi/fixed_queue.c:254

0x4011697c: hci_hal_h4_rx_handler at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/host/bluedroid/hci/hci_hal_h4.c:170

0x401109f3: osi_thread_run at C:/Users/simon/esp/esp-idf-v4.3.1/components/bt/common/osi/thread.c:68`

xiongweichao, 2023-07-26:
@SimonPVS An [illegal pointer](https://github.com/espressif/esp-idf/blob/9a1cc59338aa151bc6333f85c8169d36a478fe2a/components/bt/host/bluedroid/hci/packet_fragmenter.c#L174) was used. Is there a memory leak?

SimonPVS, 2023-07-26:
@xiongweichao, how can I verify this?

xiongweichao, 2023-07-26:
@SimonPVS  You can refer to [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#how-to-diagnose-memory-leaks).

SimonPVS, 2023-07-26:
But the memory leak would be in the bluetooth driver?

xiongweichao, 2023-07-26:
It is possible.

SimonPVS, 2023-08-10:
Hi @xiongweichao,

After more testing, i was not able to find a memory leak.
I have a periodically re-occuring task which prints the heap every 10 seconds.

I can see when i connect to bluetooth, the heap starts decreasing slowly, but it stabalizes around 40k. 
So the heap does not seems to be the problem?

I am not sure anymore if the problem even is bluetooth related. Because i got a similar failure, but it does not seems to be triggered from a bluetooth function:
[image]

I am thinking it is something interrupt / spi related problem?

Some other things i tried in the mean time:
- Placed UART into IRAM
- Changed bluetooth core task to 10
- Bluetooth transmit and receive functions to IRAM
- Processor speed to 240MHz

Could you please provide some more info on how to proceed? It would be very useful if i could replicate the failure more often, because now it can take multiple days before the failure occures.