espressif/esp-idf #11892: memory leak in esp_http_client library

parabuzzle, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board using ESP32-WROVER-E series module

### Power Supply used.

External 3.3V

### What is the expected behavior?

I expect to make an https get using the esp_http_client library with no leaked memory.

### What is the actual behavior?

I am doing a basic https get wrapped in the heap tracer and I see it leaks around 400 bytes per request.

### Steps to reproduce.

This is my `main.c`

```c
#include <stdio.h>
#include <inttypes.h>
#include "esp_system.h"
#include "esp_netif.h"
#include "esp_event.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "event_handlers.h"

#include "esp_log.h"
#include "esp_tls.h"
#include "esp_http_client.h"
#if CONFIG_MBEDTLS_CERTIFICATE_BUNDLE
#include "esp_crt_bundle.h"
#endif

#include "esp_heap_trace.h"

#define URL "https://jsonplaceholder.typicode.com/todos/1"

#define WIFI_SSID "my ssid"
#define WIFI_PASSWORD "mypassword!"

[... 204 lines cut ...]
    16 bytes (@ 0x3ffc5738, Internal) allocated CPU 0 ccount 0x4f30b1d8 caller 0x4012789b:0x4012792e
0x4012789b: mem_malloc at C:/Users/parab/esp-v5.1/esp-idf/components/lwip/lwip/src/core/mem.c:209

0x4012792e: do_memp_malloc_pool at C:/Users/parab/esp-v5.1/esp-idf/components/lwip/lwip/src/core/memp.c:254

    92 bytes (@ 0x3ffd5fe4, Internal) allocated CPU 0 ccount 0x4fa5bd34 caller 0x4012789b:0x40128a27
0x4012789b: mem_malloc at C:/Users/parab/esp-v5.1/esp-idf/components/lwip/lwip/src/core/mem.c:209

0x40128a27: pbuf_alloc at C:/Users/parab/esp-v5.1/esp-idf/components/lwip/lwip/src/core/pbuf.c:284

    16 bytes (@ 0x3ffd6044, Internal) allocated CPU 0 ccount 0x4fa5f420 caller 0x4012789b:0x4012792e
0x4012789b: mem_malloc at C:/Users/parab/esp-v5.1/esp-idf/components/lwip/lwip/src/core/mem.c:209

0x4012792e: do_memp_malloc_pool at C:/Users/parab/esp-v5.1/esp-idf/components/lwip/lwip/src/core/memp.c:254

====== Heap Trace Summary ======
Mode: Heap Trace Leaks
396 bytes 'leaked' in trace (8 allocations)
records: 8 (1000 capacity, 62 high water mark)
total allocations: 300
total frees: 1977
================================
I (8441) main: Looping in 1 sec...
```
```

nopnop2002, 2023-07-17:
ESP-IDF runs a system task in the background that constantly allocates and frees memory.

So it's fairly certain that your application will continue to consume memory right after it starts.

After some time, the memory consumption does not change.

I kept your sample running and now it's stable.

```
I (449943) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (451063) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223288
I (452973) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (454093) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223288
I (455983) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (457113) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223280
I (459013) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (460133) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223276
I (462033) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (463153) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223276
I (465053) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (466193) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223280
I (468103) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (469233) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223276
I (471143) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (472283) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223280
I (474183) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (475323) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223088
I (477223) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (478363) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223088
I (480253) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (481383) esp-x509-crt-bundle: Certificate validated
esp_get_free_heap_size()                              : 223280
I (483273) MAIN: Fetching Data from URL: https://jsonplaceholder.typicode.com/todos/1
I (484403) esp-x509-crt-bundle: Certificate validated
```

parabuzzle, 2023-07-26:
ok. I was able to track it down to the clean up task. I added a vTaskDelay after the client clean up and before checking the heap and it was fine. so obviously is the background tasks. We can close this.

closed 2023-07-26 by parabuzzle