espressif/esp-idf #11893: BT_NIMBLE_HOST_BASED_PRIVACY config option gone

Stoician, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Should be able to turn on BT_NIMBLE_HOST_BASED_PRIVACY and call functions such as  ble_hs_pvcy_rpa_config() when the compile target is ESP32C3.

### What is the actual behavior?

After updating to v5.0.3 the menuconfig option disappears and the compile fails.

The option returns and compiling works again if I undo the change on line 483 in components/bt/host/nimble/Kconfig.in made with the following commit:
https://github.com/espressif/esp-idf/commit/13f62f9d4bfee1b022f0f63246f12108212f09f4

Was this line committed by mistake?

### Steps to reproduce.

^

rahult-github, 2023-07-18:
ESP32-C3 has controller privacy enabled by default. So enabling host based privacy will cause issues on this chip. So the change is done to disable host based privacy. 

Controller based privacy is recommended to be used.

closed 2023-07-18 by rahult-github

Stoician, 2023-07-18:
I reverted to the original code so that host based privacy was turned off and removed the function call.
Advertising stopped working without any error message.
After changing,
ble_gap_adv_start(BLE_OWN_ADDR_RPA_RANDOM_DEFAULT... 
to
ble_gap_adv_start(BLE_OWN_ADDR_RPA_PUBLIC_DEFAULT...
advertisement started working again.

But now I am seeing this issue: https://github.com/espressif/esp-idf/issues/10915
That is, it also initializes with the same address, only changing to a random one after 15 minutes.