espressif/esp-idf #11894: Generate compile_commands.json for all ULP subprojects

andylinpersonal, 2023-07-17:
### Is your feature request related to a problem?

I would like to generate `compile_commands.json` for all ULP-RISC-V binaries to use auto-completion and code navigation (with clang or vscode intellisense LSPs).

### Describe the solution you'd like.

1. Add `-DCMAKE_EXPORT_COMPILE_COMMANDS` to the subproject created in the `${IDF_PATH}/components/ulp/project_include.cmake`. 
2. Merge all generated `compile_commands.json` into a single file to the topmost level of cmake's workspace.

Currently, I've only done step.1 and merge all `compile_commands.json`s to `build/compile_commands.json` in a `launch.json` task.

SoucheSouche, 2024-09-04:
0494dc45d1f3de324fc0ccbfc5da50c37e98754c

closed 2024-09-04 by SoucheSouche

KaeLL, 2024-09-04:
It didn't address the merging part.
I run a `jq` oneliner to merge the bootloader and app compile_commands when I'm fiddling with the bootloader, but I'm sure it can be done in python and added to idf.py somewhere, optionally or otherwise, so that it can merge all `compile_commands.json` files.
@sudeep-mohanty thoughts?

sudeep-mohanty, 2024-09-06:
@KaeLL Thanks for the suggestion. Indeed, there can several ways of merging the various `compile_commands.json` but if it's done as part of the build then it would be most seamless. In that regard, I will consult the tools team and check how best this can be done.

andylinpersonal, 2024-09-07:
By migrating to the newer version of `clangd`, which supports multiple `compile_commands.json`, can solve this problem and have better hinting and completion experience.

[image]

genpfault, 2025-05-12:
@andylinpersonal: "the newer version of `clangd`, which supports multiple `compile_commands.json`"

Which version is that?

alexcui03, 2026-08-19:
> [@andylinpersonal](https://github.com/andylinpersonal): "the newer version of `clangd`, which supports multiple `compile_commands.json`"
> 
> Which version is that?

i use clangd 19.1.2 with the following configuration in `.clangd` file, it works well.
```
CompileFlags:
  Remove: [-f*, -m*]
  CompilationDatabase: build

---
If:
  PathMatch: main/ulp/.*

CompileFlags:
  CompilationDatabase: build/esp-idf/main/ulp_main
```
`main/ulp` is the directory where i placed sources for ulp lp core, and `ulp_main` is the first argument of `ulp_add_project`.