espressif/esp-idf #11895: ESP-WHO human face detection\web not working

krupis, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf v4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-EYE

### Power Supply used.

USB

### What is the expected behavior?

Device connects to the configured Station and runs properly (I should be able to see live camera footage via web)

### What is the actual behavior?

Devices starts, connects to STA and crashes due to 
```
***ERROR*** A stack overflow in task who_camera has been detected.
```

### Steps to reproduce.

1. Open the example project ```esp-who\examples\human_face_detection\web```
2. Select target as ESP32-S3
3. Open menuconfig and configure WiFi station name and password
4. Build and Flash the device

### Debug Logs.

```plain
ELF file SHA256: 24f66d86ec6c48ee

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x420de09a
0x420de09a: esp_pm_impl_waiti at C:/Users/Lukas/esp4/esp-idf/components/esp_pm/pm_impl.c:839

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x470
load:0x403c9700,len:0xc00
load:0x403cc700,len:0x2af8
SHA-256 comparison failed:
Calculated: fa346a3c59af2029c0b1aa9d0b0917587366ecc88675145fcd37144952a791b8
Expected: ba09d14d25b7c1504e5dc523b718551a170a8f6087f0384b37b94f022d4cd44e
Attempting to boot anyway...
entry 0x403c98e8
[... 245 lines cut ...]
I (666) wifi:pm start, type: 0

I (666) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
I (676) wifi:<ba-add>idx:0 (ifx:0, b4:b0:24:ec:5a:a5), tid:0, ssn:3, winSize:64
I (736) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1666) esp_netif_handlers: sta ip: 192.168.0.182, mask: 255.255.255.0, gw: 192.168.0.1
I (1666) camera wifi: got ip:192.168.0.182
W (4796) cam_hal: Failed to get the frame on time!

***ERROR*** A stack overflow in task who_camera has been detected.

Backtrace: 0x40375efa:0x3fcd1af0 0x4037dd69:0x3fcd1b10 0x403816a2:0x3fcd1b30 0x4037fa73:0x3fcd1bb0 0x4037de20:0x3fcd1bd0 0x4037de16:0x00000000 |<-CORRUPTED
0x40375efa: panic_abort at C:/Users/Lukas/esp4/esp-idf/components/esp_system/panic.c:408

0x4037dd69: esp_system_abort at C:/Users/Lukas/esp4/esp-idf/components/esp_system/esp_system.c:137

0x403816a2: vApplicationStackOverflowHook at C:/Users/Lukas/esp4/esp-idf/components/freertos/port/xtensa/port.c:407

0x4037fa73: vTaskSwitchContext at C:/Users/Lukas/esp4/esp-idf/components/freertos/tasks.c:3505

0x4037de20: _frxt_dispatch at C:/Users/Lukas/esp4/esp-idf/components/freertos/port/xtensa/portasm.S:436

0x4037de16: _frxt_int_exit at C:/Users/Lukas/esp4/esp-idf/components/freertos/port/xtensa/portasm.S:231
```
```

igrr, 2023-07-17:
It seems there is already a similar issue reported in esp-who project: https://github.com/espressif/esp-who/issues/249.

krupis, 2023-07-18:
> It seems there is already a similar issue reported in esp-who project: [espressif/esp-who#249](https://github.com/espressif/esp-who/issues/249).

You are right. This seems like an old issue I'm surprised the fix hasn't been officially released,

Thanks for pointing this out!

closed 2024-05-10 by igrr