espressif/esp-idf #11896: [ESP32H2] IDF v5.1 Monitor: Input on app console causes monitor to freeze

heliochronix, 2023-07-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm running an Arch system with kernel 6.4.3, and I have the latest `release/v5.1` branch pulled (as of this posting). I'm having an issue with the IDF Monitor when talking to an ESP32-H2 board running a console app.

During the `get-started/hello_world` or `get-started/blink` program, output from the monitor is normal, though input on the keyboard seems to cause it to hang up for a bit before it resumes outputting stuff from the target.

However, if I try to run something like `openthread/ot_cli` that has a console, it will output fine at first but any keyboard input seems to freeze things up, to the point I can't even use CTRL-] or CTRL-T CTRL-X to exit the console and I'm forced to manually kill the python process. I can still get CTRL-T CTRL-H to output help text, but resetting, quitting, and rebuilding/flashing the target all seem to do nothing, and all output from the target stops. The only way to get it back seems to be killing the monitor python process and restarting it, which causes the reset to trigger.

Any help in troubleshooting this would be appreciated! I'm relatively new, so apologies if I've missed some common troubleshooting steps.

dobairoland, 2023-07-18:
Duplicate report here: https://esp32.com/viewtopic.php?f=13&t=34673

heliochronix, 2023-07-18:
> Duplicate report here: https://esp32.com/viewtopic.php?f=13&t=34673

Apologies, I made that post first but then decided I'd rather have a GH issue to track this, plus I was able to add more details here.

peterdragun, 2023-07-18:
Hi, are you using the native USB port of ESP32H2? Can you see the same behavior when using the UART port (if available)? Have you tried setting 'Channel for console output' to USB Serial? - You can do it with the menuconfig command under 'Component config > ESP System Settings > Channel for console output'

igrr, 2023-07-18:
@heliochronix Could you please also try examples/system/console/basic with the setting mentioned by @peterdragun above (CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG)? That example definitely works over USB_SERIAL_JTAG when that option is enabled.

igrr, 2023-07-18:
My bad, I forgot that we haven't backported https://github.com/espressif/esp-idf/commit/bb3f1fd2619d9344e7ceda4fced7abc41231a198 to release/v5.1 yet.

Please try to `git cherry-pick bb3f1fd2619d9344e7ceda4fced7abc41231a198`, it should apply cleanly on release/v5.1.

heliochronix, 2023-07-18:
It didn't exactly apply cleanly, but I did resolve the conflict. Also there appears to be a typo in `get_version` where it checks for `CHIP_ESP32H4` and sets `model = "ESP32-H4"` (resulting in a build error because `CHIP_ESP32H4` doesn't exist) instead of `CHIP_ESP32_H2` and setting `model = "ESP32-H2"`. I fixed this as well locally.

As for the results of the test, the behavior is identical. It outputs everything, but I can't type anything and CTRL-] doesn't work, along with CTRL-T and any subsequent commands except for the help text.

heliochronix, 2023-07-18:
Following up on the other testing, connecting up to the UART directly seems to work. It's seemingly just USB Serial/JTAG that is causing issues on the monitor.

igrr, 2023-07-18:
In the case of examples/system/console/basic, could you please confirm that you have CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y in the sdkconfig file?

igrr, 2023-07-18:
> When I tested this on the other app I was still having issues. Let me double check that though...

As far as I can tell, openthread/ot_cli example, unlike system/console/basic, doesn't implement console support via USB_SERIAL_JTAG, [there is only UART support](https://github.com/espressif/esp-idf/blob/a2d76ad38ad4be46bdc7ce2becab45fbe71ecc48/components/openthread/src/port/esp_openthread_uart.c#L60-L97). (cc @chshu)

chshu, 2023-07-21:
@heliochronix You may follow this https://github.com/espressif/esp-idf/issues/11924#issuecomment-1645595686 to enable USB port for ot_cli example.

closed 2023-08-08 by espressif-bot