espressif/esp-idf #11897: ESP32-C3 CAN bus (TWAI) filter

prazsmarid, 2023-07-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
Can somebody calculate for me the [b]acceptance_code [/b] and [b]acceptance_mask[/b] values?
I want to receive only the 0x17330800, 0x17330810 and 0x5F0 IDs.
Thanks in advance.

wanckl, 2023-07-18:
@prazsmarid I'm sad to tell you that your wishes may hard to archived, let see your request:
```
17330800
17330810
000005f0
```
The `000005f0` is so far away from others, there are no same bit(s) among three IDs. :face_exhaling:    

An alternative is using `Dual Filter Mode`, this mode split 32bits id and mask to two 16bits, then you can:   
Filter 1:
```
id = 0x1733 << 3;
mask = 0;    // will accept only 0x1733xxxx IDs
```
Filter 2:
```
id = 0;
mask = 0;    // will accept only 0x0000xxxx IDs
```
This way your filter config will be:
```
.acceptance_code = 0xb9980000,
.acceptance_mask = 0,
.single_filter = false,
```

prazsmarid, 2023-07-18:
Thank you for trying to help, I have tried the values which you suggested:
```
.acceptance_code = 0xb9980000,
.acceptance_mask = 0,
.single_filter = false,
```
The filter1 which filters the 0x1733xxxx IDs now it seems to be good, I not have too much traffic on the 0x1733xxxx IDs...
But the filter2 which should filter the 000005f0 is not passing any kind off messages. Do you have idea about why the filter2 not passes the 0x0000xxxx messages?

procoderer, 2024-05-19:
I am also having trouble finding the correct acceptance code and mask for what I want. I want to receive only messages with IDs 0x355 and 0x356. Can someone help me calculate this please?

prazsmarid, 2024-05-19:
> I am also having trouble finding the correct acceptance code and mask for what I want. I want to receive only messages with IDs 0x355 and 0x356. Can someone help me calculate this please?

I gave up the project with the ESP32-C3, the TWAI peripherial is a toy...
In your case the ESP32-C3 is good, but I have no idea about the calculations. Somewhere (github or the original Espresiv forum) someboody calculated the values which were in my question and explained these values very well, but not I not found it.

nopnop2002, 2024-05-21:
$ python3

>>> print(0x17330800 & 0x17330810 & 0x5F0)
0