espressif/esp-idf #11898: ESP-IDF make  error

Shivdoke39, 2023-07-18:
lnv173@DESKTOP-5FHABSQ MINGW32 ~/esp32-ota
# make menuconfig
Makefile:8: C:/msys32/home/lnv173/esp/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target 'C:/msys32/home/lnv173/esp/esp-idf/make/project.mk'.  Stop.

lnv173@DESKTOP-5FHABSQ MINGW32 ~/esp32-ota
# printenv IDF_PATH
C:/msys32/home/lnv173/esp/esp-idf

lnv173@DESKTOP-5FHABSQ MINGW32 ~/esp32-ota
# make menuconfig
Makefile:8: C:/msys32/home/lnv173/esp/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target 'C:/msys32/home/lnv173/esp/esp-idf/make/project.mk'.  Stop.

igrr, 2023-07-18:
Looks like you are trying to build the project which uses the legacy GNU Make based build system using a version of IDF which doesn't support it anymore. Please check the [migration guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#migrating-from-gnu-make-build-system) regarding migration to the CMake based build system. v4.x versions of IDF include a helper script which can help you migrate the project from Make to CMake.

Alvin1Zhang, 2023-08-01:
@Shivdoke39 Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

ESP-Marius, 2023-11-17:
Closing this as there hasnt been any updates, feel free to re-open if you are still having this issue and need help resolving it.

closed 2023-11-17 by ESP-Marius