espressif/esp-idf #11901: i2c_tools console not working with esp32c3

josh-gc, 2023-07-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

## Issue:
When I run esp-idf/examples/peripherals/i2c/i2c_tools on an esp32-c3-mini-1 over the usb pins (D+/D- connected to uUSB connector) my console gives the error below and does not allow me to enter any commands:

```
I (375) sleep: Enable automatic switching of GPIO sleep configuration
I (382) cpu_start: Starting scheduler.

 ==============================================================
 |             Steps to Use i2c-tools                         |
 |                                                            |
 |  1. Try 'help', check all supported commands               |
 |  2. Try 'i2cconfig' to configure your I2C bus              |
 |  3. Try 'i2cdetect' to scan devices on the bus             |
 |  4. Try 'i2cget' to get the content of specific register   |
 |  5. Try 'i2cset' to set the value of specific register     |
 |  6. Try 'i2cdump' to dump all the register (Experiment)    |
 |                                                            |
 ==============================================================

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.

Your terminal application does not support escape sequences.

Line editing and history features are disabled.

On Windows, try using Putty instead.
```

When I run this same code on an esp32 I get no errors and this input line:
```
Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
i2c-tools> 
```
I get the same issue with the console example but am able to do everything else over this connection including the `printf` and `ESP_LOGx` commands.

## Configuration
Computer: M2 MacBook Pro (Ventura 13.4.1)
IDE: VS Code
esp-idf: release/v4.4 (commit 17180803405e92a695f27623f1067323aef511db)

### esp32-c3-mini-1
```
"idf.adapterTargetName": "esp32c3",
"idf.port": "/dev/cu.usbmodem1114101",
"idf.flashType": "UART"
```
### esp32
```
"idf.adapterTargetName": "esp32",
"idf.port": "/dev/cu.usbserial-01BFAACF",
"idf.flashType": "UART"
```

Is there a solution for this?

josh-gc, 2023-07-18:
@igrr Thank you! That solved the issue!

Jrodmez, 2024-09-12:
I couldn't find CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG in menuconfig, but it is possible to update the sdkconfig directly:    
                      
[image]

This solve the issue for me.