espressif/esp-idf #11903: OTA invalid magic byte

baldhead69, 2023-07-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am implementing the example "advanced_https_ota_example" [https://github.com/espressif/esp-idf/bl ... _example.c](https://github.com/espressif/esp-idf/blob/v5.1/examples/system/ota/advanced_https_ota/main/advanced_https_ota_example.c) with some particular changes.

After ota app update at new partition( 0x1010000 ), appear this error message, and ESP32-S3 boot from old partition ( 0x20000 ).
```
E (129) esp_image: image at 0x1010000 has invalid magic byte (nothing flashed here?)
E (137) boot: OTA app partition slot 1 is not bootable
```

https://www.esp32.com/viewtopic.php?f=13&t=34682&p=117009#p117009

Thank's.

mythbuster5, 2023-07-19:
related docs:  https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#bit-address-flash-chips

baldhead69, 2023-07-20:
> related docs:  https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/spi_flash/spi_flash_optional_feature.html#bit-address-flash-chips 

My board use octal flash communication( i configured in this way ), so, i think that docs not apply and my memory chip is different, i think it is MX25UM25645G.

And i dont have sure if the problem is the memory size ( 32MB ).

baldhead69, 2023-07-21:
The problem, i think, is flash memory size( 32MB ).

I changed the partition table to address below 16MB and it worked( csv file in annex ).

How to make it work for addresses above 16MB too ???

Update answer here:
https://www.esp32.com/viewtopic.php?f=13&t=34682&p=117182#p117182

ESP32DE, 2023-08-01:
hi @baldhead69 
https://github.com/espressif/esptool/issues/883
best wishes
;-)

baldhead69, 2023-08-01:
hi @ESP32DE,

In my case the ota update by wifi achieves flash the image on flash memory without problems.

When the ota update restart the chip, the message "OTA invalid magic byte" appear and the esp32-s3 use the old firmware to start.

The problem only appears if the address of the new firmware is above 16 MB( specified in the partition file ).

Thank's.

mythbuster5, 2023-10-12:
[ota_32M.txt](https://github.com/espressif/esp-idf/files/12877177/ota_32M.txt)
 Sorry for the late reply. I'm working on other stuff these days... rename it to ota_32M.patch, and apply it. I think this will solve your problem

baldhead69, 2023-10-17:
Hi @mythbuster5,

I applied the patch and errors appear.
I dont know if i applied the patch correctly.

[image]

mythbuster5, 2023-10-17:
5.1? My version is based on the latest, let me check.

baldhead69, 2023-10-17:
5.1 is the latest stable version, no ?

baldhead69, 2023-10-19:
@mythbuster5 ,

you can change the latest stable version of esp idf.

I download the latest stable version, just tell me what it is, I think it's version 5.1.1 (give me the download link so we can be sure we are working with the same version of esp-idf).

baldhead69, 2023-10-25:
@mythbuster5 ,

you can change the latest stable version of esp idf.

I download the latest stable version, just tell me what it is, I think it's version 5.1.1 (give me the download link so we can be sure we are working with the same version of esp-idf).

@mythbuster5 ,

Completed ?

mythbuster5, 2023-10-31:
Please pick this commit to your branch https://github.com/espressif/esp-idf/commit/6cea72b76b90e752ad135d7cbc7074b786244584 The backport on 5.1 has not been arrived.

baldhead69, 2023-10-31:
Work on latest stable version ?

I think latest stable version is v5.1.1

baldhead69, 2023-11-07:
@mythbuster5,

I updated esp-idf to v5.2-dev-3775-gb4268c874a and got errors yet.

@igrr, could you help me with this issue ? 

[image]

baldhead69, 2023-11-08:
Hi,

Someone can open this issue again, since this hasn't been solved yet.

baldhead69, 2023-11-14:
?????????

ginkgm, 2023-11-15:
Hi @baldhead69 

5.1.2 is going to be released soon, you may wait for that.

baldhead69, 2023-11-23:
I want to know why this issue was not reopened ???

Open this issue again.

This issue is not solved yet.

I downloaded esp-idf v5.1.2 and the SAME problem persists.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x18 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379c47
0x40379c47: esp_restart_noos_dig at C:/esp-idf/components/esp_system/port/esp_system_chip.c:57 (discriminator 1)

SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce3818,len:0x17fc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcb4
load:0x403cc700,len:0x3008
entry 0x403c9984
I (38) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (38) boot: compile time Nov 23 2023 19:31:29
I (38) boot: Multicore bootloader
I (41) boot: chip revision: v0.1
I (45) boot.esp32s3: Boot SPI Speed : 80MHz
I (49) boot.esp32s3: SPI Mode       : SLOW READ
I (55) boot.esp32s3: SPI Flash Size : 32MB
I (59) boot: Enabling RNG early entropy source...
I (65) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (76) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (91) boot:  2 ota_0            OTA app          00 10 00020000 00fe7000
I (98) boot:  3 ota_1            OTA app          00 11 01010000 00fe7000
I (106) boot: End of partition table
E (110) esp_image: image at 0x1010000 has invalid magic byte (nothing flashed here?)
E (118) boot: OTA app partition slot 1 is not bootable
I (124) esp_image: segment 0: paddr=00020020 vaddr=3c0c0020 size=86c20h (551968) map
I (257) esp_image: segment 1: paddr=000a6c48 vaddr=3fc9d000 size=05678h ( 22136) load
I (263) esp_image: segment 2: paddr=000ac2c8 vaddr=40378000 size=03d50h ( 15696) load
I (267) esp_image: segment 3: paddr=000b0020 vaddr=42000020 size=bb358h (766808) map
I (444) esp_image: segment 4: paddr=0016b380 vaddr=4037bd50 size=11290h ( 70288) load
I (470) boot: Loaded app from partition at offset 0x20000
```

Spritetm, 2023-11-24:
Reopened as apparently this is not fixed yet.

baldhead69, 2023-11-28:
Considering that my idea is to divide the 32 MB memory into two 16 MB partitions and that for accessing the first 16 MB only need 3 bytes of address ( 2^24 = 16.777.216 -> 0 to 16.777.215 ), 16.777.215 = 11111111 11111111 11111111 and for acessing 32 MB only need 25 bits ( 2^25 = 33.554.432 -> 0 to 33.554.431 ), 33.554.431 = 0001 11111111 11111111 11111111, means that only bit 24 changes the value when accessing the two partitions.  

For performance reasons, wouldn't there be a way (command on flash memory) informing the memory that which will be accessed  in that moment, ie: 0 - 16MB or 16MB-32MB ?

baldhead69, 2023-12-20:
@Spritetm ,

Do you have some position on this ?

Thanks.

igrr, 2023-12-20:
The fix on release/v5.1 branch is linked above — https://github.com/espressif/esp-idf/commit/bb7544e65aba750aad7dfd641a2b166b9e0a2ffe. This commit is not in v5.1.2 release, it will be included in v5.1.3.

baldhead69, 2023-12-20:
@igrr 

I have version 5.1.2.

How do I use this commit in version 5.1.2  ?

baldhead69, 2023-12-25:
????

baldhead69, 2024-01-04:
@igrr

igrr, 2024-01-04:
@baldhead69 Please try this:

```
git cherry-pick bb7544e
git cherry-pick 8f6213c9
```

(the 2nd commit is the fix for the compilation issue reported in #12850, linked above)

baldhead69, 2024-01-10:
@igrr 

[image]

esp-idf version: v5.2-dev-3775-gb4268c874a

baldhead69, 2024-01-10:
> The fix on release/v5.1 branch is linked above — [bb7544e](https://github.com/espressif/esp-idf/commit/bb7544e65aba750aad7dfd641a2b166b9e0a2ffe). This commit is not in v5.1.2 release, it will be included in v5.1.3.

Is esp_idf v5.1.3 already available ?

baldhead69, 2024-01-11:
[fetch.txt](https://github.com/espressif/esp-idf/files/13896958/fetch.txt)

I tried this:
fatal: bad revision while cherry-picking a commit!
This happens when your local git remains oblivious of the existence of the commit. To fix this, fetch all the branches, and then cherry-pick again.

From here:
https://blog.mayadata.io/openebs/avoiding-git-panic-part-i-novice#:~:text=fatal%3A%20bad%20revision%20while%20cherry,and%20then%20cherry%2Dpick%20again.

But not worked.

baldhead69, 2024-02-08:
The ota is working now.

Solution here:
https://www.esp32.com/viewtopic.php?f=13&t=37734

Alvin1Zhang, 2024-02-08:
Thanks for sharing the feedback, feel free to reopen.

closed 2024-02-08 by Alvin1Zhang