espressif/esp-idf #11906: Unable to install release versions/v4.4 and below via VS code extension.

akamal816, 2023-07-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf release/v4.4 or lower

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Espressif VS code extension should be able to install previous versions of esp-idf branches. 

### What is the actual behavior?

Espressif VS code extension is unable to install previous versions of esp-idf branches. 

### Steps to reproduce.

1. Install Espressif VS Code Extension
2. Select Express install option
3. Choose any release version prior to 5.0
4. Install will fail. 
...

### More Information.

Hello all, I would like to install the Espressif vs code extension with a branch version of the esp-idf library 5.0 or less. 
For whatever reason, it seems like the CAN drivers were deprecated after 5.0 as I can't find them in the 'components' folder. 

I am able to successfully install the Espressif vs code extension with branch versions 5.0 and later of esp-idf library. 

Installing versions less than 5, the idf_tool.py file raises an import error mentioning "Module not found virtualenv". 
I verified that the python interpreter that espressif is using does in fact have virtualenv installed. 
(I've reinstalled virtualenv via the terminal and still I get the same issue). 

I've read on the forums that 'virutalenv' is quite problematic and that's why in the 5.0 release they've switched to 'venv'. 

Really I just need CAN functionality so if installing versions prior to 5.0 isn't possible, is there anyway to port the CAN functionality of previous esp-idf versions to 5.0 or greater.

dobairoland, 2023-07-19:
> I verified that the python interpreter that espressif is using does in fact have virtualenv installed.

This looks like a vscode issue and I don't think we can do anything about it here. Could you please check the guided at https://github.com/espressif/vscode-esp-idf-extension and if the issue persist then report it there.

As for the other issue, I'm not a CAN or TWAI expert but I see the change more or less just a rename of the name. See https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/peripherals.html#twai-driver and please search for TWAI for more information.

akamal816, 2023-07-19:
I've taken a look at the [CAN documentation espressif provides](https://docs.espressif.com/projects/esp-idf/en/release-v3.3/api-reference/peripherals/can.html) and compared it to the [TWAI documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html) and I agree. It seems to be a 1 to 1 copy with the exception of bus speeds and the obvious rename. 

Not sure why Espressif decided to rename CAN to TWAI. I'm assuming it has to do with saving money on licensing fees. 

Nonetheless I'll stick with TWAI as opposed to trying to get previous versions to work. Thank you @dobairoland !!!

closed 2023-07-19 by akamal816