espressif/esp-idf #11907: RMT driver and Wi-fi driver coexistence.

guirespi, 2023-07-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 WROVER KIT

### Power Supply used.

USB

### What is the expected behavior?

Wi-fi and RMT driver for WS2812 should coexist without freezing the program. The code I made starts a task that constantly creates and delete anoter task that turn the WS2812 leds.

### What is the actual behavior?

When try to connect to an non existent SSID and the RMT driver is used to turn on and turn off a WS2812 led strip, the program freezes. Neither Wi-fi or led strip do something after.

### Steps to reproduce.

Compile and run the following code . The base is the 'station' project in the examples folder (esp_idf>examples>wifi>getting_started). Then I add the component led_strip (esp_idf>examples>common_components) to the components folder. 

This a draft code, the production code uses a different implementation of led_strip component but, I achieve to recreate the same scenario with espressif's components. In the production code, the program freezes after several hours of use and it is always at the use of the RMT driver.

```
/* WiFi station Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sys.h"
#include "../components/led_strip/include/led_strip.h"

/* The examples use WiFi configuration that you can set via project configura
 * tion menu

[... 210 lines cut ...]
I (390) cpu_start: Project name:     wifi_station
I (395) cpu_start: App version:      v4.4.2-dirty
I (401) cpu_start: Compile time:     Jul 19 2023 11:53:27
I (407) cpu_start: ELF file SHA256:  3bc20b1ca34f9941...
I (413) cpu_start: ESP-IDF:          v4.4.2-dirty
I (418) heap_init: Initializing. RAM available for dynamic allocation:
I (425) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (431) heap_init: At 3FFB7570 len 00028A90 (162 KiB): DRAM
I (438) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (444) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (451) heap_init: At 40092FF8 len 0000D008 (52 KiB): IRAM
I (458) spi_flash: detected chip: generic
I (461) spi_flash: flash io: dio
W (465) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (480) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (550) wifi station: ESP_WIFI_MODE_STA
Init turn on
Finish turn on
Init turn on
```

### More Information.

I don't know if this is related to #7602 but, I tried the same code with an S3 and stills freezes or crash.

Xartrick, 2023-08-03:
RMT and Wi-Fi have issue working together, see [led_strip - RMT-based driver for WS2812B/SK6812/APA106 LED strips — esp-idf-lib 1.0 documentation](https://esp-idf-lib.readthedocs.io/en/latest/groups/led_strip.html).

> If you try to use this driver simultaneously with Wi-Fi, you may encounter RMT transmission bugs. To avoid them, simply initialize device from the task bound to the second processor core.

You can run the task interacting with you led strip on another core by using [xTaskCreatePinnedToCore()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html#_CPPv423xTaskCreatePinnedToCore14TaskFunction_tPCKcK22configSTACK_DEPTH_TYPEPCv11UBaseType_tPC12TaskHandle_tK10BaseType_t).

guirespi, 2023-08-08:
I'll try it in my example and will comment the results.

Sherry616, 2024-03-20:
Hi @guirespi, could you please share the latest status of this issue? Thanks.

guirespi, 2024-03-20:
Hi @Sherry616, I've already separated the WIFI and RMT tasks into different cores. The bug is less frequent but still happen. It's been a couple months since I worked on the project that uses the RMT driver and WIFI, but, I remember I use Nimble BLE too. So, do I need to keep the Nimble task in another core too?

suda-morris, 2024-03-21:
There're some limitations you may want to know when using the RMT driver. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/rmt.html#faq Although this documentation is written for the new RMT driver which only appears in the esp-idf v5.x

For the legacy driver you're using the esp-idf v4.x, I would provide the following suggestions:

*  increase the `rmt_config_t::mem_block_num` 
* Install the RMT driver on a separate CPU core to avoid competing for the same CPU resources with other interrupt heavy peripherals (e.g. WiFi, Bluetooth).

BTW, can you use a logic analyzer to capture the signal on the RMT GPIO when it fails to blink the LED? Can you also provide your crash log?

guirespi, 2024-03-21:
Hi,

For now it's quite difficult to use the logic analyzer on the project. The logs were provided in this issue, the program just stops waiting for an interruption.

Regards.

Alvin1Zhang, 2024-12-23:
Thanks for reporting, since 4.4 is end of life, suggest to update to more newer versions, feel free to reopen if the issue still happens

closed 2024-12-23 by Alvin1Zhang