espressif/esp-idf #11908: TinyUSB support for ESP32-C6

DanTup, 2023-07-19:
### Describe the solution you'd like.

I wanted to use some ESP32-C6's (that I already have) for some new projects that involve USB (possibly both HID and serial). I had seen some USB examples in the IDF repository and had assumed they'd work for all modern ESP32 boards.

However, now I see that the examples (and tinyusb components) only have ESP32-S2/ESP32-S3 as the target and trying to use them on a C6 results in errors like:

```
CMake Error at /Users/danny/Dev/Microcontrollers/ESP-IDF-stable/esp-idf/tools/cmake/build.cmake:540 (message):
  HINT: Component "espressif/esp_tinyusb" has suitable versions for other
  targets: "esp32s2", "esp32s3".  Is your current target "esp32c6" set
  correctly?
```

I understand these modules have different CPUs, though I have no idea of the scale of work to support the C6 (it seems like tinyusb does support some RISC-V chips already?).

suda-morris, 2023-07-20:
ESP32C6 doesn't have the USB-OTG IP, what's why TinyUSB is not supported.

DanTup, 2023-07-20:
Oh, I see. I thought USB OTG was for being a USB Host (my plan was to only be a USB device). Is it possible to use C6 as a USB device at all (eg. without using TinyUSB)?

If not, are the S2/S3 the only boards that have the hardware to support this? I was trying to limit the number of different boards I use and had picked the C6 because of Zigbee/Thread support (and I had naively assumed the C6 being new was an improved version of the earlier boards 😄).

Thanks!

suda-morris, 2023-07-20:
> Is it possible to use C6 as a USB device at all (eg. without using TinyUSB)?

C6 can work as a USB CDC Device thanks to another USB-related peripheral called "USB-Serial-JTAG". https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf#usj

You may want to  check this example for the usage: https://github.com/espressif/esp-idf/blob/master/examples/system/console/basic/main/console_example_main.c#L110

DanTup, 2023-07-20:
Thanks - I'll take a look, but it sounds like it won't support being a HID device?

Just to ensure I understand correctly - the S3 has specific hardware that is required to be a USB device (like HID or MSC) that isn't present on the C6? I had assumed that as long as USB is connected to the chip that any kind of use would be possible from software, but it sounds like that was a bad assumption (I'm a newbie in this area) :-)

I suspect the best thing will be to pick up some S3 boards. I presume ESP32-S3 is the newest available board that will support this? 

Thank you!

DanTup, 2023-07-26:
I picked up some S2 boards and will experiment with them. Thanks!

closed 2023-07-26 by DanTup