espressif/esp-idf #11909: flasher_args.json is missing entry for `bootloader` when built with `secure boot v2` enabled.

rretanubun, 2023-07-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.5-183-g99279e7a14

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

When I issue `idf.py build` with an sdkconfig that contains these extra flags, I expect the generated `flasher_args.json` to continue to contain an entry for the `bootloader`
```
CONFIG_APP_BUILD_TYPE_APP_2NDBOOT=y
CONFIG_APP_BUILD_BOOTLOADER=y
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_SIZE=y
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
CONFIG_BOOTLOADER_WDT_ENABLE=y
CONFIG_BOOTLOADER_FLASH_XMC_SUPPORT=y
CONFIG_SECURE_BOOT_SUPPORTS_RSA=y
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_ESP_SYSTEM_PANIC_PRINT_REBOOT=y
CONFIG_ESP_COREDUMP_CHECK_BOOT=y
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=y
CONFIG_ESP32S2_PANIC_PRINT_REBOOT=y

// (I think) these are the extra flags that made the difference
>> CONFIG_SECURE_BOOT=y
>> CONFIG_SECURE_BOOT_V2_ENABLED=y
>> CONFIG_SECURE_SIGNED_ON_BOOT=y
```

### What is the actual behavior?

I think the flasher args files are partially generated, but somehow not `collated` together correctly, here are samples of the generated files in the failing case.

### `flash_bootloader_args` (generated, but not being taken in by the other files?)
--flash_mode dio --flash_freq 80m --flash_size 16MB
0x1000 bootloader/bootloader.bin

### `flash_project_args` (missing `bootloader.bin` entry)
--flash_mode dio --flash_freq 80m --flash_size 16MB
0x450000 bayview.bin
0xd000 partition_table/partition-table.bin
0x27000 ota_data_initial.bin

### `flash_args.in` (missing `bootloader.bin` entry)
--flash_mode dio --flash_freq 80m --flash_size 16MB
0x450000 $<TARGET_PROPERTY:__idf_build_target,EXECUTABLE_NAME>.bin
0xd000 partition_table/partition-table.bin
0x27000 ota_data_initial.bin

All this leads to `flasher_args.json` not having an entry for the bootloader.

### Steps to reproduce.

1. Build a project without secure boot v2 and observe that `flasher_args.json` (and `flash_*` files) are generated okay.
2. 1. Build a project with secure boot v2 enabled and observe that `flasher_args.json` (and `flash_*` files) are generated with missing `bootloader.bin`.

### More Information.

If needed, I am okay with having a custom `CONFIG_ADD_BOOTLOADER_TO_FLASHER_ARGS` sdkconfig so I can trigger this manually, I am at a lost in navigating the CMAKE tooling around the generation of `flasher_args` files.

KonstantinKondrashov, 2023-07-20:
Hi @rretanubun!
It is not a bug, it is done intentionally to split the flashing bootloader and app+partitin_table. **idf.py flash doesn’t flash the bootloader if Secure Boot is enabled**. The build system prints a command that needs to use to flash the bootloader (like esptool.py write_flash ).
See more in the doc https://docs.espressif.com/projects/esp-idf/en/v4.4.5/esp32/security/secure-boot-v2.html#how-to-enable-secure-boot-v2
Thanks.

rretanubun, 2023-07-20:
@KonstantinKondrashov: 

Understood. I am not saying it is a bug either (apologies if the issue filing implied it).

In our production system, we use `idf.py build` to prepare the binary artifacts and then we take `flasher_args.json` and post process it with other tooling for flashing things in the correct order (e.g. a manual step where we sign using HSM key) without using `idf.py flash`.

I understand that there are good reasons to split the flashing process. I need advice on a good place to do a `manual override` to support our system (maybe in `/esp-idf/components/esptool_py/project_include.cmake`?) that is controlled by a custom sdkconfig flag to make `flasher_args.json` contain the bootloader entry.

Thank you for your time!

rretanubun, 2023-07-20:
Actually, I think I found the solution I am looking for https://github.com/espressif/esp-idf/commit/873901e7aababb3e183766bd5f476eaddb8b1cc0

Thanks for everyone's help.

closed 2023-07-20 by rretanubun