espressif/esp-idf #11911: `httpd_req_get_url_query_str` does not work with websockets

kevinhikaruevans, 2023-07-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.0

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-devkitc-1

### Power Supply used.

USB

### What is the expected behavior?

If it returns `ESP_OK`, it should write the query string to the buffer passed in.

If it is not supported with websockets, it should return an error code.

### What is the actual behavior?

It does not write anything to the output (or maybe it's writing an empty string).

### Steps to reproduce.

1. Create HTTP server with a websocket endpoint
2. Add a query string to the ws endpoint, e.g. `/ws?something=abcd`
3. Observe that `httpd_req_get_url_query_len(req) > 0` and `httpd_req_get_url_query_str == ESP_OK`, but the query string is empty

### Debug Logs.

```plain
N/A
```

kevinhikaruevans, 2023-07-19:
Here's roughly the code that I'm using:

```c
static esp_err_t ws_get_identifier(httpd_req_t* req, char* dest, size_t dest_size) {
    if (req == NULL) {
        return ESP_OK;
    }

    char buffer[128] = {0};
    size_t buffer_length = 0;

    // first check query string
    buffer_length = httpd_req_get_url_query_len(req);

    if (buffer_length > sizeof(buffer)) {
        // TODO: off by 1 err here?
        ESP_LOGW(TAG, "query length exceeds buffer length");
        return ESP_ERR_INVALID_SIZE;
    }

    if (buffer_length > 0) {
        ESP_RETURN_ON_ERROR(httpd_req_get_url_query_str(req, buffer, buffer_length + 1), TAG, "cannot get query string");

        ESP_LOGI(TAG, "query string = '%s'", buffer);  // this is returning printing "query string = ''"
    }

    return ESP_OK;
}
```

chipweinberger, 2023-07-19:
Ya it's pretty annoying. I use a workaround using the user_ctx to store the uri

```
    // this is a bit of a hack. For some reason, incoming websocket
    // packets have httpd_req_t->uri && httpd_req_t->method zeroed out. 
    // So we need some other manner to associate 
    // httpd_req_t -> pd_registered_api_t. user_ctx to the rescue =)
    // The user_ctx is *per http endpoint registration*.
    char* user_ctx = NULL;
    user_ctx = (char*) calloc(strlen(uri) + 1, 1);
    memcpy(user_ctx, uri, strlen(uri));

    httpd_uri_t http_uri = {
        .uri = uri,
        .method = ....,
        .handler =  ....,
        .user_ctx = "/websocket/my-socket"
        .is_websocket = true,
        .handle_ws_control_frames = ...,
        .supported_subprotocol = ...,
    };

    esp_err_t err = httpd_register_uri_handler(server, &http_uri);
```

ESP-YJM, 2023-10-18:
@kevinhikaruevans Any update for this issue. I think @chipweinberger solution could solve your issue.

kevinhikaruevans, 2023-10-18:
I found an alternative work around. 

I think `httpd_req_get_url_query_str` should *not* return ESP_OK in this case however, since it's not actually working.

hrushikesh430, 2025-01-03:
Hi @kevinhikaruevans, I was trying to figure out your problem, could you please send the patch file of your code, that will help me to reproduce your issue.

Alvin1Zhang, 2025-01-10:
Thanks for reporting, fix is available at https://github.com/espressif/esp-idf/commit/bbb8b849e2eb495ff725eff1d3ea3a36a859330c. Feel free to reopen.

closed 2025-01-10 by Alvin1Zhang