espressif/esp-idf #11912: I2C sample code issue

Lalithrahulj, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF version.
v5.0.1 

Operating System used.
Windows

How did you build your project?
Command line with idf.py

Development Kit.
ESP32-S3-mini-1 

Power Supply used.
USB

What is the expected behavior?
I2C SCL and SDA Lines to remain in idle state when no transaction taking place.

I have interfaced a I2C slave with the ESP32 S3 master and built the project on the i2c_sample example.
When I Try to read data continuously on the I2C line, I started getting Timeout error.
On the Scope I observed that SDA lines are toggling even when no transaction occurs.

When I try to interface the same in arduino IDE, Everything seems to work fine and the SDA lines are also stable

Lalithrahulj, 2023-07-20:
Here is the initialization I am doing, Freq is set to 100KHz. Just after do this part I see the SDA Line starts toggling.
[image]

        int i2c_master_port = 1;
         i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = GPIO_NUM_4,         // select GPIO specific to your project
        .scl_io_num = GPIO_NUM_5,         // select GPIO specific to your project
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = I2C_MASTER_FREQ_HZ,  // select frequency specific to your project
        .clk_flags = 0,                          // you can use I2C_SCLK_SRC_FLAG_* flags to choose i2c source clock here
    };
    i2c_param_config(i2c_master_port, &conf);
    

    ret = i2c_driver_install(i2c_master_port, conf.mode, I2C_MASTER_RX_BUF_DISABLE, I2C_MASTER_TX_BUF_DISABLE, 0);
    if(ret != ESP_OK)
    {
        ESP_LOGE(TAG,"I2C Installation Failed...");
    }
    ESP_LOGI(TAG,"I2C Installation Success...");

o-marshmallow, 2023-07-20:
Hi @Lalithrahulj ,

Can you try to `cherry-pick` the following commit to see if your issue disappears ed70f1c66b31429b230ed194f1cd7f693c16332e

Lalithrahulj, 2023-07-20:
Hi @o-marshmallow ,

I will try doing that and update the response!!

o-marshmallow, 2023-07-21:
@Lalithrahulj Which I2C device are you using? Maybe the device is performing clock stretching, which forces the clock signal to low.

@mythbuster5 Any idea?

Lalithrahulj, 2023-07-21:
Hi @o-marshmallow Today,I reinstalled the IDF with v5.1 and performed the cherry-pick. I observe that the clock is clocking now but the SDA line is still unstable after reading the 1st data from the slave..
I am attaching the responses I got! 

1.Response before Cherry-pick:
[image]

2.Response After Cherry-pick after reading 1st response from slave:
[image]

[image]
[image]

Lalithrahulj, 2023-07-21:
@o-marshmallow 
I have a update: Tested with ESP32 S3 Wroom and the I2C Lines are stable and communicating fine!  I'll try to change the ESP32-S3-mini Hardware and test and update the response!

Lalithrahulj, 2023-07-24:
@o-marshmallow The Cherry-pick Worked on ESP32 S3. Closing the Issue.

o-marshmallow, 2023-07-24:
@Lalithrahulj Great, thanks for the confirmation!