espressif/esp-idf #11913: Installation fails with online installer

rbruneau-altyor, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Online installer should allow to install ESP-IDF without exit code

### What is the actual behavior?

Whenever I try to install ESP-IDF with online installer, I get the following error message:
`Installation has failed with exit code 1`

I am using universal windows installer v2.21, and running it without antivirus or admin rights does not change the behavior. 

### Steps to reproduce.

1. Download online installer v2.21
2. Install ESP-IDF

### More Information.

I actually want to use the installer to install amazon freeRTOS custom ESP-IDF version, but cannot use it as it fails with exit code 1. Even when downloading ESP-IDF through the installer, I get the same error.

dobairoland, 2023-07-20:
@rbruneau-altyor Could you please share the installation logs? The above error message is very general and we cannot help based on that.

dobairoland, 2023-07-20:
In case you don't have the logs and/or you decide to re-run the installer then a newer version (2.22) has just been released a couple of minutes ago. Please try it.

rbruneau-altyor, 2023-07-20:
That is when I try to install amazon-freertos esp-idf 4.2:

`2023-07-20 13:40:08.065   Log opened. (Time zone: UTC+02:00)
2023-07-20 13:40:08.065   Setup version: Inno Setup version 6.2.2
2023-07-20 13:40:08.065   Original Setup EXE: C:\Users\rbruneau\Downloads\esp-idf-tools-setup-online-2.22.exe
2023-07-20 13:40:08.065   Setup command line: /SL5="$E0B1C,3283833,844288,C:\Users\rbruneau\Downloads\esp-idf-tools-setup-online-2.22.exe" 
2023-07-20 13:40:08.065   Windows version: 10.0.19044  (NT platform: Yes)
2023-07-20 13:40:08.065   64-bit Windows: Yes
2023-07-20 13:40:08.065   Processor architecture: x64
2023-07-20 13:40:08.065   User privileges: None
2023-07-20 13:40:08.067   Administrative install mode: No
2023-07-20 13:40:08.067   Install mode root key: HKEY_CURRENT_USER
2023-07-20 13:40:08.067   64-bit install mode: Yes
2023-07-20 13:40:09.677   Created temporary directory: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp
2023-07-20 13:40:09.683   -- DLL function import --
2023-07-20 13:40:09.683   Function name: idpAddFile
2023-07-20 13:40:09.683   DLL name: files:idp.dll
2023-07-20 13:40:09.683   Extracting temporary file: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.690   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.690   Importing the DLL function.
2023-07-20 13:40:09.737   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.737   -- DLL function import --
2023-07-20 13:40:09.737   Function name: idpAddFileComp
2023-07-20 13:40:09.737   DLL name: files:idp.dll
2023-07-20 13:40:09.737   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.737   Importing the DLL function.
2023-07-20 13:40:09.737   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.737   -- DLL function import --
2023-07-20 13:40:09.737   Function name: idpAddMirror
2023-07-20 13:40:09.737   DLL name: files:idp.dll
2023-07-20 13:40:09.737   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.737   Importing the DLL function.
2023-07-20 13:40:09.737   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.737   -- DLL function import --
2023-07-20 13:40:09.737   Function name: idpAddFtpDir
2023-07-20 13:40:09.737   DLL name: files:idp.dll
2023-07-20 13:40:09.737   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.737   Importing the DLL function.
2023-07-20 13:40:09.737   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.737   -- DLL function import --
2023-07-20 13:40:09.737   Function name: idpAddFtpDirComp
2023-07-20 13:40:09.737   DLL name: files:idp.dll
2023-07-20 13:40:09.737   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.737   Importing the DLL function.
2023-07-20 13:40:09.737   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.737   -- DLL function import --
2023-07-20 13:40:09.737   Function name: idpClearFiles
2023-07-20 13:40:09.737   DLL name: files:idp.dll
2023-07-20 13:40:09.737   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.737   Importing the DLL function.
2023-07-20 13:40:09.737   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.738   -- DLL function import --
2023-07-20 13:40:09.738   Function name: idpFilesCount
2023-07-20 13:40:09.738   DLL name: files:idp.dll
2023-07-20 13:40:09.738   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.738   Importing the DLL function.
2023-07-20 13:40:09.738   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.738   -- DLL function import --
2023-07-20 13:40:09.738   Function name: idpFtpDirsCount
2023-07-20 13:40:09.738   DLL name: files:idp.dll
2023-07-20 13:40:09.738   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.738   Importing the DLL function.
2023-07-20 13:40:09.738   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.738   -- DLL function import --
2023-07-20 13:40:09.738   Function name: idpFileDownloaded
2023-07-20 13:40:09.738   DLL name: files:idp.dll
2023-07-20 13:40:09.738   Dest DLL name: C:\Users\rbruneau\AppData\Local\Temp\is-25UCV.tmp\idp.dll
2023-07-20 13:40:09.738   Importing the DLL function.
2023-07-20 13:40:09.738   Successfully imported the DLL function. Delay loaded? No
2023-07-20 13:40:09.738   -- DLL function import --
[... 807 lines cut ...]
                              action_func(args)
                            File "C:\Workspace\amazon-freertos\vendors\espressif\esp-idf\tools\idf_tools.py", line 1268, in action_install_python_env
                              subprocess.check_call(run_args, stdout=sys.stdout, stderr=sys.stderr)
                            File "subprocess.py", line 413, in check_call
                          subprocess.CalledProcessError: Command '['C:\\Workspace\\Espressif\\amazon-esp-idf-v4.2\\python_env\\idf4.2_py3.11_env\\Scripts\\python.exe', '-m', 'pip', 'install', '--no-warn-script-location', '-r', 'C:\\Workspace\\amazon-freertos\\vendors\\espressif\\esp-idf\\requirements.txt']' returned non-zero exit status 1.
                          
2023-07-20 13:41:09.778   --------
2023-07-20 13:41:09.779   Message box (OK):
[... cut ...]
                          Display the installation log now?
2023-07-20 13:41:27.786   User chose Yes.
2023-07-20 13:41:27.830   CurStepChanged raised an exception.
2023-07-20 13:41:27.944   Need to restart Windows? No
2023-07-20 13:41:27.948   -- Run entry --
2023-07-20 13:41:27.948   Run as: Original user
2023-07-20 13:41:27.948   Type: ShellExec
2023-07-20 13:41:27.948   Filename: 
2023-07-20 13:41:27.973   -- Run entry --
2023-07-20 13:41:27.973   Run as: Original user
2023-07-20 13:41:27.973   Type: ShellExec
2023-07-20 13:41:27.973   Filename: 
`

rbruneau-altyor, 2023-07-20:
Same problem with v2.22 installer

dobairoland, 2023-07-20:
It looks like you are trying to install v4.2 and not v5.1 mentioned in the report. The workaround for the above issue has already been added into the v4.2 release branch: https://github.com/espressif/esp-idf/commit/9e9cbd541b1a46524b3ab250ba82bb544091fd08. Your release probably doesn't have it. Please check if you are able to get any newer bugfix version. Unfortunately we are not able to add more fixes or workarounds to v4.2 because it reached its end-of-life and not supported anymore.

You might be able to work around the issue if you can apply the above changes to your ESP-IDF copy.

Also, you might be able to find a much older offline installer with the required ESP-IDF version included - this won't work with custom ESP-IDF copies.

closed 2023-07-20 by dobairoland

giorgioprovenzale, 2023-09-04:
I have the same issue on 4.4.1, with multiple  `Error compiling Cython file` like the log reported