espressif/esp-idf #11914: Socket send error

Coloryr, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.3-182-gf2973dc4c6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-CAM

### Power Supply used.

USB

### What is the expected behavior?

[image]

### What is the actual behavior?

Socket send data not restart

### Steps to reproduce.

1. Run
2. Open `Socket`
3. Send data

### More Information.

[image]
[image]
[image]

espressif-abhikroy, 2023-08-04:
It appears that there might be an issue with an invalid address being sent in one of the ```socket_server_send()``` calls. 
Kindly investigate the matter thoroughly. 
If the problem persists, I would appreciate it if you could provide a simple example that can help recreate the issue, facilitating further troubleshooting. 

Thank you.

Coloryr, 2023-08-04:
> It appears that there might be an issue with an invalid address being sent in one of the `socket_server_send()` calls. Kindly investigate the matter thoroughly. If the problem persists, I would [...]

[softAP.zip](https://github.com/espressif/esp-idf/files/12259001/softAP.zip)

Coloryr, 2023-08-08:
Is look like the `pic->buf` is NULL.

closed 2023-08-08 by Coloryr