espressif/esp-idf #11915: sha256的数据放在PSRAM时计算不正确

dengbq1234, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Esp32-s3-krovo-3

### Power Supply used.

External 5V

### What is the expected behavior?

通过mbedtls_sha256*相关的API，能够计算出正确的sha256值，无论数据是放在片内RAM还是PSRAM.

### What is the actual behavior?

数据在片内RAM时，能够计算出正确的sha256值，放在PSRAM时，计算错误.

### Steps to reproduce.

见工程源码:
[hello_world_ks3.zip](https://github.com/espressif/esp-idf/files/12104717/hello_world_ks3.zip)

运行结果:
[image]

正确的sha256是:
3fe3881ad5ad724f2a63c38f26b37c56c0132840b9e8388d950620954777dd6b

### More Information.

必现问题

mahavirj, 2023-07-26:
@dengbq1234 

Can you please check if the following patch helps to fix this issue:

```
diff --git components/mbedtls/port/sha/dma/sha.c components/mbedtls/port/sha/dma/sha.c
index a0fff4acd5..3146fa957a 100644
--- components/mbedtls/port/sha/dma/sha.c
+++ components/mbedtls/port/sha/dma/sha.c
@@ -215,7 +215,6 @@ int esp_sha_dma(esp_sha_type sha_type, const void *input, uint32_t ilen,
 {
     int ret = 0;
     unsigned char *dma_cap_buf = NULL;
-    int dma_op_num = ( ilen / (SOC_SHA_DMA_MAX_BUFFER_SIZE + 1) ) + 1;
 
     if (buf_len > block_length(sha_type)) {
         ESP_LOGE(TAG, "SHA DMA buf_len cannot exceed max size for a single block");
@@ -249,6 +248,15 @@ int esp_sha_dma(esp_sha_type sha_type, const void *input, uint32_t ilen,
         buf = dma_cap_buf;
     }
 
+    /* Number of DMA operations based on maximum chunk size in single operation */
+    uint32_t dma_op_num;
+    if (ilen > 0) {
+        dma_op_num = (ilen + SOC_SHA_DMA_MAX_BUFFER_SIZE - 1) / SOC_SHA_DMA_MAX_BUFFER_SIZE;
+    } else {
+        /* For zero input length, we must allow at-least 1 DMA operation to see
+         * if there is any pending data that is yet to be copied out */
+        dma_op_num = 1;
+    }
 
     /* The max amount of blocks in a single hardware operation is 2^6 - 1 = 63
        Thus we only do a single DMA input list + dma buf list,
```

dengbq1234, 2023-07-31:
D:\Espressif\frameworks\esp-idf-v4.4.4>git apply ..\..\sha256.patcherror: mbedtls/port/sha/dma/sha.c: No such file or directory补丁打不上，要不把完整的sha.c文件发出来吧.

抄送人：dengbq1234 ***@***.***>,  Mention ***@***.***>
主题：Re:_[espressif/esp-idf]_sha256的数据放在PSRAM时计算不正确_(IDFGH-10690)_(Issue_#11915)
日期：2023年07月26日 15点11分

mahavirj, 2023-07-31:
@dengbq1234 

Please try to apply with this command `git apply -p0 </path/to/patch>`

dengbq1234, 2023-08-01:
D:\Espressif\frameworks\esp-idf-v4.4.4>git apply -p0 ..\..\sha256.patcherror: corrupt patch at line 28
D:\Espressif\frameworks\esp-idf-v4.4.4>more ..\..\sha256.patchdiff --git components/mbedtls/port/sha/dma/sha.c components/mbedtls/port/sha/dma/sha.cindex a0fff4acd5..3146fa957a 100644--- components/mbedtls/port/sha/dma/sha.c+++ components/mbedtls/port/sha/dma/sha.c@@ -215,7 +215,6 @@ int esp_sha_dma(esp_sha_type sha_type, const void *input, uint32_t ilen, {     int ret = 0;     unsigned char *dma_cap_buf = NULL;-    int dma_op_num = ( ilen / (SOC_SHA_DMA_MAX_BUFFER_SIZE + 1) ) + 1;
     if (buf_len > block_length(sha_type)) {         ESP_LOGE(TAG, "SHA DMA buf_len cannot exceed max size for a single block");@@ -249,6 +248,15 @@ int esp_sha_dma(esp_sha_type sha_type, const void *input, uint32_t ilen,         buf = dma_cap_buf;     }
+    /* Number of DMA operations based on maximum chunk size in single operation */+    uint32_t dma_op_num;+    if (ilen > 0) {+        dma_op_num = (ilen + SOC_SHA_DMA_MAX_BUFFER_SIZE - 1) / SOC_SHA_DMA_MAX_BUFFER_SIZE;+    } else {+        /* For zero input length, we must allow at-least 1 DMA operation to see+         * if there is any pending data that is yet to be copied out */+        dma_op_num = 1;+    }
     /* The max amount of blocks in a single hardware operation is 2^6 - 1 = 63        Thus we only do a single DMA input list + dma buf list,
D:\Espressif\frameworks\esp-idf-v4.4.4>

日期：2023年07月31日 17点34分

closed 2023-08-02 by espressif-bot

mahavirj, 2023-08-02:
Fix has been merged with https://github.com/espressif/esp-idf/commit/735c0c325b532b5798099e806a2a0a44050a835f, you can try on master branch now.