espressif/esp-idf #11916: Bluedroid 128 bit UUID Guide not working

Svahnen-Perific, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to follow the Readme [here](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_client#settings-for-uuid128) but cant get past step 3, since it fails to build with different variations this error:
variable "remote_filter_service_uuid" is not a type name

My code below:
`static uint8_t gatts_service_uuid128[ESP_UUID_LEN_128] = {0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xe9, 0xfe, 0x00, 0x00};
static esp_bt_uuid_t remote_filter_service_uuid = {
    .len = ESP_UUID_LEN_128,  // Changed from ESP_UUID_LEN_16
    .uuid = {
        //.uuid128 = {0xfb,0x34,0x9b,0x5f,0x80,0x00,0x00,0x80,0x00,0x10,0x00,0x00,0xe9,0xfe,0x00,0x00},
        .uuid128 = {0},
    },
};
// memcpy(&remote_filter_service_uuid.uuid.uuid128, gatts_service_uuid128, ESP_UUID_LEN_128); // Doesn't work
memcpy(remote_filter_service_uuid.uuid.uuid128, gatts_service_uuid128, ESP_UUID_LEN_128);  // Doesn't work
// esp_bt_uuid_t *remote_filter_service_uuid_pointer;
// memcpy((remote_filter_service_uuid_pointer->uuid.uuid128), gatts_service_uuid128, ESP_UUID_LEN_128);// Doesn't work

static uint8_t gatts_char_uuid128[ESP_UUID_LEN_128] = {0x01, 0x96, 0x12, 0x16, 0x54, 0x92, 0x75, 0xb5, 0xa2, 0x45, 0xfd, 0xab, 0x39, 0xc4, 0x4b, 0xd4};
static esp_bt_uuid_t remote_filter_char_uuid = {
    .len = ESP_UUID_LEN_128,  // Changed from ESP_UUID_LEN_16
    .uuid = {
        //.uuid128 = {0x01, 0x96, 0x12, 0x16, 0x54, 0x92, 0x75, 0xb5, 0xa2, 0x45, 0xfd, 0xab, 0x39, 0xc4, 0x4b, 0xd4},
        .uuid128 = {0},
    },
};
// memcpy(&remote_filter_char_uuid.uuid.uuid128, gatts_char_uuid128, ESP_UUID_LEN_128); // Doesn't work
memcpy(remote_filter_char_uuid.uuid.uuid128, gatts_char_uuid128, ESP_UUID_LEN_128);  // Doesn't work`

It would help immensely if the Readme would include a working example where its easier to see where and how to implement step 3 (memcpy).

zhp0406, 2023-08-09:
The provided code appears to be correct:

```
memcpy(remote_filter_service_uuid.uuid.uuid128, gatts_service_uuid128, ESP_UUID_LEN_128); //correct
memcpy(remote_filter_char_uuid.uuid.uuid128, gatts_char_uuid128, ESP_UUID_LEN_128); //correct
```

However, you mentioned that you've encountered issues. Could you please provide some error logs or additional details about the problem you're facing? This information would be helpful in diagnosing and resolving the issue you're experiencing.

I've also tested it by modifying the example "esp-idf/examples/bluetooth/bluedroid/ble/gatt_server," and I didn't encounter any issues. Below is my most recent example code:

```
    case ESP_GATTS_REG_EVT:
        ESP_LOGI(GATTS_TAG, "REGISTER_APP_EVT, status %d, app_id %d", param->reg.status, param->reg.app_id);
        gl_profile_tab[PROFILE_A_APP_ID].service_id.is_primary = true;
        gl_profile_tab[PROFILE_A_APP_ID].service_id.id.inst_id = 0x00;
        gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.len = ESP_UUID_LEN_128;
        static uint8_t gatts_service_uuid128[ESP_UUID_LEN_128] = {0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xe9, 0xfe, 0x00, 0x00};
        memcpy(&gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.uuid.uuid128, gatts_service_uuid128, ESP_UUID_LEN_128); 
        gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.uuid.uuid16 = GATTS_SERVICE_UUID_TEST_A;

        esp_err_t set_dev_name_ret = esp_ble_gap_set_device_name(TEST_DEVICE_NAME);
        if (set_dev_name_ret){
            ESP_LOGE(GATTS_TAG, "set device name failed, error code = %x", set_dev_name_ret);
}
```
You can test the client using the same approach as before with the example "esp-idf/examples/bluetooth/bluedroid/ble/gatt_client."

[image]

Svahnen-Perific, 2023-08-09:
Thx for the respons, we where able to compile by moving memcpy into app_main, the guide never mentioned where it was meant to run, but we placed it after the error checks.

The device we are connecting to is a Nordic nrf5340 configured as server and perf, and it now connects but only prints this:
```
I (6846) GATTC_DEMO: searched device S1-12323XXXX

I (6846) GATTC_DEMO: connect to the remote device.
I (6856) GATTC_DEMO: stop scan successfully
I (7846) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (7846) GATTC_DEMO: REMOTE BDA:
I (7846) GATTC_DEMO: d8 df ef 17 3f 2e 
I (7856) GATTC_DEMO: open success
I (8696) GATTC_DEMO: discover service complete conn_id 0
I (8706) GATTC_DEMO: Get service information from remote device
I (8706) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (8726) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 247, conn_id 0
I (12946) GATTC_DEMO: update connection params status = 0, min_int = 20, max_int = 80,conn_int = 80,latency = 0, timeout = 400
```

But if we connect with the server example from Espressif we get things like for example this `I (2693) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1` and that makes me think it still does not find the services, even tho we can see them if connecting from a phone app.

Svahnen-Perific, 2023-08-09:
I have now tried it together with the example you suggested "esp-idf/examples/bluetooth/bluedroid/ble/gatt_server" and made the same changes you posted to the ESP_GATTS_REG_EVT case. But still get the same output and no mentions of services found.

Client code:
```
static uint8_t gatts_remote_filter_service_uuid128[ESP_UUID_LEN_128] = {0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xe9, 0xfe, 0x00, 0x00};

static esp_bt_uuid_t remote_filter_service_uuid = {
    .len = ESP_UUID_LEN_128,
    .uuid = {
        .uuid128 = {0},
    },
};

static esp_bt_uuid_t remote_filter_char_uuid = {
    .len = ESP_UUID_LEN_16,
    .uuid = {
        .uuid16 = REMOTE_NOTIFY_CHAR_UUID,
    },
};

static esp_bt_uuid_t notify_descr_uuid = {
    .len = ESP_UUID_LEN_16,
    .uuid = {
        .uuid16 = ESP_GATT_UUID_CHAR_CLIENT_CONFIG,
    },
};
```
```
        case ESP_GATTC_SEARCH_RES_EVT: {
            printf("GATT service discovery result is got\n");
            ESP_LOGI(GATTC_TAG, "SEARCH RES: conn_id = %x is primary service %d", p_data->search_res.conn_id, p_data->search_res.is_primary);
            ESP_LOGI(GATTC_TAG, "start handle %d end handle %d current handle value %d", p_data->search_res.start_handle, p_data->search_res.end_handle, p_data->search_res.srvc_id.inst_id);
            if (p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_128) {
                if (memcmp(p_data->search_res.srvc_id.uuid.uuid.uuid128, gatts_remote_filter_service_uuid128, ESP_UUID_LEN_128) == 0) {
                    ESP_LOGI(GATTC_TAG, "service uuid128 found");
                    get_server = true;
                    gl_profile_tab[PROFILE_A_APP_ID].service_start_handle = p_data->search_res.start_handle;
                    gl_profile_tab[PROFILE_A_APP_ID].service_end_handle = p_data->search_res.end_handle;
                } else {
                    ESP_LOGE(GATTC_TAG, "service not found");
                }
            }
            break;
        }
```
```
memcpy(remote_filter_service_uuid.uuid.uuid128, gatts_remote_filter_service_uuid128, ESP_UUID_LEN_128);
```

Server code:
```
        case ESP_GATTS_REG_EVT:
            ESP_LOGI(GATTS_TAG, "REGISTER_APP_EVT, status %d, app_id %d", param->reg.status, param->reg.app_id);
            gl_profile_tab[PROFILE_A_APP_ID].service_id.is_primary = true;
            gl_profile_tab[PROFILE_A_APP_ID].service_id.id.inst_id = 0x00;
            gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.len = ESP_UUID_LEN_128;
            static uint8_t gatts_service_uuid128[ESP_UUID_LEN_128] = {0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xe9, 0xfe, 0x00, 0x00};
            memcpy(&gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.uuid.uuid128, gatts_service_uuid128, ESP_UUID_LEN_128);
            gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.uuid.uuid16 = GATTS_SERVICE_UUID_TEST_A;

            esp_err_t set_dev_name_ret = esp_ble_gap_set_device_name(TEST_DEVICE_NAME);
            if (set_dev_name_ret) {
                ESP_LOGE(GATTS_TAG, "set device name failed, error code = %x", set_dev_name_ret);
            }
```

Client log:
```
I (4626) GATTC_DEMO: ESP_GATTS_DEMO
I (4636) GATTC_DEMO: 
[... 86 lines cut ...]

I (1054) GATTS_DEMO: the gatts demo char length = 3

I (1054) GATTS_DEMO: prf_char[0] =11

I (1064) GATTS_DEMO: prf_char[1] =22

I (1064) GATTS_DEMO: prf_char[2] =33

I (1074) GATTS_DEMO: CREATE_SERVICE_EVT, status 0,  service_handle 44

I (1084) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40

I (1084) GATTS_DEMO: SERVICE_START_EVT, status 0, service_handle 44

I (1094) GATTS_DEMO: ADD_CHAR_EVT, status 0,  attr_handle 46, service_handle 44

I (1104) GATTS_DEMO: ADD_DESCR_EVT, status 0, attr_handle 47, service_handle 44

I (1164) GATTS_DEMO: ESP_GATTS_CONNECT_EVT, conn_id 0, remote 24:0a:c4:1d:49:72:
I (1174) GATTS_DEMO: CONNECT_EVT, conn_id 0, remote 24:0a:c4:1d:49:72:
I (1344) GATTS_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 32,latency = 0, timeout = 400
I (2634) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (2634) GATTS_DEMO: ESP_GATTS_MTU_EVT, MTU 500
```

zhp0406, 2023-08-10:
@Svahnen-Perific 
The **client-side** has performed a conversion from a 128-bit UUID to a 16-bit UUID. Therefore, please ensure that you include the following code in your implementation:

```
p_data->search_res.srvc_id.uuid.len == ESP_UUID_LEN_16
```

Furthermore, note that the correct code snippet for the `memcpy` operation is:

```
memcpy(gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.uuid.uuid128, gatts_service_uuid128, ESP_UUID_LEN_128);
```

Do not use:

```
memcpy(&gl_profile_tab[PROFILE_A_APP_ID].service_id.id.uuid.uuid.uuid128, gatts_service_uuid128, ESP_UUID_LEN_128);
```

zhp0406, 2023-08-10:
[0001-test-bt-Test-128-bit-uuid-both-in-server-and-client.zip](https://github.com/espressif/esp-idf/files/12310634/0001-test-bt-Test-128-bit-uuid-both-in-server-and-client.zip)
I've written a test for you. You can use "git am 0001-test-bt-Test-128-bit-uuid-both-in-server-and-client.patch" to apply this patch and then proceed with the testing.
@Svahnen-Perific

Svahnen-Perific, 2023-08-10:
After testing with your patch the log shows this:
```
128 bits may have been converted to 16 bits.
service found uuid 16,uuid is 0xfee9
```
Does this mean there is something we are doing wrong or does SDK 5 (master) not yet support 128 bit? Is there another version we should be using?

zhp0406, 2023-08-10:
28_bit_value = 16_bit_value * 296 + Bluetooth_Base_UUID
128_bit_value = 32_bit_value * 296 + Bluetooth_Base_UUID

Svahnen-Perific, 2023-08-10:
I'm sorry but I don't understand what you mean?

zhp0406, 2023-08-10:
@Svahnen-Perific 
according to BLUETOOTH CORE SPECIFICATION Version 5.4 | Vol 3, Part B page 1181 
A 128-bit UUID can be transformed into a 16-bit UUID,so:
128-bit UUID {0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xe9, 0xfe, 0x00, 0x00} converts to the 16-bit UUID 0xfee9,which means Quintic Corp.

Svahnen-Perific, 2023-08-10:
I see that the UUID you added on the server side does not match the variable gatts_remote_uuid128 in the client file, are they not meant to match each other?

Server:
```
gatts_service_uuid128[ESP_UUID_LEN_128] = {0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0xe9, 0xfe, 0x00, 0x00};
```
Client:
```
gatts_remote_uuid128[ESP_UUID_LEN_128] = {0xA5, 0x7F, 0xC1, 0x01, 0x87, 0xB1, 0x41, 0x7C, 0x90, 0x27, 0x8F, 0x7E, 0x54, 0x65, 0x4C, 0xD4};
```

Svahnen-Perific, 2023-08-11:
Does that mean that in order for us to be able to find the services from Espressif the 128 bit UUID have to be created from the Bluetooth_Base_UUID?
After re-applying the patch it started working between 2 ESP32 using the UUID you supplied but its not working when changed to the one we need to connect too.

zhp0406, 2023-08-11:
@Svahnen-Perific 
question 1: No,You can use any 128-bit UUID.
question 2: can you give more description or a patch?

Svahnen-Perific, 2023-08-11:
We are trying to connect to a Nordic chip, I will send over the documentation we are using.
[How_we_use_Bluetooth_LE_in_Amina_S.pdf](https://github.com/espressif/esp-idf/files/12322328/How_we_use_Bluetooth_LE_in_Amina_S.pdf)

zhp0406, 2023-08-24:
@Svahnen-Perific 
[0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch](https://github.com/espressif/esp-idf/files/12428026/0001-fix-bt-bluedroid-Fix-uuid-length-error-when-uuid-is-.patch)
[0002-fix-bt-bluedroid-Test-BLE-128-bit-uuid.patch](https://github.com/espressif/esp-idf/files/12428028/0002-fix-bt-bluedroid-Test-BLE-128-bit-uuid.patch)
Converting a 128-bit UUID to a 16-bit UUID might mislead users. Therefore, we have disabled the automatic conversion from a 128-bit UUID to a 16-bit UUID. I have provided two patches: the first patch disables the automatic conversion feature, and the second patch is for testing. 
thanks you for report the problems.

meakashrao, 2024-06-17:
@Svahnen-Perific I updated to the stable release of 5.2.2 and still this is a trouble? I am applying the patch file to the location where my ESP files are located and the project files are located in a different directory. 

[image]

It would be helpful if you could share as to where did you apply the patch file to get 128 bit UUID's working