espressif/esp-idf #11920: ESP32-S3 Assert failed: tlsf_free tlsf.c:1119 (!block_is_free(block) && "block already marked as free")

Stencht, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-DevKitC-1 (ESP32-S3-WROOM-1 N8R8)

### Power Supply used.

USB

### What is the expected behavior?

I have an application which runs an edge-impulse model on an ESP32-S2 (IDF-version v5.0-rc1). I expect that this would also work on an ESP32-S3-WROOM-1 N8R8. 

### What is the actual behavior?

The same application, _on the same idf-version v5.0-rc1_, crashes on remove_free_block (esp-idf/components/heap/tlsf/tlsf.c:337) - which is called from ei_free(void *ptr) in run_classifier().

The error when I tried it with v5.0-rc1: `Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.`

In an attempt to fix this, I tried the same application but with esp-idf v5.1. 
This resulted in following error: `assert failed: tlsf_free tlsf.c:1119 (!block_is_free(block) && "block already marked as free")`

### Steps to reproduce.

Not yet

### Debug Logs.

```plain
assert failed: tlsf_free tlsf.c:1119 (!block_is_free(block) && "block already marked as free")

Backtrace: 0x40375e2a:0x3fccf620 0x40380cbd:0x3fccf640 0x403889fd:0x3fccf660 0x40386f22:0x3fccf780 0x40386dcc:0x3fccf7a0 0x40376932:0x3fccf7c0 0x40388a2d:0x3fccf7e0 0x420107ad:0x3fccf800 0x42010bb2:0x3fccf820 0x4200e7c7:0x3fccf840 0x4200ef3f:0x3fccf880 0x4200f015:0x3fccf920 0x403837d9:0x3fccf9e0
0x40375e2a: panic_abort at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/esp_system/panic.c:452

0x40380cbd: esp_system_abort at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x403889fd: __assert_func at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/newlib/assert.c:81

0x40386f22: tlsf_free at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/heap/tlsf/tlsf.c:1119 (discriminator 1)

0x40386dcc: multi_heap_free_impl at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/heap/multi_heap.c:231

0x40376932: heap_caps_free at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/heap/heap_caps.c:388

0x40388a2d: free at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/newlib/heap.c:39

0x420107ad: ei_free(void*) at C:/Projects/shower_monitor/edge-impulse-sdk/porting/espressif/ei_classifier_porting.cpp:85

0x42010bb2: trained_model_reset(void (*)(void*)) at C:/Projects/shower_monitor/tflite-model/trained_model_compiled.cpp:562 (discriminator 3)

0x4200e7c7: run_nn_inference(ei_impulse const*, ei::ei_matrix*, ei_impulse_result_t*, void*, bool) at C:/Projects/shower_monitor/edge-impulse-sdk/classifier/inferencing_engines/tflite_eon.h:111
 (inlined by) run_nn_inference(ei_impulse const*, ei::ei_matrix*, ei_impulse_result_t*, void*, bool) at C:/Projects/shower_monitor/edge-impulse-sdk/classifier/inferencing_engines/tflite_eon.h:228

0x4200ef3f: process_impulse at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:117
 (inlined by) process_impulse at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:211
 (inlined by) process_impulse at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:140

0x4200f015: vInferencingTask(void*) at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:564 (discriminator 13)
 (inlined by) vInferencingTask at C:/Projects/shower_monitor/main/services/analyser.cpp:232 (discriminator 13)

0x403837d9: vPortTaskWrapper at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

### More Information.

- I tried this both with the original idf-version of the application (v5.0-rc1) as with the latest stable version (v5.1).
- The application also uses WiFi, tried to remove this from the application but same situation
- It might be an error with the edge impulse sdk, but I don't see where this is different from the ESP32-S2 implementation.

Any ideas are welcome! I am hoping I am overlooking a very obvious difference between ESP32-S2 and ESP32-S3...

wuyuanyi135, 2023-07-21:
i ran into such a problem when stackoverflow or heap corruption occurs. try turn on stackoverflow detection and heap corruption detection in the menuconfig

Stencht, 2023-07-22:
Thank you for the tip @wuyuanyi135.
I see now that there is a heap corruption... I'm trying to hunt the issue now, but no experience with this yet, so tips are very welcome :)
I've also tried to increase the stack size for the task (and did a hail Mary), but no success.

&I'm still baffled that this issue occurs with the ESP32-S3, but not with the ESP32-S2...

```
CORRUPT HEAP: Bad tail at 0x3fcc59ff. Expected 0xbaad5678 got 0x51ff66ff

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)

Backtrace: 0x40375e4e:0x3fcd0570 0x40380ce1:0x3fcd0590 0x40388bdd:0x3fcd05b0 0x40386f11:0x3fcd06d0 0x40376956:0x3fcd06f0 0x40388c0d:0x3fcd0710 0x420107ad:0x3fcd0730 0x42010bb2:0x3fcd0750 0x4200e7c7:0x3fcd0770 0x4200ef3f:0x3fcd07b0 0x4200f015:0x3fcd0850 0x403837fd:0x3fcd0910
0x40375e4e: panic_abort at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/esp_system/panic.c:452

0x40380ce1: esp_system_abort at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40388bdd: __assert_func at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/newlib/assert.c:81

0x40386f11: multi_heap_free at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/heap/multi_heap_poisoning.c:276 (discriminator 1)

0x40376956: heap_caps_free at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/heap/heap_caps.c:388

0x40388c0d: free at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/newlib/heap.c:39

0x420107ad: ei_free(void*) at C:/Projects/shower_monitor/edge-impulse-sdk/porting/espressif/ei_classifier_porting.cpp:85

0x42010bb2: trained_model_reset(void (*)(void*)) at C:/Projects/shower_monitor/tflite-model/trained_model_compiled.cpp:562 (discriminator 3)

0x4200e7c7: run_nn_inference(ei_impulse const*, ei::ei_matrix*, ei_impulse_result_t*, void*, bool) at C:/Projects/shower_monitor/edge-impulse-sdk/classifier/inferencing_engines/tflite_eon.h:111
 (inlined by) run_nn_inference(ei_impulse const*, ei::ei_matrix*, ei_impulse_result_t*, void*, bool) at C:/Projects/shower_monitor/edge-impulse-sdk/classifier/inferencing_engines/tflite_eon.h:228

0x4200ef3f: process_impulse at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:117
 (inlined by) process_impulse at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:211
 (inlined by) process_impulse at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:140

0x4200f015: vInferencingTask(void*) at c:\projects\shower_monitor\edge-impulse-sdk\classifier/ei_run_classifier.h:564 (discriminator 13)
 (inlined by) vInferencingTask at C:/Projects/shower_monitor/main/services/analyser.cpp:232 (discriminator 13)

0x403837fd: vPortTaskWrapper at C:/Users/benna/esp/esp-idf-v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

Stencht, 2023-07-22:
No solution yet. 
The heap and stack looks OK:
```
I (5263) analyser: Trying to take mutex for InferencingTask...
I (5683) analyser: Running classifier...
I (5683) analyser: heap integrety 1
I (5683) analyser: FreeHeapSize: 159276
I (5693) analyser: Bytes free in stack: 5580
I (6043) analyser: Inference results off/on: 1.00/0.00
I (6043) analyser: Entering InferencingTask FOR-loop...
I (6043) analyser: Trying to take mutex for InferencingTask...
I (6053) analyser: Running classifier...
I (6053) analyser: heap integrety 1
I (6063) analyser: FreeHeapSize: 159012
I (6063) analyser: Bytes free in stack: 5372
CORRUPT HEAP: Bad head at 0x3fcd1954. Expected 0xabba1234 got 0x0018fff3

assert failed: multi_heap_free multi_heap_poisoning.c:276 (head != NULL)
```

I've changed the creation of the FreeRTOS tasks from:

```
xReturned = xTaskCreate(
					vInferencingTask, /* Function that implements the task. */
					"vInferencingTask", /* Text name for the task. */
					EI_TASK_STACK_SIZE, /* Stack size in words, not bytes. */
					( void * ) 1, /* Parameter passed into the task. */
					EI_TASK_PRIORITY, /* Priority at which the task is created, default tskIDLE_PRIORITY. */
					&xHandleInference ); /* Used to pass out the created task's handle. */
```

to
```
xReturned = xTaskCreatePinnedToCore(
					vInferencingTask, /* Function that implements the task. */
					"vInferencingTask", /* Text name for the task. */
					EI_TASK_STACK_SIZE, /* Stack size in words, not bytes. */
					( void * ) 1, /* Parameter passed into the task. */
					EI_TASK_PRIORITY, /* Priority at which the task is created, default tskIDLE_PRIORITY. */
					&xHandleInference, /* Used to pass out the created task's handle. */
                    EI_TASK_CORE_ID ); 
```

But the same result...

Could it have something to do with the dual core of the ESP32-S3 instead of the single core of the ESP32-S2?

wuyuanyi135, 2023-07-22:
It is possible that you have concurrent access to some shared resource without proper locking. If possible, try not to use threading and run everything in series. If the problem is gone, check the racing conditions. For memory access violation it might be a bit harder to locate. But generally if the crash is easily reproducible you can always bisect the faulty code.

wuyuanyi135, 2023-07-23:
```
0x42010bb2: trained_model_reset(void (*)(void*)) at C:/Projects/shower_monitor/tflite-model/trained_model_compiled.cpp:562 (discriminator 3)
0x4200e7c7: run_nn_inference(ei_impulse const*, ei::ei_matrix*, ei_impulse_result_t*, void*, bool) at C:/Projects/shower_monitor/edge-impulse-sdk/classifier/inferencing_engines/tflite_eon.h:111
 (inlined by) run_nn_inference(ei_impulse const*, ei::ei_matrix*, ei_impulse_result_t*, void*, bool) at C:/Projects/shower_monitor/edge-impulse-sdk/classifier/inferencing_engines/tflite_eon.h:228

```
Go to the corresponding line, see what resource is being released. If that is something you recognize, see if it might be released somewhere else. Or if this is the generated code or library code, you may consult the documentation if the it is thread-safe (tensorflow?). If not, make sure not to use it in multiple thread without proper mutex. If you want to use multithreading for data loading, use producer-consumer pattern. If you want to make use of both CPU, check if they have option to enable to prevent race condition. Thats all general tips I could provide.

Stencht, 2023-07-23:
Thanks again for the insights @wuyuanyi135.
The issue is indeed something thread related. I use 2 FreeRTOS tasks, one for loading and one for consuming data:

- A quick combine of those two tasks in one task (so loading and consuming in series, but still in a task), gives the same issue. 
- Just calling that same function (which does the data loading and consuming in series) in the main thread without a task works just fine.

So I'm messing something up. I'm already using a mutex, but I will have to quadruple check everything.
My mind still has a hard time wrapping around why it works on the ESP32-S2 single core though, I would assume that it would also have thrown this error.

Anyway, many thanks for the advise!

wuyuanyi135, 2023-07-23:
esp32-s2 is a single core device so yes it might be attributed to the multi-core nature of s3. 

> A quick combine of those two tasks in one task (so loading and consuming in series, but still in a task), gives the same issue.

Can you try to give sufficiently large stack (like 8192) and pin the created thread to a core as well?

Stencht, 2023-07-23:
I've tried to pin the task to core 0 (and later also to core 1, just to be sure), but no success. 
The stack size is also sufficiently large (4096*3).

I do see some interesting behaviour: I've included some printf statements before the function call which throws the heap corruption (in the external library):
```
ei_printf("Heap integrety %i\n", heap_caps_check_integrity_all(true));               // Added
ei_printf("FreeHeapSize: %zu\n", xPortGetFreeHeapSize());                                // Added
ei_printf("Bytes free in stack: %zu\n", uxTaskGetStackHighWaterMark(NULL));  // Added
vTaskDelay(10);                // Added
config->model_reset(ei_aligned_free); // Heap corruption happens here
```
When the heap corruption occurs I get following printout:
```
Heap integrety 1
FreeHeapSize: 142416
Bytes free in stack: 9628
CORRUPT HEAP: Bad head at 0x3fcd3c80. Expected 0xabba1234 got 0xfffeffab
```
But with these extra lines, the program actually works better. Once the first call of `config->model_reset` is successful (which usually takes a couple of crashes), the program keeps working, no more crashing. Even when the device goes to deepsleep and wakes up. 

After realising this, I removed the print lines and only kept the delay:
```
vTaskDelay(10);                // Added
config->model_reset(ei_aligned_free); // Heap corruption happens here
```
This resulted in a different behaviour, being that it also takes a couple of crashes to work, but on wakeup from deepsleep it crashes again... Increasing the delay to `vTaskDelay(100)`, results in the same behaviour...

Ideas? :)

wuyuanyi135, 2023-07-23:
If timing causes random crash it is almost for sure that there is racing condition. I did not used the library so I could not tell. What if you do not reset the model? I guess you could just leave it for future reuse?

Stencht, 2023-07-24:
After a lot of pain, I've found the issue. 
I've changed
```
if(NOT CMAKE_BUILD_EARLY_EXPANSION)
add_definitions(-DEI_CLASSIFIER_TFLITE_ENABLE_ESP_NN=1) # enables ESP-NN optimizations by Espressif
endif()
```
to
```
if(NOT CMAKE_BUILD_EARLY_EXPANSION)
add_definitions(-DEI_CLASSIFIER_TFLITE_ENABLE_ESP_NN=0) 
endif()
```

Not completely sure why, because EESP32-S3 is supported by ESP_NN. But at the moment I'm just glad I got it working again...
Thank you for your help @wuyuanyi135 .

closed 2023-07-24 by Stencht

wuyuanyi135, 2023-07-24:
good to know that. you may also want to raise the issue to tf developers. they might have some fix and may need to emphasize this in the doc.

srikanthpalvai, 2023-09-06:
> After a lot of pain, I've found the issue. I've changed
>
> ```
> if(NOT CMAKE_BUILD_EARLY_EXPANSION)
> add_definitions(-DEI_CLASSIFIER_TFLITE_ENABLE_ESP_NN=1) # enables ESP-NN optimizations by Espressif [...]
Where can I find out this // enables ESP-NN optimizations by Espressif
is it configured from idf.py menuconfig ?

Divirambling, 2025-10-17:
> After a lot of pain, I've found the issue. I've changed
>
> ```
> if(NOT CMAKE_BUILD_EARLY_EXPANSION)
> add_definitions(-DEI_CLASSIFIER_TFLITE_ENABLE_ESP_NN=1) # enables ESP-NN optimizations by Espressif [...]

https://ecmascript-next8.github.io/ueySgqRl72Nh-nMXD8DS7BPJYdKg3dByQpikqkK2QajyoUYTHSd313ZK7rGFVD0X8luHFIMWQA/