espressif/esp-idf #11924: ESP32-C6 USB Serial (CDC) not working correctly

thedevleon, 2023-07-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-C6-DevKitC-1-N8

### Power Supply used.

USB

### What is the expected behavior?

I am compiling the openthread cli sample (`/examples/openthread/ot_cli`), but instead of using the UART port, I want to use the native USB CDC serial port via the USB-port labeled USB. We want to build a product using the ESP32-C6 that will only require a single USB port for debugging and serial input/output.

Following the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-guides/usb-serial-jtag-console.html), I enabled the USB CDC Serial console by setting `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y`. 

Everything compiles and flashes fine, and the output looks good initially. However, right after the cli task is started, the console is flooded with wierd characters.

Furthermore, if I connect via the UART port instead, I still get logging output (I would expect NO output at all), but I cannot enter any commands.

### What is the actual behavior?

I would expect the CLI to work just like it does with the UART port, and I would expect the UART port not to send anything at all.

### Steps to reproduce.

1. Open the ot_cli example
2. Run menuconfig and set the "Channel for console output" to "USB Serial/JTAG Controller" and "Channel for console secondary output" to "No secondary console".
3. Build, Flash and Monitor
4. Get garbage on the monitor after the cli thread has been started

### Debug Logs.

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x408007fe
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xd14
load:0x4086e610,len:0x2cc4
load:0x40875720,len:0x17ec
entry 0x4086c410
I (23) boot: ESP-IDF v5.1-dirty 2nd stage bootloader    
I (23) boot: compile time Jul 20 2023 16:09:11
I (24) boot: chip revision: v0.0
I (26) boot.esp32c6: SPI Speed      : 40MHz
I (31) boot.esp32c6: SPI Mode       : DIO
I (36) boot.esp32c6: SPI Flash Size : 2MB
[... 97 lines cut ...]
> ����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
> ����������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������������
> ����
```
```

### More Information.

I have tried compiling with the latest master branch, however the sample does no longer compile. I get the following error:

```
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c: In function 'light_sleep_uart_prepare':
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:149:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  149 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:503:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  503 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |                                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:149:61: note: each undeclared identifier is reported only once for each function it appears in
  149 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:503:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  503 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |                                     
```

thedevleon, 2023-07-20:
I also tried compiling the `examples/system/console/basic` sample, which compiles fine in it's stock configuration, but as soon as I change to the "USB Serial/JTAG Controller" for "Channel for console output" in menuconfig, I again get the following error:

```
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c: In function 'light_sleep_uart_prepare':
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:134:61: error: 'CONFIG_ESP_CONSOLE_UART_BAUDRATE' undeclared (first use in this function); did you mean 'CONFIG_ESP_CONSOLE_UART_NUM'?
  134 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:460:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  460 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |                                                                                                                 ^~~~~~~~~~~~~~~~~~~~~~
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:134:61: note: each undeclared identifier is reported only once for each function it appears in
  134 | #define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
      |                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/thele/esp/esp-idf/components/esp_hw_support/sleep_modes.c:460:113: note: in expansion of macro 'UART_FLUSH_US_PER_CHAR'
  460 |             (sleep_duration < (int64_t)((UART_LL_FIFO_DEF_LEN - uart_ll_get_txfifo_len(CONSOLE_UART_DEV) + 1) * UART_FLUSH_US_PER_CHAR) + SLEEP_UART_FLUSH_DONE_TO_SLEEP_US)) {
      |  
```

thedevleon, 2023-07-20:
I was able to get the `examples/system/console/basic` sample to work by fixing the offending line in `sleep_modes.c` to

```
// hotfix?
#ifndef CONFIG_ESP_CONSOLE_UART_BAUDRATE
#define UART_FLUSH_US_PER_CHAR              0
#else
#define UART_FLUSH_US_PER_CHAR              (10*1000*1000 / CONFIG_ESP_CONSOLE_UART_BAUDRATE)
#endif
```

However, the ot-cli sample still prints garbage with this fix. So it has to be something related to the ot-cli sample.

igrr, 2023-07-20:
Hi @thedevleon, ot_cli example doesn't support USB_SERIAL_JTAG peripheral as console, please see the recent discussion in https://github.com/espressif/esp-idf/issues/11896#issuecomment-1640577640.

Thank you for reporting the compilation failure for `examples/system/console/basic` for C6 with `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y`, we have missed adding a test for such combination. Will fix this soon!

thedevleon, 2023-07-20:
Also, it seems the UART port is still used for bootloader output. I remember seeing that there are eFuses to burn to disable this output, but I can't find it. Can you link me to this?

Also, can I use the pins for UART as soon as the bootloader is finished writing output? Or does the output need to be disabled via the eFuses so that those pins are freed up for other uses?

igrr, 2023-07-20:
> I guess in esp_openthread_uart.c, esp_openthread_uart_init or esp_openthread_uart_init_port would need to be adjusted to support calls to usb_serial_jtag_driver_install?

@chshu could you please help with this question?

> Also, it seems the UART port is still used for bootloader output. I remember seeing that there are eFuses to burn to disable this output, but I can't find the documentation for it. Can you link me to this?

This should probably be `EFUSE_UART_PRINT_CONTROL` efuse field, it should be described in the ESP32-C6 technical reference manual.

However, the simpler option might to use a different UART TX pin for communicating with your external device. Since UART pins can be freely configured, you can pass a different TX GPIO number when initializing the UART driver. This way you will still be able to access UART logs (just in case) but the UART TX line going to the external device will not be disturbed by the boot process.

chshu, 2023-07-21:
@thedevleon Could you apply this patch: [ot_cli_usb.patch](https://github.com/espressif/esp-idf/files/12126601/ot_cli_usb.patch), by excuting `git apply ot_cli_usb.patch` under the esp-idf folder. Then the `ot_cli` should work with the USB port.

We will merge the change to esp-idf soon.

thedevleon, 2023-07-21:
> @thedevleon Could you apply this patch: [ot_cli_usb.patch](https://github.com/espressif/esp-idf/files/12126601/ot_cli_usb.patch), by excuting `git apply ot_cli_usb.patch` under the esp-idf folder. Then the `ot_cli` should work with the USB port.
> 
> We will merge the change to esp-idf soon.

Thank you so much for the patch, it can confirm that it works on my side.

closed 2023-07-21 by thedevleon