espressif/esp-idf #11925: Manual override for flasher_args.json to mark "encrypted" true

rretanubun, 2023-07-20:
### Is your feature request related to a problem?

Follow up to https://github.com/espressif/esp-idf/issues/11909

We found `CONFIG_SECURE_BOOT_FLASH_BOOTLOADER_DEFAULT` really useful for producing a flasher_args.json that a another (non `idf.py` tool) can take an post process for production deployment.

unfortunately, in the secure build config, the `encrypted` entry remains "false"

### Describe the solution you'd like.

A similar `override` to `CONFIG_SECURE_BOOT_FLASH_BOOTLOADER_DEFAULT` that can mark entries to be `encrypted" : "true", like this:

    "bootloader" : { "offset" : "<snip>", "file" : "bootloader/bootloader.signed.bin", "encrypted" : "true" },
    "app" : { "offset" : "<snip>", "file" : "application.signedbin", "encrypted" : "true" },
    "partition-table" : { "offset" : "0xd000", "file" : "partition_table/partition-table.bin", "encrypted" : "true" },
    "otadata" : { "offset" : "0x27000", "file" : "ota_data_initial.bin", "encrypted" : "true" },

even if this is not an output that `idf.py flash` is not supposed to use.

### Describe alternatives you've considered.

right now we are doing manual `sed` string substitution on the output of `flasher_args.json`

mahavirj, 2023-07-21:
> unfortunately, in the secure build config, the encrypted entry remains "false"

Can you please share your sdkconfig for flash encryption related options? Please note that the `encrypted` flag in the flasher config would come into picture provided the flash encryption feature is enabled in the development mode (`CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT` set to `y`).

rretanubun, 2023-07-21:
@mahavirj : Of course, here is all the `cat sdkconfig | grep "ENCRYPT"` for our build

```
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT is not set
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
# CONFIG_PARTITION_TABLE_SINGLE_APP_ENCRYPTED_NVS is not set
CONFIG_NVS_ENCRYPTION=y
CONFIG_SPI_FLASH_ENABLE_ENCRYPTED_READ_WRITE=y
CONFIG_FLASH_ENCRYPTION_ENABLED=y
# CONFIG_FLASH_ENCRYPTION_INSECURE is not set
```

I understand that `idf.py` has its own logic to mark `encrypt: false` in the flasher args. In our system we only use `idf.py build` to generate binaries and `flasher_args.json` and then uses a different tooling to stage/sign binaries into the device.

I am looking for advice for a good incision point to make all the entries in `flasher_args.json` be listed as `encrypt: true` (even if it is a HACK override that can't be mainlined).

dobairoland, 2023-07-25:
> even if this is not an output that idf.py flash is not supposed to use.

`flasher_args.json` is an internal file ESP-IDF tools supposed to use. Putting there items which our tools would ignore could be misleading. Honestly I don't think we or you should try to do this.

If you want to use that file then you can even parse it as it is in standard JSON format, make any necessary changes and write it out your tools would use. But be aware, that the format of the file can change at any time since it is considered as an internal file - rules for ESP-IDF semantic versions don't apply there.

If I might suggest that you can get the same information from `sdkconfig.json` and then write your own flasher_args-like JSON file for your tools.

closed 2023-08-02 by dobairoland